espressif/esptool #594: Flash Size support of 512KB for ESP32 (current minimum is 1MB)

shachar-limor, 2021-01-13:
Hi,
I'd like to use the ESP32-S0WD with a small binary, no OTA, to be fitted into a 4Mb == 512KB SPI flash.
However the toolchain only supports a minimum of 1MB flash.
Can you explain why?
Is it possible to add support for 512KB flash?

Thanks,
Shachar

Using:
Win10 64bit
ESP-IDF v4.3-dev-2136-gb0150615d-dirty
Python 3.7.3
ESP32-S0WD (custom devboard)
Eclipse 2012-9
Eclipse plugin info:

Executing C:\Users\ShacharLimor\AppData\Local\Programs\Python\Python37\python.exe C:\Users\ShacharLimor\esp\esp-idf\tools\idf_tools.py list
WARNING: directory for tool riscv-none-embed-gcc version riscv-none-embed-gcc-8.2.0 is present, but tool was not found
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* riscv-none-embed-gcc: Toolchain for RISC-V (optional)
  - riscv-none-embed-gcc-8.2.0 (recommended)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.10.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

dobairoland, 2021-01-14:
Hi @shachar-limor. Personally, I don't have access to such small flash and never tried by myself but looking at the code I think it is already supported:

```python
DETECTED_FLASH_SIZES = {0x12: '256KB', 0x13: '512KB', 0x14: '1MB',
                        0x15: '2MB', 0x16: '4MB', 0x17: '8MB', 0x18: '16MB'}
```

Are you running `esptool.py --chip esp32 --port <YOURPORT> flash_id` or you are attempting to do something else?

shachar-limor, 2021-01-14:
Hi,
1) Documentation shows minimum is 1MB
[image]

2) I am using eclipse plugin which then triggers esptool.py - configuration via "menuconfig" in eclipse only has a 1MB-16MB supported 
[image]

So the smallest binary file built by the tools is set to 1MB.
I am able to build a "hello world" and actually flash it in to a 4Mb flash, but then at runtime, when the startup code of the boot loader runs, it flags the mismatch between actual flash size (512KB) versus binary size (1024KB = 1MB).

I am running ESP32, I believe the smaller flash sizes are used for ESP8266.
What do you think?
Shachar

dobairoland, 2021-01-14:
Ok, I see. You are right. Only flash sizes from 1MB are supported for ESP32.

I asked around and the reason is because smaller sizes are not practical if you have some connectivity (wifi or bluetooth) enabled which is the primary purpose of ESP32 as being an IoT chip.

512KB seems to be supported by the ROM code. Can you please try to disable the check in the bootloader? Maybe that will be enough for you. I'm sorry I cannot check this as I don't have access to HW with that small flash.

shachar-limor, 2021-01-14:
Hi Roland,
Thanks for your comments and I will give it a try.
My motivation is obviously to get the BOM cost down to the minimum while still supporting the required functionality for my application.

I was wondering if there is a cleaner way ("officially" using the toolchain) to enable it, not just disable the check. I don't want to hack things to much and then run into trouble later..

FYI, my app is based on the BLE gatt_server_service_table running on a single core ESP32 - I can get the footprint smaller by disabling the BLE SM (Security Module).
I might be reading this incorrectly but is the footprint down to 384KB? Or maybe I will be able to drop some additional non-critical items and reduce the size more.
Anyway if this is smaller than 512KB it should fit just fine, right? Or am I missing something here?
[image]

Thanks,
Shachar

dobairoland, 2021-01-14:
Hi Shachar. But unfortunately, the opposite can happen as well, i.e. you have to add something and the size increases rapidly. 

Otherwise, your assessment seems to be right.

Please take a look at the last paragraph of https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spi_flash.html#spi-flash-size which mentions the possibility to overwrite the chip size at runtime. That might help you as well. 

I don't think that expecting official support for this is realistic given that breaking changes would have to be added for various tools and usually nobody really needs this. But probably some of the possible workarounds might work for you.

shachar-limor, 2021-01-17:
Dear Roland,
Since we are going for mass production, I wanted to save a few $$ in the cost of SPI (well, cents is more accurate but it is still significant in may application).
Rethinking this - I think that using the "main road" and sticking to the tools' officially supported size is more robust and will be a better choice for mass production, it is probably worth the extra cost.

Anyway, thanks for your support, it was very educational for me.
Regards,
Shachar

shachar-limor, 2021-01-17:
Issue has been clarified. No further pursuit needed from my side.

closed 2021-01-17 by shachar-limor