espressif/esptool #595: esptool tries to patch the sha256 checksum at the wrong offset

erikcorry, 2021-01-14:
* Operating system:
Linux

* Python version: (`python -V` to check this)
Python 2.7.17

# Full esptool.py command line as run:
esptool.py --chip esp32 elf2image --flash_mode dio --flash_freq 40m --flash_size 4MB --min-rev 0 --elf-sha256-offset 0xb0 -o /home/erik/Toitware/toit/build/esp32_esp32_display/toit.bin /home/erik/Toitware/toit/build/esp32_esp32_display/toit.elf

# Full output from esptool.py

A fatal error occurred: Contents of segment at SHA256 digest offset 0xb0 are not all zero. Refusing to overwrite.

# What is the expected behaviour?

Create a bin file

# Do you have any other information from investigating this?

I have a pull request on its way to fix this

# Is there any other information you can think of which will help us reproduce this problem?

I'm using the 4.1 release branch of esp-idf and the espressif master branch of esptool_py 8270199920415049eea65990a1ca93a549113979

dobairoland, 2021-01-14:
Hi @erikcorry. Please re-check the issue with the latest master branch because I think this was probably fixed one month later than the commit you are using: https://github.com/espressif/esptool/commit/365a0ba96305e0cbde32cfa8305207d67f811917

This is a possible duplicate of #560.

erikcorry, 2021-01-14:
The issue was a bit different to the one I thought I had.
We 4k-align the rodata section with an assembly instruction.  The fixed 0xe0 offset for the sha256 hash assumes that the rodata is at a certain address 0x20.  (What is the first 0x20 part?).

I guess I should make a new read-only section for my data that needs to be aligned.

dobairoland, 2021-01-15:
@erikcorry 

```c
/*
    (0x20 offset above is a convenience for the app binary image generation.
    Flash cache has 64KB pages. The .bin file which is flashed to the chip
    has a 0x18 byte file header, and each segment has a 0x08 byte segment
    header. Setting this offset makes it simple to meet the flash cache MMU's
    constraint that (paddr % 64KB == vaddr % 64KB).)
  */
```
(source https://github.com/espressif/esp-idf/blob/master/components/esp32/ld/esp32.ld) The comment is for `iram0_2_seg` and not for `drom0_0_seg` but I assume that the same applies.

erikcorry, 2021-01-15:
The technique of replacing the sha checksum at a fixed offset seems quite fragile to me.  I would prefer if the position was specified as the name of an ELF section, and an offset within that section.  Then it would still work if the section was moved for some reason. But perhaps there is code somewhere that needs it to be at exactly offset 0xe0 in the ROM image?

dobairoland, 2021-01-15:
The offset is hardcoded and is not subject to change: https://github.com/espressif/esp-idf/blob/master/components/esptool_py/project_include.cmake#L29

I don't see anything wrong with esptool. Can we close the issue?

www.esp32.com could be a better place for asking questions regarding the chip architecture and design decision. You might get better answers there from the admins.

closed 2021-01-15 by erikcorry