espressif/esptool #597: espefuse - Flash Encryption burn_key command with ESP32-S2 location invalid error

ThomasCham, 2021-01-19:
# Operating system

Windows 10

# Python version

Python 3.7

# What Chip

ESP32-S2

# What development board or other hardware is the chip attached to

ESP-Prog

# Is anything else attached to the development board, except for the serial flasher connections?

No

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

No IDE, Windows command prompt

# Full esptool.py command line that was run:

espefuse.py --port COM33 burn_key flash_encryption my_flash_encryption_key.bin

# Full output from esptool.py

Connecting....
Detecting chip type... ESP32-S2
usage: espefuse.py burn_key [-h] [--no-write-protect] [--no-read-protect]
                            [--force-write-always]
                            {MAC_SPI_8M_0,BLOCK1,BLOCK_SYS_DATA,BLOCK2,BLOCK_USR_DATA,BLOCK3,BLOCK_KEY0,BLOCK4,BLOCK_KEY1,BLOCK5,BLOCK_KEY2,BLOCK6,BLOCK_KEY3,BLOCK7,BLOCK_KEY4,BLOCK8,BLOCK_KEY5,BLOCK9,BLOCK_SYS_DATA2,BLOCK10}
                            keyfile
                            {USER,RESERVED,XTS_AES_256_KEY_1,XTS_AES_256_KEY_2,XTS_AES_128_KEY,HMAC_DOWN_ALL,HMAC_DOWN_JTAG,HMAC_DOWN_DIGITAL_SIGNATURE,HMAC_UP,SECURE_BOOT_DIGEST0,SECURE_BOOT_DIGEST1,SECURE_BOOT_DIGEST2}
                            [BLOCK] [KEYFILE] [KEYPURPOSE] [BLOCK] [KEYFILE]
                            [KEYPURPOSE] [BLOCK] [KEYFILE] [KEYPURPOSE]
                            [BLOCK] [KEYFILE] [KEYPURPOSE] [BLOCK] [KEYFILE]
                            [KEYPURPOSE] [BLOCK] [KEYFILE] [KEYPURPOSE]
                            [BLOCK] [KEYFILE] [KEYPURPOSE] [BLOCK] [KEYFILE]
                            [KEYPURPOSE] [BLOCK] [KEYFILE] [KEYPURPOSE]
                            [BLOCK] [KEYFILE] [KEYPURPOSE] [BLOCK] [KEYFILE]
                            [KEYPURPOSE] [BLOCK] [KEYFILE] [KEYPURPOSE]
                            [BLOCK] [KEYFILE] [KEYPURPOSE] [BLOCK] [KEYFILE]
                            [KEYPURPOSE] [BLOCK] [KEYFILE] [KEYPURPOSE]
                            [BLOCK] [KEYFILE] [KEYPURPOSE] [BLOCK] [KEYFILE]
                            [KEYPURPOSE] [BLOCK] [KEYFILE] [KEYPURPOSE]
                            [BLOCK] [KEYFILE] [KEYPURPOSE] [BLOCK] [KEYFILE]
                            [KEYPURPOSE]
espefuse.py burn_key: error: argument block: invalid choice: 'flash_encryption' (choose from 'MAC_SPI_8M_0', 'BLOCK1', 'BLOCK_SYS_DATA', 'BLOCK2', 'BLOCK_USR_DATA', 'BLOCK3', 'BLOCK_KEY0', 'BLOCK4', 'BLOCK_KEY1', 'BLOCK5', 'BLOCK_KEY2', 'BLOCK6', 'BLOCK_KEY3', 'BLOCK7', 'BLOCK_KEY4', 'BLOCK8', 'BLOCK_KEY5', 'BLOCK9', 'BLOCK_SYS_DATA2', 'BLOCK10')

# Do you have any other information from investigating this?

I was following the guide here to enable flash encryption for my device: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/security/flash-encryption.html#using-host-generated-key

When trying to run the command under Step 3, the tool failes with this error implying that the location 'flash_encryption' BLOCK doesn't exist? Any ideas on which location should be used for the Flash Encryption key? 

Thanks

radimkarnis, 2021-01-19:
Hello @ThomasCham,

I am sorry for any inconvenience caused. 

[The guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/security/flash-encryption.html#using-host-generated-key) you are referring too seems to contain information that does not apply to ESP32-S2. The `flash_encryption` alias for BLOCK1 is a remnant from the ESP32 docs.

The ESP32-S2 (and C3) SoCs implement a new XTS-AES compliant algorithm for flash encryption. The 256 bit (or 512 bit) long key is stored in one or two eFuse blocks from BLOCK4 to BLOCK9. The exact information is available in the [technical reference manual](https://www.espressif.com/sites/default/files/documentation/esp32-s2_technical_reference_manual_en.pdf), chapter `29. External Memory Encryption and Decryption (XTS_AES)`.

We will fix and verify the documentation. I would advise you to wait for the updated docs, as any wrong eFuses related command can brick your device.

If you still want to proceed, the correct command for S2 in this case is: 
`espefuse.py --port PORT burn_key BLOCK_KEY0 my_flash_encryption_key.bin XTS_AES_128_KEY`.

Thank you for reporting this issue.
Radim

ThomasCham, 2021-01-19:
Hi @radimkarnis,

Thanks for looking into this so quickly! I thought that might be the case as the ESP32 and ESP32-S2 webpages for those topics were identical. I did check the TRM like you said and saw BLOCK_KEY0 could be the one I was after but didn't want to risk bricking my device.

I'd like to enable both Flash Encryption and Secure Boot V2 on my device. Is it possible you could check the Secure Boot V2 page for ESP32-S2 also to check it has the correct info?: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/security/secure-boot-v2.html

Please let me know when the documentation is updated.

Thanks,
Thomas

radimkarnis, 2021-01-19:
@ThomasCham, the Secure Boot V2 information is correct! 

I will notify you once the docs are updated.

radimkarnis, 2021-02-04:
Hello @ThomasCham,

the ESP32-S2 docs have been updated. It is now safe to follow the flash encryption using host generated key guide.

Radim

closed 2021-02-04 by radimkarnis