espressif/esptool #598: "sudo esptool.py" gives error "ModuleNotFoundError: No module named 'serial' "

manibharathytu, 2021-01-21:
# Operating system
Ubuntu 20.04

# Python version
Python 3.8.5

# What Chip
ESP8266

# What development board or other hardware is the chip attached to
ESP-01

If your hardware is custom or unusual, please attach a photo to the issue.
Just ESP-01 with usb to serial programmer made for ESP-01 itself
Photo : https://i.imgur.com/9ISUE9s.png

# Is anything else attached to the development board, except for the serial flasher connections?
No. Only programmer and ESP-01

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
Running from command line.

# Full esptool.py command line that was run:
sudo python3 /home/mani/.local/bin/esptool.py -p /dev/ttyUSB0 write_flash 0x01000 /home/mani/Downloads/tasmota.bin

# Full output from esptool.py
```
mani@mani-Latitude-E6540:~/.local/bin$ sudo python3 /home/mani/.local/bin/esptool.py -p /dev/ttyUSB0 write_flash 0x01000 /home/mani/Downloads/tasmota.bin
Pyserial is not installed for /usr/bin/python3. Check the README for installation instructions.
Traceback (most recent call last):
  File "/home/mani/.local/bin/esptool.py", line 38, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'
```
Image: https://i.imgur.com/XRINRBr.png

# Do you have any other information from investigating this?
Cmd : esptool.py
Output : (Prints the usage info.) 
Thought : esptool.py installed fine.

Cmd : esptool.py -p /dev/ttyUSB0 write_flash 0x01000 /home/mani/Downloads/tasmota.bin
Error : serial.serialutil.SerialException: [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'
Thought : May be I should try sudo.

Cmd : sudo esptool.py -p /dev/ttyUSB0 write_flash 0x01000 /home/mani/Downloads/tasmota.bin
Error : sudo: esptool.py: command not found
Thought : May be I should use the full path to esptool.py. Found the path with whereis cmd.

Cmd : sudo /home/mani/.local/bin/esptool.py -p /dev/ttyUSB0 write_flash 0x01000 /home/mani/Downloads/tasmota.bin
Error : ModuleNotFoundError: No module named 'serial'
Thought : From a python3 terminal, I'm able to import serial successfully though. May be I should use python3. Although the esptool.py has python3 in its first line. "#!/usr/bin/python3"

Cmd : sudo python3 /home/mani/.local/bin/esptool.py -p /dev/ttyUSB0 write_flash 0x01000 /home/mani/Downloads/tasmota.bin
Error : ModuleNotFoundError: No module named 'serial'

# Is there any other information you can think of which will help us reproduce this problem?

Before ubuntu, I tried in windows with Tasmotizer exe. I tried to fash with it. It gave the error
"Timed out waiting for packet header"

Not sure if this info is useful as now I'm using ubuntu and esptool.py.

dobairoland, 2021-01-21:
Hi @manibharathytu. In gneral, the use of `sudo` is not recommended for security concerns. Moreover, in this case pyserial was installed for your user and not for root. That is why you getting the ModuleNotFoundError.

The issue is related to your user account not having access to the port. This is the related section from the README:

> In Linux, the current user may not have access to serial ports and a "Permission Denied" error will appear. On most Linux distributions, the solution is to add the user to the dialout group with a command like sudo usermod -a -G dialout <USERNAME>. Check your Linux distribution's documentation for more information.

dobairoland, 2021-01-26:
I assume that the above solves the issue and closing this.

closed 2021-01-26 by dobairoland