espressif/esptool #600: How i can get progress from a funtion while burning ?

xuhongv, 2021-02-06:
* Operating system: windows10
* Python version: 3.8

# question:

How i can get progress while burning flash ?  
Can you provide me with such an interface ？

Thanks.

dobairoland, 2021-02-09:
Hi @xuhongv. What progress have you in mind? 

Esptool already gives progress about burning flash, i.e. amount already written in the given moment, total image size, percentage of task completion.

xuhongv, 2021-02-09:
Oh ，I understand that the file Esptool.py amount already written in the given moment, total image size, percentage of task completion.

but , I want to not alter the Esptool.py Code and I want to get progress about burning flash  to make PC flasher by PyQT5.

123swk123, 2021-02-09:
Use signal and slots in pyQt. emit the signal at your desired location in write_flash function Regards, Karthikeyan Sw

xuhongv, 2021-02-09:
Thanks. I

123swk123, 2021-02-09:
For option 1
```python
while len(image) > 0:
            print_overwrite('Writing at 0x%08x... (%d %%)' % (address + seq * esp.FLASH_WRITE_SIZE, 100 * (seq + 1) // blocks))
            sys.stdout.flush()
            block = image[0:esp.FLASH_WRITE_SIZE]
# emit signal here 
            if compress:
                esp.flash_defl_block(block, seq, timeout=timeout_per_mb(COMP_BLOCK_WRITE_TIMEOUT_PER_MB, uncsize))
            else:
                # Pad the last block
                block = block + b'\xff' * (esp.FLASH_WRITE_SIZE - len(block))
                if encrypted:
                    esp.flash_encrypt_block(block, seq)
                else:
                    esp.flash_block(block, seq)
            image = image[esp.FLASH_WRITE_SIZE:]
            seq += 1
            written += len(block)
```

dobairoland, 2021-02-09:
Esptool is a terminal program and currently there is no plan to add support for graphical libraries.

Thank you @123swk123 for helping with possible workarounds.

closed 2021-02-09 by dobairoland