espressif/esptool #601: ESP32 Wrover Kit v4: Timeout when flashing

sylents-dev-e, 2021-02-08:
Hello all,

i have the following error when flashing my device:
CODE: SELECT ALL

Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
Setup:
- ESP-IDF 4.2
- ESP32 Wrover Kit v4
- USB Port: Interface0 = WinUSB; Interface1 = FTDIBUS
- Tried powering with USB and externally, does not matter, tried connecting 10uF between EN/GND, tried pushing BOOT button while flashing

I was aiming to test my CAN/TWAI-Bus with an external CAN-Tranceiver (SN65HVD230 CAN Board). I used < 4.2 ESP-IDF before and after a updated to 4.2 i had to use the TWAI-Interface. From then on the CAN/TWAI-Bus did unexpected things. So i decided to test the twai_network_master example. When i ran it (can_tx=GPIO21 and can_rx=GPIO22) the tranceiver became incredibly hot and the boards LEDs were turning red. After this incident the described flashing problem popped out. In case i want to flash the LED D6 blinks and the timeout error occurs during flashing. Apparently, i caused some kind of short circuit somewhere. When i connect the can transceiver to VCC/GND it still gets overheated (even without can_tx/rx connected) hence this IC is dead i would say.
Do you think the board itself is kind of broken?

Thank you in advance!

Greets

dobairoland, 2021-02-09:
Hi @sylents-dev-e. 

You can try to invoke esptool with the `--trace` argument to see the details about the communication. Does it show some successful read or just write?

You can test it even without esptool. You can connect a serial terminal program to the port and reset the board manually into bootloader mode. Do you receive anything? 

Wrover kit is very well tested with esptool. I don't think additional capacitor or pushing the boot button during flashing would be necessary.

sylents-dev-e, 2021-02-09:
Hello @dobairoland,

thank you for your reply.

`C:PATH>esptool --trace erase_flash
esptool.py v3.0
Found 1 serial ports
Serial port COM5
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.115 Timed out waiting for packet header
.TRACE +0.065 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU`

How to reset it via terminal? I don't receive anything at this point.

dobairoland, 2021-02-09:
> How to reset it via terminal? I don't receive anything at this point.

You connect the terminal and then you reset the board by the buttons on the Wrover kit.

sylents-dev-e, 2021-02-09:
yes as i already mentioned unfortunatly, there is no reaction coming form the board.

dobairoland, 2021-02-09:
I'm sorry to tell you but in that case esptool cannot do anything. The issue is with your hardware.

dobairoland, 2021-02-09:
I can suggest to ask about this on https://esp32.com/ in the Hardware sub-forum. I have no idea what caused this or if you can save your hardware.

closed 2021-02-09 by dobairoland

sylents-dev-e, 2021-02-09:
Yeah thank you, i opened an issue there some days ago.