espressif/esptool #602: 'Serial' object has no attribute 'dtr'

SaudAlfurhud, 2021-02-09:
# Operating system
macOS 10.15.7 

# Python version
Python 3.9.

# What Chip
ESP8266 (ESP-01)

# What development board or other hardware is the chip attached to
Arduino UNO

# Is anything else attached to the development board, except for the serial flasher connections?
No

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
Arduino ESP8266 IDE version 1.8.13

# Full esptool.py command line that was run:
`python esptool.py -p /dev/cu.usbmodem141101 erase_flash`

# Full output from esptool.py

```
esptool.py v3.1-dev
Serial port /dev/cu.usbmodem141101
Connecting...
Traceback (most recent call last):
  File "esptool.py", line 4290, in <module>
    _main()
  File "esptool.py", line 4283, in _main
    main()
  File "esptool.py", line 3866, in main
    before=args.before)
  File "esptool.py", line 115, in get_default_connected_device
    connect_attempts)
  File "esptool.py", line 346, in detect_chip
    detect_port.connect(connect_mode, connect_attempts, detecting=True)
  File "esptool.py", line 566, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False)
  File "esptool.py", line 517, in _connect_attempt
    self._setRTS(True)   # EN=LOW, chip in reset
  File "esptool.py", line 489, in _setRTS
    self._port.setDTR(self._port.dtr)
AttributeError: 'Serial' object has no attribute 'dtr'
```

# Is there any other information you can think of which will help us reproduce this problem?
I also have a Windows 10 PC, esptool command works correctly on this PC.

radimkarnis, 2021-02-09:
Hello @SaudAlfurhud,

I would recommend reading through this discussion (https://github.com/espressif/esptool/issues/324) about the same issue. It mentions a few ideas that might help you!

Let me know if you manage to fix your problem.
Radim

SaudAlfurhud, 2021-02-09:
> Hello @SaudAlfurhud,
> 
> I would recommend reading through this discussion (#324) about the same issue. It mentions a few ideas that might help you!
> 
> Let me know if you manage to fix your problem.
> Radim

SOLVED.
Thank you for your replay after many tries and a lot of search i found the tool doesn't work with python2, only work with python3 

the command thats work:
`python esptool.py -p /dev/cu.usbmodem141101 erase_flash`

radimkarnis, 2021-02-09:
@SaudAlfurhud, I am glad you were able to solve your issue!

Esptool is actually both Python 2 and 3 compatible, so there might be a chance that you have an older version (<3.0) of the `pyserial` package installed in your Python 2 environment.

I am closing this issue. Reopen if you feel like further discussion is needed.

closed 2021-02-09 by radimkarnis