espressif/esptool #605: MacOS Big Sur esptool.py not found in /bin

matekocchimilk, 2021-02-23:
I have a problem like [https://github.com/espressif/esptool/issues/540](url) is 

> File "esptool.py", line 57, in <module>
>   File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
>   File "serial/tools/list_ports.py", line 29, in <module>
>   File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
>   File "serial/tools/list_ports_posix.py", line 31, in <module>
>   File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
>   File "serial/tools/list_ports_osx.py", line 32, in <module>
> ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found

but I can't found "/Library/Frameworks/Python.framework/Versions/3.8/bin/esptool.py" to comment

when typing `pip show -f esptool` the esptool was located in "/Users/USERNAME/esptool" the right path must be  `something/bin/`

# Operating system
MacOS Big Sur 11.2.1

# Python version
Python 3.8.6rc

# What Chip

ESP8266 ESP32dev module

# What development board or other hardware is the chip attached to

KB32-FT Starterkit

martinespgano, 2021-02-24:
Hi, @matekocchimilk, thank you for opening an issue. Could you please provide us with the command that fails with this issue? Please, also try to `python setup.py install` in the esptool repository.

matekocchimilk, 2021-02-25:
Thank you for replying, The failed command is while I'm compiling code show errors above. I try  `python setup.py install` . esptool.py is located in `/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages ` (the current `python.framework` directory)  but  inside esptool.py has nothing that can comment (ref. on [https://github.com/espressif/esptool/issues/url](url) )

In esptool.py file

>  #!/usr/local/bin/python
>   #EASY-INSTALL-SCRIPT: 'esptool==3.1.dev0','esptool.py'
> __requires__ = 'esptool==3.1.dev0'
> __import__('pkg_resources').run_script('esptool==3.1.dev0', 'esptool.py')

martinespgano, 2021-02-26:
Thank you for the information. There is a chance that the problem is caused by the library `pyserial` that esptool uses. Could you please provide a version of `pyserial` that you are using?

matekocchimilk, 2021-02-26:
Thank you for advice, I checked my pyserial version  by typing `pip show pyserial`in command line. My pyserial version is **3.5** . The pyserial is located in `/Users/Library/Python/2.7/lib/python/site-packages` .

dobairoland, 2021-03-05:
Hi @matekocchimilk. I think your Python environments are somehow mixed together. Some of your Python paths contain 2.7 and some of them 3.8. The best way is to use a `python` executable consistently with all commands so it would work with the same environment.

```sh
python --version   # check that this is there right version you want to use
which python     # check where it was installed
python -m pip install esptool          # sometimes pip uses other python. This will eliminate that problem
python -m esptool version            # this is the esptool we just installed. Invoke esptool this way in order to remain in the same environment 
python -m pip freeze | grep pyserial   # check that pyserial is of version 3.5 here as well 
```

Similarly, if you want to use the latest esptool from this repo then you can do the following:
```
python setup.py build install
python -m esptool <all the required arguments>       # ensures that the esptool installed in the previous line will be used 
```

dobairoland, 2021-03-14:
I'm closing this because I believe this is not an issue on the esptool side. Feel free to comment if you need further assistance with this.

closed 2021-03-14 by dobairoland