espressif/esptool #606: Download mode works on linux but not on windows

joelsernamoreno, 2021-02-26:
Hello!

I am trying to flash a code on ESP32-PICO through Arduino IDE by selecting the ESP32 Generic board.

It is important for you to know that my device is a custom board with ESP32-PICO and it does not have a download mode button, it only has reset button.

If I upload a code to the board with arduino IDE in Ubuntu 18.04, esptool automatically detects my device and uploads the code without pressing the reset button. My device enters download mode automatically

My problem is when I try to do this on Windows 10 with Arduino IDE. Esptool does not connect to esp32 and I get a timed out waiting for packet header error.

Do you know why my device starts in download mode on linux without pressing any button and on windows it doesn't work? I have tried pressing the reset button on windows, but that doesn't work either. ESP32-PICO does not enter download mode automatically. If it helps, my device uses USB to UART CP210N.

Here you have more information:

# Operating system
Ubuntu 18.04 and Windows 10

# Python version
Python 2.7.17

# What Chip
ESP32-PICO (Custom device)

[image]

# Is anything else attached to the development board, except for the serial flasher connections?

2x CC110 modules:
sck = 18
int miso = 19
int mosi = 23
int csn1 = 5
int csn2 = 27
int gdo0_1 = 2
int gdo2_1 = 4
int gdo0_2 = 25
int gdo2_2 = 26

3x pushbuttons
push1 = 34
push2 = 35
push3 = 32

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

Arduino IDE 1.8.13 and ESP32 by espressif system 1.0.4 (Windows 10 and Ubuntu 18.04)

Thanks!

dobairoland, 2021-02-27:
Hi @joelsernamoreno. Unfortunately, this is a very common issue with custom boards. The same board works with Linux but not with Windows. Or this can also happen when one switches from one Windows machine to another one. The issue is with the Windows USB driver and not with ESPTOOL.

- If you connect a logic analyzer to the EN & GPIO0 pins then you will see different results even they were generated with the same ESPTOOL.
- You can try to connect GPIO0 to GND with a cable during flash. This should help to put the device into download mode and flash it. This might be inconvenient if you want to flash the device many times.
- You can try to add a capacitor between EN and GND. All boards designed by Espressif Systems have this. This is the main reason this issue is present only for custom boards. 
- You can try to use some kind of bridge between the device and the computer, e.g. a USB 2.0 hub.

chegewara, 2021-03-04:
Hi @dobairoland 
i dont want to start new issue yet, because your answer my be explanation of my issue. Here is my story (i hope it is not hijacking):
- arduino IDE on windows,
- board: esp32 S2 saola 1W,
- flashing over USB,
- when i enter bootload mode with buttons all works fine,
- when i am using auto-enable mode then board is switching to download mode (i see `waiting for download` message), but after couple seconds chip is restarted
Here you can see logs from bootloader and from arduino IDE:
```
12:21:36.286 -> ESP-ROM:esp32s2-rc4-20191025
12:21:36.286 -> Build:Oct 25 2019
12:21:36.286 -> rst:0x3 (RTC_SW_SYS_RST),boot:0x0 (DOWNLOAD(USB/UART0/1/SPI))
12:21:36.286 -> Saved PC:0x40025fe9
12:21:36.286 -> waiting for download
12:21:41.847 -> ESP-ROM:esp32s2-rc4-20191025
12:21:41.847 -> Build:Oct 25 2019
12:21:41.847 -> rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
12:21:41.847 -> Saved PC:0x400000c6
12:21:41.847 -> SPIWP:0xee
12:21:41.881 -> mode:DIO, clock div:1
12:21:41.881 -> load:0x3ffe6100,len:0x8
12:21:41.881 -> load:0x3ffe6108,len:0x620
12:21:41.881 -> load:0x4004c000,len:0xa40
12:21:41.881 -> load:0x40050000,len:0x284c
12:21:41.881 -> entry 0x4004c190
```
```
esptool.py v3.0-dev
Serial port COM7
Connecting........_
Traceback (most recent call last):
  File "esptool.py", line 3682, in <module>
  File "esptool.py", line 3675, in _main
  File "esptool.py", line 3330, in main
  File "esptool.py", line 515, in connect
  File "esptool.py", line 492, in _connect_attempt
  File "esptool.py", line 431, in sync
  File "esptool.py", line 369, in command
  File "esptool.py", line 332, in write
  File "site-packages\serial\serialwin32.py", line 323, in write
serial.serialutil.SerialTimeoutException: Write timeout
```

Thanks

dobairoland, 2021-03-04:
Hi @chegewara. Could you please try with esptool from the master branch? Please also use the `--trace` option so I can see what is happening.

I don't have too much experience with Saola boards on Windows yet. I've tried to do flash and worked for me. I've noticed that when I put the chip into bootloader mode then the port number has changed. I'm wondering if this could happen in your case. Or is it always COM7 in your case? I have COM6 in normal mode and COM5 in bootloader mode. However, I can flash & monitor by setting the port to COM6. I haven't connected the USB-to-UART cable to the micro USB port so it could not interfere.

chegewara, 2021-03-04:
Ok, i will try.

I have implemented now persistent mode in my library, so port number is not changing. Also ive been trying with espressif USB library by setting "Serial connected to USB", which is enabling persistent mode and also does not change port. In both cases i am having the same result as described above.

Until now i tested with esptool.py from IDF commit `b0150615dff529662772a60dcb57d5b559f480e2`, but the same result.
Sadly i dont know how to enable `--trace` in arduino.

I am testing on virtualbox VM, where USB is passed to VM and UART is monitoring on linux thats how i know whats going on.

dobairoland, 2021-03-04:
> Sadly i dont know how to enable --trace in arduino.

You can use ESP-IDF Command Line (cmd.exe + export.bat) and run esptool.py directly from there.

> I am testing on virtualbox VM

I don't think it will work in Virtualbox. I always have to hold the boot button during flashing when the flashing is done from Virtualbox guest. I had this issue with Linux host & Windows guest, Windows host & Linux guest, on different computers with different ESP chips. I don't think Virtualbox is able to transfer the DTR & RTS signals without introducing additional delays between them.

chegewara, 2021-03-05:
I tested on baremetal windows 10 with esptool installed with `pip install esptool`.  It works nicely on ubuntu 20 and here is result from windows:
```
C:\Windows\system32>esptool.py --port com3 chip_id
esptool.py v3.0
Serial port com3
Connecting....
Traceback (most recent call last):
  File "C:\Users\darek\AppData\Local\Programs\Python\Python38\Scripts\esptool.py-script.py", line 33, in <module>
    sys.exit(load_entry_point('esptool==3.0', 'console_scripts', 'esptool.py')())
  File "c:\users\darek\appdata\local\programs\python\python38\lib\site-packages\esptool.py", line 3962, in _main
    main()
  File "c:\users\darek\appdata\local\programs\python\python38\lib\site-packages\esptool.py", line 3541, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace,
  File "c:\users\darek\appdata\local\programs\python\python38\lib\site-packages\esptool.py", line 309, in detect_chip
    detect_port.connect(connect_mode, connect_attempts, detecting=True)
  File "c:\users\darek\appdata\local\programs\python\python38\lib\site-packages\esptool.py", line 519, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False)
  File "c:\users\darek\appdata\local\programs\python\python38\lib\site-packages\esptool.py", line 499, in _connect_attempt
    self.sync()
  File "c:\users\darek\appdata\local\programs\python\python38\lib\site-packages\esptool.py", line 437, in sync
    self.command(self.ESP_SYNC, b'\x07\x07\x12\x20' + 32 * b'\x55',
  File "c:\users\darek\appdata\local\programs\python\python38\lib\site-packages\esptool.py", line 376, in command
    self.write(pkt)
  File "c:\users\darek\appdata\local\programs\python\python38\lib\site-packages\esptool.py", line 339, in write
    self._port.write(buf)
  File "c:\users\darek\appdata\local\programs\python\python38\lib\site-packages\serial\serialwin32.py", line 325, in write
    raise SerialTimeoutException('Write timeout')
serial.serialutil.SerialTimeoutException: Write timeout
```
Here is also change of state lines during that command call (DTR/RTS):
```
new bitrate: 9600
connection state changed, dtr: 1, rts: 1
new bitrate: 9600
new bitrate: 115200
connection state changed, dtr: 1, rts: 1
new bitrate: 115200
new bitrate: 115200
connection state changed, dtr: 1, rts: 1
new bitrate: 115200
connection state changed, dtr: 0, rts: 1 
connection state changed, dtr: 0, rts: 1
connection state changed, dtr: 1, rts: 1
connection state changed, dtr: 1, rts: 0
connection state changed, dtr: 0, rts: 0
new bitrate: 115200
connection state changed, dtr: 0, rts: 0
new bitrate: 115200
new bitrate: 115200
connection state changed, dtr: 0, rts: 0
new bitrate: 115200
connection state changed, dtr: 0, rts: 0
```

Thanks

chegewara, 2021-03-05:
Similar result with esptool from idf master:
```
E:\esp-idf\components\esptool_py\esptool>esptool.py --port com3 chip_id
esptool.py v3.1-dev
Serial port com3
Connecting...
Traceback (most recent call last):
  File "E:\esp-idf\components\esptool_py\esptool\esptool.py", line 4290, in <module>
    _main()
  File "E:\esp-idf\components\esptool_py\esptool\esptool.py", line 4283, in _main
    main()
  File "E:\esp-idf\components\esptool_py\esptool\esptool.py", line 3864, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
  File "E:\esp-idf\components\esptool_py\esptool\esptool.py", line 114, in get_default_connected_device
    _esp = ESPLoader.detect_chip(each_port, initial_baud, before, trace,
  File "E:\esp-idf\components\esptool_py\esptool\esptool.py", line 346, in detect_chip
    detect_port.connect(connect_mode, connect_attempts, detecting=True)
  File "E:\esp-idf\components\esptool_py\esptool\esptool.py", line 566, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False)
  File "E:\esp-idf\components\esptool_py\esptool\esptool.py", line 538, in _connect_attempt
    self.sync()
  File "E:\esp-idf\components\esptool_py\esptool\esptool.py", line 476, in sync
    self.command(self.ESP_SYNC, b'\x07\x07\x12\x20' + 32 * b'\x55',
  File "E:\esp-idf\components\esptool_py\esptool\esptool.py", line 413, in command
    self.write(pkt)
  File "E:\esp-idf\components\esptool_py\esptool\esptool.py", line 376, in write
    self._port.write(buf)
  File "C:\Users\darek\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages\serial\serialwin32.py", line 325, in write
    raise SerialTimeoutException('Write timeout')
serial.serialutil.SerialTimeoutException: Write timeout
```

dobairoland, 2021-03-05:
Thank you @chegewara for doing additional experiments. I'm sorry but I cannot advice you right now what to do. 

I don't think that this is an issue with esptool but I admit that somehow this should be addressed. Esptool generates some sequence of DTR and RTS signals and the Windows driver changes it. Esptool has no chance to detect this and come up with another sequence and timing which will Windows transform into what was the intended sequence and timing in the beginning.

I have faced something similar only once with my ESP32-S2-based USB bridge. I had to filter out the DTR=RTS=1 change shown in the middle of your log:
```
connection state changed, dtr: 0, rts: 1 
connection state changed, dtr: 0, rts: 1
connection state changed, dtr: 1, rts: 1
connection state changed, dtr: 1, rts: 0
connection state changed, dtr: 0, rts: 0
```
I achieved this by starting a timer and assigning DTR=RTS=1 only if no further transitions arrived in the following 10 ms. I achieved this in my tinyusb-based application, not in esptool.

So I think something would be possible in ESP-IDF as well, maybe in `components/esp_system/port/soc/esp32s2/usb_console.c`. I haven't figured out this yet and I'm also blocked by the fact that I cannot reproduce this. I'll ask around and see if someone else has encountered this.

Please let me know if you achieve some progress with this or have additional thoughts or ideas.

chegewara, 2021-03-05:
Maybe this will be more helpful:
```
esptool.py v3.0
Serial port com7
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 1 bytes: 01
TRACE +0.000 Read 1 bytes: 08
TRACE +0.025 Read 1 bytes: 04
TRACE +0.000 Read 1 bytes: 00
TRACE +0.005 Read 1 bytes: 07
TRACE +0.000 Read 1 bytes: 07
TRACE +0.000 Read 1 bytes: 12
TRACE +0.000 Read 1 bytes: 20
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 1 bytes: 00
TRACE +0.015 Read 1 bytes: 00
TRACE +0.002 Read 1 bytes: c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.156 Read 17 bytes:
    c001080400070712 2000000000c0c001 | ........ .......
    08                                | .
TRACE +0.015 Received full packet: 010804000707122000000000
TRACE +3.047 Timed out waiting for packet content
.TRACE +0.095 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.124 Timed out waiting for packet header
.TRACE +0.094 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.124 Timed out waiting for packet header
.TRACE +0.095 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.124 Timed out waiting for packet header
.TRACE +0.095 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.124 Timed out waiting for packet header
.TRACE +1.907 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
[... 59 lines cut ...]
22:40:49.874 -> Build:Oct 25 2019
22:40:49.874 -> rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
22:40:49.874 -> Saved PC:0x40012c4f
22:40:49.874 -> SPIWP:0xee
22:40:49.874 -> mode:DIO, clock div:1
22:40:49.874 -> load:0x3ffe6100,len:0x8
22:40:49.874 -> load:0x3ffe6108,len:0x620
22:40:49.907 -> load:0x4004c000,len:0xa40
22:40:49.907 -> load:0x40050000,len:0x284c
22:40:49.907 -> entry 0x4004c190
22:40:50.503 -> DTR1: 0, RTS: 0, state: 0
22:40:50.503 -> DTR0: 0, RTS: 0, state: 0
22:40:50.536 -> DTR1: 0, RTS: 0, state: 0
22:40:50.536 -> DTR0: 0, RTS: 0, state: 0
22:40:50.536 -> Received data: 46
22:40:50.668 -> DTR1: 0, RTS: 0, state: 0
22:40:50.668 -> DTR0: 0, RTS: 0, state: 0
22:40:50.768 -> DTR1: 0, RTS: 0, state: 0
22:40:50.768 -> DTR0: 0, RTS: 0, state: 0
22:41:00.795 -> DTR1: 0, RTS: 0, state: 0
22:41:00.795 -> DTR0: 0, RTS: 0, state: 0
22:41:00.828 -> DTR1: 0, RTS: 0, state: 0
22:41:00.828 -> DTR0: 0, RTS: 0, state: 0
```
Ive been trying for few days to figure this out, but no luck. What i found is when i add small delay exactly before `esp_reset()` then it cause `esptool` to stuck instead of timeout (only on windows).

chegewara, 2021-03-06:
Ive been playing another full day with it with no result. Changing sleep time (short sleep) between DTR/RTS signals makes a difference, but does not solve problem.
The only time i can successfully flash using native USB is when console_usb example is pre-flashed. The thing is that console_usb example is not using tinyusb, but instead is using USB descriptors from rom, so after reset to bootloader both descriptors from point of view of OS are 1:1 exactly the same.

I also did some tests on arduino, with my library, and i can have CDC mixed with HID and arduino IDE is still able to flash device on linux, as long as CDC endpoints match rom endpoints.

ozgurbostan, 2021-03-07:
Hi @chegewara,

Is it possible to use USB Console feature in Arduino IDE?

chegewara, 2021-03-07:
Only if you write your own library.
Maybe it is possible to use full console_usb component/example, but i didnt try to do it. I have very important project to finish and not much time left to work on USB library.
The closest option to use arduino IDE to flash with USB on windows is when i fool system, switch to DFU, but this requires to use upload button 2 times and com port is changed in between.

ozgurbostan, 2021-03-07:
Thanks for your support until so far. If you recommend me anything to start to write my own library based on USB Console feature, I may try to do so. If possible, you may show me how and where to start?

dobairoland, 2021-03-08:
> The only time i can successfully flash using native USB is when console_usb example is pre-flashed. The thing is that console_usb example is not using tinyusb, but instead is using USB descriptors [...]

Thank you @chegewara for additional information. Yes, I was using that example when I said that I cannot reproduce. I just assumed that it is the only supported way for now. We are stuck with the tinyusb support a little bit because don't have any active maintainer at the moment. We have other tinyusb tasks blocking this and I cannot estimate when will be able to move forward. I'm sorry for this. I can suggest to open a feature request in ESP-IDF in order to get notified about this in the future.

I will close this issue in a couple of days if I don't receive additional information from the original reporter. This is a different issue even thought the manifestation is the same.

chegewara, 2021-03-08:
Yes, i didnt want to hijack issue, but i didnt want to open new one in hope it may be related.
Thank you for all support and if i can help with tinyusb just let me know

ozgurbostan, 2021-03-08:
> Thank you @chegewara for additional information. Yes, I was using that example when I said that I cannot reproduce. I just assumed that it is the only supported way for now.  
> 

Hi @dobairoland,

Is there an easy way to use USB Console feature in Arduino IDE; if possible, could you please give me some recommendations from where and how to start?

dobairoland, 2021-03-09:
Thank you @chegewara. I left you a PM at esp32.com about this topic.

dobairoland, 2021-03-09:
@ozgurbostan I'm sorry I don't use Arduino IDE and this is offtopic here. Please take a look at esp32.com and try to find your answer there.

closed 2021-03-09 by dobairoland

chegewara, 2021-03-09:
Hi @dobairoland 
i cant PM to you on forum