espressif/esptool #608: esptool.py v2.6.1: A fatal error occurred: Timed out waiting for packet content

parinacota, 2021-03-04:
Hello,

I'm encountering this issue with 3 brand new ESP32-VROOM32D chips, and got the same result when trying to flah a (basic) stetch with Arduino & VSCode.

# Operating system
Windows 10 (2 distinct workstations)
Mac OS Big Sur

# Python version
Python 3.9.2

# What Chip
ESP32-WROOM32D (3 distincts new chips)

# What development board or other hardware is the chip attached to
- Downloader for ESP-wroom-32 (with 2104 driver)
- and FTDI USB to TTL - with RX, TX, GND cabling

# Is anything else attached to the development board, except for the serial flasher connections?
NO connection (unless test cables for GPIO0 and 2 to enable flah mode)

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
Arduino IDE, VSCode + Platform IO

# Full esptool.py command line that was run:
# Full output from esptool.py
[...Arduino messages...]
"C:\\Users\\bertr\\OneDrive\\Documents\\ArduinoData\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-size" -A "C:\\Users\\bertr\\AppData\\Local\\Temp\\arduino_build_308714/motorController.ino.elf"
Le croquis utilise 282169 octets (21%) de l'espace de stockage de programmes. Le maximum est de 1310720 octets.
Les variables globales utilisent 17668 octets (5%) de mémoire dynamique, ce qui laisse 310012 octets pour les variables locales. Le maximum est de 327680 octets.
C:\Users\bertr\OneDrive\Documents\ArduinoData\packages\esp32\tools\esptool_py\2.6.1/esptool.exe --chip esp32 --port COM15 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 C:\Users\bertr\OneDrive\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4/tools/partitions/boot_app0.bin 0x1000 C:\Users\bertr\OneDrive\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4/tools/sdk/bin/bootloader_dio_80m.bin 0x10000 C:\Users\bertr\AppData\Local\Temp\arduino_build_308714/motorController.ino.bin 0x8000 C:\Users\bertr\AppData\Local\Temp\arduino_build_308714/motorController.ino.partitions.bin 
esptool.py v2.6
Serial port COM15
Connecting........
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: e8:db:84:2b:f0:d4
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 8192 bytes to 47...

**A fatal error occurred: Timed out waiting for packet content
A fatal error occurred: Timed out waiting for packet content**

# Do you have any other information from investigating this?
Result of espfuse.py:
C:\Users\bertr>espefuse.py --port COM15 summary
Connecting........_____....._____....._____....._____....._____....
Detecting chip type... ESP32
espefuse.py v3.0
EFUSE_NAME (Block)                       Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
BLK3_PART_RESERVE (BLOCK0):              BLOCK3 partially served for ADC calibration data   = False R/W (0b0)
ADC_VREF (BLOCK0):                       Voltage reference calibration                      = 1142 R/W (0b00110)

Config fuses:
XPD_SDIO_FORCE (BLOCK0):                 Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = False R/W (0b0)
XPD_SDIO_REG (BLOCK0):                   If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = False R/W (0b0)
XPD_SDIO_TIEH (BLOCK0):                  If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 1.8V R/W (0b0)
CLK8M_FREQ (BLOCK0):                     8MHz clock freq override                           = 51 R/W (0x33)
SPI_PAD_CONFIG_CLK (BLOCK0):             Override SD_CLK pad (GPIO6/SPICLK)                 = 0 R/W (0b00000)
SPI_PAD_CONFIG_Q (BLOCK0):               Override SD_DATA_0 pad (GPIO7/SPIQ)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_D (BLOCK0):               Override SD_DATA_1 pad (GPIO8/SPID)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_HD (BLOCK0):              Override SD_DATA_2 pad (GPIO9/SPIHD)               = 0 R/W (0b00000)
SPI_PAD_CONFIG_CS0 (BLOCK0):             Override SD_CMD pad (GPIO11/SPICS0)                = 0 R/W (0b00000)
DISABLE_SDIO_HOST (BLOCK0):              Disable SDIO host                                  = False R/W (0b0)

Efuse fuses:
WR_DIS (BLOCK0):                         Efuse write disable mask                           = 0 R/W (0x0000)
RD_DIS (BLOCK0):                         Efuse read disable mask                            = 0 R/W (0x0)
CODING_SCHEME (BLOCK0):                  Efuse variable block length scheme
   = NONE (BLK1-3 len=256 bits) R/W (0b00)
KEY_STATUS (BLOCK0):                     Usage of efuse block 3 (reserved)                  = False R/W (0b0)

Identity fuses:
MAC (BLOCK0):                            Factory MAC Address
   = e8:db:84:2b:f0:d4 (CRC 0x8b OK) R/W
MAC_CRC (BLOCK0):                        CRC8 for factory MAC address                       = 139 R/W (0x8b)
CHIP_VER_REV1 (BLOCK0):                  Silicon Revision 1                                 = True R/W (0b1)
CHIP_VER_REV2 (BLOCK0):                  Silicon Revision 2                                 = False R/W (0b0)
CHIP_VERSION (BLOCK0):                   Reserved for future chip versions                  = 2 R/W (0b10)
CHIP_PACKAGE (BLOCK0):                   Chip package identifier                            = 1 R/W (0b001)
MAC_VERSION (BLOCK3):                    Version of the MAC field                           = 0 R/W (0x00)

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                Flash encryption mode counter                      = 0 R/W (0b0000000)
UART_DOWNLOAD_DIS (BLOCK0):              Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):             Flash encryption config (key tweak bits)           = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE (BLOCK0):          Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                     Secure boot V1 is enabled for bootloader image     = False R/W (0b0)
ABS_DONE_1 (BLOCK0):                     Secure boot V2 is enabled for bootloader image     = False R/W (0b0)
JTAG_DISABLE (BLOCK0):                   Disable JTAG                                       = False R/W (0b0)
DISABLE_DL_ENCRYPT (BLOCK0):             Disable flash encryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_DECRYPT (BLOCK0):             Disable flash decryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_CACHE (BLOCK0):               Disable flash cache in UART bootloader             = False R/W (0b0)
BLOCK1 (BLOCK1):                         Flash encryption key
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK2 (BLOCK2):                         Secure boot key
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK3 (BLOCK3):                         Variable Block 3
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).

C:\Users\bertr>espefuse.py --port COM15 dump
Connecting........_____....._____.
Detecting chip type... ESP32
BLOCK0          (                ) [0 ] read_regs: 00000000 842bf0d4 008be8db 0000a200 00000633 00000000 00000004
BLOCK1          (flash_encryption) [1 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK2          (secure_boot_v1 s) [2 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK3          (                ) [3 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
espefuse.py v3.0

Note: 
- test done with sereral baud rates (above: 115200)
- test done with / without GPIO2 grounded (same behavior)
- Serial monitor display correct flash entering mode:
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

Thanks for you ideas...

dobairoland, 2021-03-05:
Hi @parinacota. Please try the following with the latest esptool version from this repository without Arduino and share the logs.
```
esptool.py --trace --no-stub flash_id
esptool.py --trace flash_id
```

Your chip enters into bootloader mode. It can even upload the flash to RAM. Communication with flash fails.

parinacota, 2021-03-05:
Hello. Thanks for your answer.

Please find the 2 outputs of the esptool commands, executed on the basis esptool.py (version 3.1-dev)
[esptool--nostub_flahid.txt](https://github.com/espressif/esptool/files/6091744/esptool--nostub_flahid.txt)
[esptool_flahid.txt](https://github.com/espressif/esptool/files/6091746/esptool_flahid.txt)

Regards

dobairoland, 2021-03-05:
The communication seems to be right up until the very end where the flash size and manufacturer id are read out. Esptool receives a  proper response but the ID and size fields are all 1s in it. This indicate some kind of hardware issue. 

Could you please try to disconnect the FTDI chip to see if the issue persist?

Do you have some schematics of your hardware connection?

parinacota, 2021-03-06:
Ok. You were right, again an hardware issue, as warned also by the ticket template. The chip was alone on a small Downloader board... Cannot be more simple. but this board seems to have an issue (short-contact??) with Spi pins... I put some tapes on esp chip "south" connectors (except gnd and io2) and that's it, working to isolate them from the board.
[image]
I apologize, no bug or issue with espressif deliveries, just faulty prototyping hardware.
Best regards

dobairoland, 2021-03-06:
No need to apologize. I'm happy that it is solved now. :-)

closed 2021-03-06 by dobairoland