espressif/esptool #609: [S2] flash over USB

chegewara, 2021-03-11:
Hi @dobairoland 
Recently i reported issue with flashing S2 over USB on windows 10 but there was not enough data to think it is esptool issue. Today i performed another test with `USBlyzer` app on windows and i think it may shade some light on the problem.
As far as i understand logs from that app, esptool is sending a lot od `Set line codding requests` and then `Get line codding requests`. 
I think the problem occurs when S2 is already in `waiting download mode`, but esptool is still sending `Get coding line requests`. It is not a problem when cp210x is in the middle, but S2 cant respond on that and esptool is just resting connection.

I understand it work when rom USB is used, but in that case i the 1st stage bootloader is responding, even when CPU is in download mode.

dobairoland, 2021-03-11:
Thank you! I will debug this and try to come up with a solution. I probably get to this the next week.

chegewara, 2021-03-11:
And example of flashing using arduino IDE, when on S2 was installed `console_usb` example:
[image]

dobairoland, 2021-03-15:
> ... I will debug this and try to come up with a solution. I probably get to this the next week.

The priorities has been changed so I can look at this in the next couple of weeks.

radimkarnis, 2021-04-26:
Hello @chegewara,
I tried to investigate, but couldn't manage to reproduce this issue. Would you mind explaining the steps in detail?
Thank you very much

chegewara, 2021-04-26:
Sorry for not updating issue. Recently arduino has been updated and all seems to be ok, at least when i tested with arduino-esp32 v2.0.0-alpha1 on windows 10 VM.
I didnt test with esp-idf on linux yet, but i am assuming it is ok right now. If i will have problems i will reopen issue.

Thanks

radimkarnis, 2021-04-26:
I appreciate your timely response. Closing this issue.

closed 2021-04-26 by radimkarnis