espressif/esptool #610: Flash over bluetooth (HC-05)

chacn, 2021-03-12:
Setup:
An ESP8266 connected through an HC-05 bluetooth module.

Issue:
[image]

MEM_END_ROM_TIMEOUT = 0.05 is too small for the latency over this Bluetooth module, causing the programming to timeout.
The solution appears to be increasing the timeout to 0.1

radimkarnis, 2021-03-15:
Hello @chacn, 
thanks for opening this issue.
Would you mind giving more information about your setup? Are you using Arduino IDE and laptop Bluetooth to flash your ESP8266 with HC-05? Which sketch have you used to allow Bluetooth OTA update?

dobairoland, 2021-03-17:
@chacn Could you please provide the full command you've run and the full trace log? The reported piece of trace is before or after applying `MEM_END_ROM_TIMEOUT = 0.1`?

The backtrace shows a different failure. It seems to happen while uploading the flasher stub. However, `MEM_END_ROM_TIMEOUT` is used only without flasher stub and a possible timeout is ignored.

radimkarnis, 2021-03-26:
Hello, I am closing this issue due to the fact that no more information was provided. It is not possible to reproduce the situation for testing. Reopen if you feel like further discussion is needed.

closed 2021-03-26 by radimkarnis