espressif/esptool #611: Can't enter boot mode on one specific PC

Muecke1978, 2021-03-12:
During bring up of series production of our own ESP32-WROVER based device we face this strange behaviour:
We use esptool to flash ESP module on our own hardware. We have connected CP2102 USB IC in the common way:
[image]

Entering boot mode and flashing runs like a charm on all of our PCs **except** the production PC which we use in our production line. When calling esptool on this special PC, it is not able to enter boot mode.
When we compare signals of EN and IO0 we observe, that sequence of EN and IO0 is different on different PCs. We installed identical drivers on all PCs and compared electrical connection (GND etc) and find no differences. The production PC is AMD Ryzen 5 based, all other PCs have Intel processors. 
Works well:
[image]
Does not work:
[image]

Any hints on where to investigate further?
Regards
Harald

dobairoland, 2021-03-14:
Hi @Muecke1978. Unfortunately, there is a known issue with Ryzen 5 chipsets on Windows. I don't have a definite solution for this, only a couple of suggestions.

- Do you have a capacitor between EN an GND as shown for example [here](https://dl.espressif.com/dl/schematics/ESP-WROVER-KIT_SCH-3.pdf)?
- There is a report suggesting that using an USB 2.0 hub between the board and computer can help.
- You can try Linux. It will probably work on the same machine.
- Manually pulling down IO0 during flashing can also help.

I personally hope that [this update](https://www.reddit.com/r/Amd/comments/m2wqkf/updated_agesa_coming_for_intermittent_usb/) will fix this issue as well.

dobairoland, 2021-04-24:
I'm closing this for lack of activity. Please notify us if further help is required.

closed 2021-04-24 by dobairoland