espressif/esptool #613: A fatal error occurred: Invalid head of packet

JayZar21, 2021-03-17:
# Operating system
- Raspberry Raspbian (updated)

# Python version
3.7.3

# What Chip
ESP32-WROVER

# What development board or other hardware is the chip attached to
Downloader for ESP32-WROVER (https://www.evelta.com/esp32-wrover-module-adapter-downloader-burning-fixture-board/)

# Is anything else attached to the development board, except for the serial flasher connections?
NA

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
No, console and command line via python script

# Full esptool.py command line that was run:
`python3 -m esptool --trace    
    --chip esp32     
    --port /dev/ttyDUT     
    --baud 460800     
    --before default_reset     
    --after hard_reset write_flash     
    -z --flash_mode dio     
    --flash_freq 40m     
    --flash_size detect         
0x3B000 a.bin         
0x1000 b.bin         
0x40000 g.bin         
0xA000 t.bin         
0x23000 f.bin         
0x26000 c.bin         
0x36000 w.bin  
0xD40000 t1.bin  
0xE80000 t2.bin`

# Full output from esptool.py
I attach the stripped output (partially stripped from binary data):
`
 | -- Now flashing, please wait ... 
==> CALL => python3 -m esptool --trace    
                --chip esp32     
                --port /dev/ttyDUT     
                --baud 460800     
                --before default_reset     
                --after hard_reset write_flash     
                -z --flash_mode dio     
                --flash_freq 40m     
                --flash_size detect         
            0x3B000 a.bin         
            0x1000 b.bin         
            0x40000 g.bin         
            0xA000 t.bin         
            0x23000 f.bin         
            0x26000 c.bin         
            0x36000 w.bin  
            0xD40000 t1.bin  
            0xE80000 t2.bin

esptool.py v3.0
Serial port /dev/ttyDUT
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
[... 730 lines cut ...]
    --port /dev/ttyDUT     
    --baud 460800     
    --before default_reset     
    --after hard_reset write_flash     
    -z --flash_mode dio     
    --flash_freq 40m     
    --flash_size detect         
0x3B000 a.bin         
0x1000 b.bin         
0x40000 g.bin         
0xA000 t.bin         
0x23000 f.bin         
0x26000 c.bin         
0x36000 w.bin  
0xD40000 t1.bin  
0xE80000 t2.bin' returned non-zero exit status 2.`

# Do you have any other information from investigating this?
We use this approach to flash a huge amount of our chips.
The command usually succeeds. 
Sometimes it exits with this kind of error (1-3% of fails). 
Not always in the same programming point (i.e. A fatal error occurred: Invalid head of packet (0x00))

# Is there any other information you can think of which will help us reproduce this problem?
NA

radimkarnis, 2021-03-17:
Hello @JayZar21,

I think this issue is entirely HW and Raspberry USB driver related. The fact that the error only happens in 1%-3% and is not entirely reproducible (does not always happen in the same programming point) implies that there might be a HW problem, rather than a related bug in esptool.

Using higher baudrate `-b 460800` might introduce a possibility of error, if your system is prone to interference, but I understand that using lower baudrates is not an option for you when flashing large amounts of chips.

Do you have an option to test a different esp32 downloader for your purpose?

radimkarnis, 2021-03-26:
I am closing this issue. Please, reopen if you feel like further discussion is needed.

closed 2021-03-26 by radimkarnis