espressif/esptool #614: Failed to connect to ESP8266: Timed out waiting for packet header

abham916, 2021-03-18:
**# Operating system**
Windows 10 64bit

**# Python version**
3.9.2

**# What Chip**
ESP 8266 (ESP-M1)

**# What development board or other hardware is the chip attached to**

WeMos D1 R1

**# Is anything else attached to the development board, except for the serial flasher connections?**
Nothing. Connected to computer with USB

**# Are you running esptool.py from an IDE such as Arduino or Eclipse?**
Arduino ESP8266 IDE version 2.7.4

. . . . .

Hello, I'm using WeMos D1 R1 with ESP-M1 ESP8266 microprocessor (I've found that ESP-M1 is compatible with ESP-12F)
I've tried to upload Blink sketch to board, but error occurred.

```
esptool.py v2.8
Serial port COM5
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "C:\Users\abham\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "C:/Users/abham/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 2890, in main
    esp.connect(args.before)
  File "C:/Users/abham/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
```

radimkarnis, 2021-03-18:
Hello @abham916,
before trying to do some troubleshooting, let me understand this correctly - you have desoldered the ESP-12F module and soldered the ESP-M1 module to the WeMos D1 R1 board?

abham916, 2021-03-18:
Hello @radimkarnis ,
No. I have not desoldered anything. On the microprocessor it's written "ESP8266MOD ESP-M1".
When I looked for ESP-M1 chip on Google, most of searched result was ESP-M1 base on ESP8285, so I wrote ESP-12F as additional information. Sorry if I confused you.

radimkarnis, 2021-03-18:
It's okay, it is true there is a lot of conflicting information about the M1 module on the internet (I only found esp8285 based ones).

It's possible the automatic reset circuit in the WeMos D1 R1 is not working properly.

- If you manually short GPIO0 to GND while running esptool (can do this with a pair of pointy tweezers, or a breadboard jumper wire, for example) then does flashing complete successfully?

- If you run esptool and then quickly press and release the "Reset" button on the D1 Mini while esptool says "Connecting", does flashing complete successfully? May need to try this step more than one.

Also, make sure nothing is connected to the board.

abham916, 2021-03-18:
@radimkarnis I've just tried both method, but it still give same error message.

abham916, 2021-03-18:
I manually short GPIO0 to GND using breadboard
I think flashing complete successfully, but don't know why Blink does not working.

I used Blink code from the example given for Arduino IDE, and I got this message.
```
esptool.py v2.8
Serial port COM5
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 5c:cf:7f:99:e6:1e
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 261472 bytes to 193137...
Wrote 261472 bytes (193137 compressed) at 0x00000000 in 17.0 seconds (effective 122.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

If it's not related to upload problem, then I will close this.
Anyway, thank you @radimkarnis !

closed 2021-03-18 by abham916