espressif/esptool #615: ERROR: Failed building wheel for reedsolo

Franky1, 2021-03-21:
* Operating system:
* Python version:

Microsoft Windows 10 Home x64
Python 3.8.6

# Full esptool.py command line as run:

`pip install --upgrade esptool`

# Full output from esptool.py

```log
Building wheel for reedsolo (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: 'C:\WinPython\WPy64-3860\python-3.8.6.amd64\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\xxxxx\\AppData\\Local\\Temp\\pip-install-7t324qen\\reedsolo_36fa3459e2704df38ef5a96a1dde4c14\\setup.py'"'"'; __file__='"'"'C:\\Users\\xxxxx\\AppData\\Local\\Temp\\pip-install-7t324qen\\reedsolo_36fa3459e2704df38ef5a96a1dde4c14\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\xxxxx\AppData\Local\Temp\pip-wheel-u8lvkz0w'
       cwd: C:\Users\xxxxx\AppData\Local\Temp\pip-install-7t324qen\reedsolo_36fa3459e2704df38ef5a96a1dde4c14\
  Complete output (10 lines):
  Cython is installed, building creedsolo module
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.8
  copying reedsolo.py -> build\lib.win-amd64-3.8
  running build_ext
  building 'creedsolo' extension
  error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
  ----------------------------------------
  ERROR: Failed building wheel for reedsolo
  Running setup.py clean for reedsolo
Failed to build reedsolo
Installing collected packages: reedsolo, ecdsa, bitstring, esptool
    Running setup.py install for reedsolo ... error
    ERROR: Command errored out with exit status 1:
     command: 'C:\WinPython\WPy64-3860\python-3.8.6.amd64\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\xxxxx\\AppData\\Local\\Temp\\pip-install-7t324qen\\reedsolo_36fa3459e2704df38ef5a96a1dde4c14\\setup.py'"'"'; __file__='"'"'C:\\Users\\xxxxx\\AppData\\Local\\Temp\\pip-install-7t324qen\\reedsolo_36fa3459e2704df38ef5a96a1dde4c14\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\xxxxx\AppData\Local\Temp\pip-record-pggpxbrh\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\WinPython\WPy64-3860\python-3.8.6.amd64\Include\reedsolo'
         cwd: C:\Users\xxxxx\AppData\Local\Temp\pip-install-7t324qen\reedsolo_36fa3459e2704df38ef5a96a1dde4c14\
    Complete output (10 lines):
    Cython is installed, building creedsolo module
    running install
    running build
    running build_py
    creating build
    creating build\lib.win-amd64-3.8
    copying reedsolo.py -> build\lib.win-amd64-3.8
    running build_ext
    building 'creedsolo' extension
    error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\WinPython\WPy64-3860\python-3.8.6.amd64\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\xxxxx\\AppData\\Local\\Temp\\pip-install-7t324qen\\reedsolo_36fa3459e2704df38ef5a96a1dde4c14\\setup.py'"'"'; __file__='"'"'C:\\Users\\xxxxx\\AppData\\Local\\Temp\\pip-install-7t324qen\\reedsolo_36fa3459e2704df38ef5a96a1dde4c14\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\xxxxx\AppData\Local\Temp\pip-record-pggpxbrh\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\WinPython\WPy64-3860\python-3.8.6.amd64\Include\reedsolo' Check the logs for full command output.
```

# What is the expected behaviour?

pip upgrade should not fail and should install with pre-build wheels without the need to install additional ms toolchains

# Do you have any other information from investigating this?

No

dobairoland, 2021-03-22:
Hi @Franky1. I see that you've reported this issue for the reedsolo package as well (https://github.com/tomerfiliba/reedsolomon/issues/32). This is very weird and I think something is wrong with your Python. We are using Python 3.8.7 as the default one for ESP-IDF where this package is installed and used by many customers. I have no information beside yours that Visual C++ would be required. I don't have it neither and have it with 3.8.7.

Do you have some special distribution of Python? Could you please try one 3.8.x from python.org? 3.9.x is the newest one but lets keep trying to debug 3.8.x.

Franky1, 2021-03-23:
Still after switching from `WinPython 3.8.6.0` to `WinPython 3.8.8.0` on this machine, i had no success to install `esptool` respectively `reedsolo`.

Only after installing the latest **Microsoft Visual C++ 14.2 Build Tools** on this machine I was able to install `esptool` respectively `reedsolo` without any error messages.

https://wiki.python.org/moin/WindowsCompilers

lia800, 2023-01-16:
I still have problems... What can I do?
[image]