espressif/esptool #617: cannot upload stub - write to RAM error 0107

s0mm3rb, 2021-03-26:
# Operating system
raspbian 10 (buster)

# Python version
Python 2.7.16

# What Chip
TUYA TYWE3S with ESP8266EX and 2MB flash
https://tasmota.github.io/docs/devices/TYWE3S/

it has already the patched tuya firmware that kills the OTA hack
https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02

# What development board or other hardware is the chip attached to
the device is an infrared and 433 MHz programmable remote control
also known as URC-160.app
imported by PEARL Germany
https://www.pearl.de/a-NX4955-3103.shtml
(german user manual, if anyone needs it: https://www.pearl.de/pdocs/NX4955_11_182700.pdf)

# Is anything else attached to the development board, except for the serial flasher connections?
yep, 2 unknown MCUs
one can be disabled by pulling the RST to GND 
but it does not change the ESPs behaviour

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
nope, raspberry cli

# Full esptool.py command line that was run:
first I tried: `esptool.py --port /dev/ttyUSB0 --baud 9600 read_flash 0 0x100000 firmware.bin`
(as suggested at the tuya convert collab doc)
after some research I was able to get the flash info with: `esptool.py --port /dev/ttyUSB0 --baud 9600 --no-stub flash_id`
then I tried: `esptool.py --port /dev/ttyUSB0 --baud 9600 read_flash 0 0x200000 firmware.bin`
and also: `esptool.py --port /dev/ttyUSB0 --baud 9600 --no-stub read_flash 0 0x200000 firmware.bin`

# Full output from esptool.py
```
esptool.py --port /dev/ttyUSB0 --baud 9600 --no-stub flash_id
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: bc:dd:c2:7f:85:5b
Enabling default SPI flash mode...
Manufacturer: c8
Device: 4015
Detected flash size: 2MB
Hard resetting via RTS pin...
```

```
esptool.py --port /dev/ttyUSB0 --baud 9600 read_flash 0 0x200000 firmware.bin
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: bc:dd:c2:...
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 0107)
```

thanks in advance

dobairoland, 2021-03-28:
Hi @s0mm3rb. Please run the following command and post the received log:
```
esptool.py --port /dev/ttyUSB0 --baud 9600 --trace flash_id
```
Thanks!

s0mm3rb, 2021-03-28:
@dobairoland 

here's the log

thanks

```
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.005 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.062 Read 1 bytes: c0
TRACE +0.000 Read 9 bytes: 010802000707122000
TRACE +0.016 Read 1 bytes: 00
TRACE +0.000 Read 15 bytes: c0c001080200070712200000c0c001
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.001 Received full packet: 01080200070712200000
TRACE +0.015 Read 1 bytes: 08
TRACE +0.000 Read 14 bytes: 0200070712200000c0c001080200
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.014 Read 1 bytes: 07
TRACE +0.000 Read 14 bytes: 0712200000c0c001080200070712
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.016 Read 1 bytes: 20
TRACE +0.000 Read 14 bytes: 0000c0c001080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.016 Read 1 bytes: c0
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.001 Read 15 bytes: c001080200070712200000c0c00108
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.014 Read 1 bytes: 02
TRACE +0.000 Read 8 bytes: 00070712200000c0
TRACE +0.000 Received full packet: 01080200070712200000

Detecting chip type...TRACE +0.001 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=00100040
TRACE +0.000 Write 14 bytes: c0000a04000000000000100040c0
TRACE +0.030 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a020001c1f0ff0000c0
TRACE +0.000 Received full packet: 010a020001c1f0ff0000
 ESP8266
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5c00f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005c00f03fc0
TRACE +0.031 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a0200c2ddbc000000c0
TRACE +0.000 Received full packet: 010a0200c2ddbc000000
TRACE +0.001 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5800f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005800f03fc0
TRACE +0.030 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a020000b0005c0000c0
TRACE +0.000 Received full packet: 010a020000b0005c0000
TRACE +0.001 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5400f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005400f03fc0
TRACE +0.030 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a0200857f00020000c0
TRACE +0.000 Received full packet: 010a0200857f00020000
TRACE +0.001 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5000f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005000f03fc0
TRACE +0.031 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a02000000b55b0000c0
TRACE +0.000 Received full packet: 010a02000000b55b0000
Chip is ESP8266EX
TRACE +0.001 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5c00f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005c00f03fc0
TRACE +0.030 Read 1 bytes: c0
[... 809 lines cut ...]
    29a4f64f07c6ddff 7221245d0b2172fa | )..O....r!$].!r.
    2c2339620c068601 007221245d0b7cf6 | ,#9b.....r!$].|.
    2616144b26cc6246 03000b77c2ccf882 | &..K&.bF...w....
    212477b802f68cf1 8168fa2198fa3198 | !$w......h.!..1.
    fac9784d0c526134 6261357261338261 | ..xM.Ra4ba5ra3.a
    32b26136c5840082 2132922128a22126 | 2.a6....!2.!(.!&
    0b2299e8922129e0 e210a26810722133 | ."...!)....h.r!3
    a22124522134b221 36622135f9f8e268 | .!$R!4.!6b!5...h
    14926815a0d7dbdc b0c5dbdcfd069656 | ..h............V
    0e3185faf8d82d0c 057d00f0e0f44d02 | .1....-..}....M.
    f0f0f57d0c0c7862 2135b22136462500 | ...}..xb!5.!6F%.
    0000920200a20202 eae9920201ea999a | ................
    eefafee202039a9a 9aff9a9ee202049a | ................
    ff9a9ee202059aff 9a9ee202069aff9a | ................
    9ee202079aff9aee eaff8b223a924739 | ...........":.G9
    dbdc402341b022b0 b090604602000032 | ..@#A."...`F...2
    02001b223aeeeaff 2a39bd024733ef31 | ...":...*9..G3.1
    67fa2d0e426131c0                  | g.-.Ba1.
TRACE +6.842 Read 1 bytes: c0
TRACE +0.015 Read 1 bytes: 01
TRACE +0.000 Read 10 bytes: 070200c2ddbc000107c0
TRACE +0.000 Received full packet: 01070200c2ddbc000107

A fatal error occurred: Failed to write to target RAM (result was 0107)
```

dobairoland, 2021-03-29:
Hi @s0mm3rb. Everything looks good to me up until that point where the error is returned. I'm linking here https://github.com/espressif/esptool/issues/280 where the same error was reported and the fix was the change of the connection to the chip. I'm sorry but this seems to me a hardware issue I don't know what could be done on the esptool side.

s0mm3rb, 2021-03-29:
@dobairoland thanks for looking into it
I've already found that thread and tried another FTDI adapter
but it's the same with all three of them
(this https://www.amazon.com/dp/B00IJXZQ7C and this https://www.amazon.com/dp/B07PDGKDBY)

also I have a bunch of different ESPs and sonoff here and they work fine
that is the only tuya device I've tried so far

maybe one of the other connected MCUs are interfering with the signal
I can disable one, maybe I can also disable the other but I haven't found any documentation on them
here's some pictures of them, maybe some captain could help identifying them
https://imgur.com/a/tspwqrg

thanks

s0mm3rb, 2021-03-29:
I don't have an oscilloscope available atm
but I changed my setup and powered it through the onboard USB and disabled both of the other MCUs
and the trace changed a bit:

```
esptool.py --port /dev/ttyUSB0 --baud 9600 --trace flash_id
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.003 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.064 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 01080200070712200000c0
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.016 Read 1 bytes: c0
TRACE +0.000 Read 15 bytes: 01080200070712200000c0c0010802
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.015 Read 1 bytes: 00
TRACE +0.000 Read 13 bytes: 070712200000c0c00108020007
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.016 Read 1 bytes: 07
TRACE +0.000 Read 15 bytes: 12200000c0c0010802000707122000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.016 Read 1 bytes: 00
TRACE +0.000 Read 15 bytes: c0c001080200070712200000c0c001
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.001 Received full packet: 01080200070712200000
TRACE +0.014 Read 1 bytes: 08
TRACE +0.000 Read 13 bytes: 0200070712200000c0c0010802
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.016 Read 1 bytes: 00
TRACE +0.000 Read 7 bytes: 070712200000c0
TRACE +0.000 Received full packet: 01080200070712200000

Detecting chip type...TRACE +0.001 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=00100040
TRACE +0.000 Write 14 bytes: c0000a04000000000000100040c0
TRACE +0.030 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a020001c1f0ff0000c0
TRACE +0.000 Received full packet: 010a020001c1f0ff0000
 ESP8266
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5c00f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005c00f03fc0
TRACE +0.031 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a0200c2ddbc000000c0
TRACE +0.000 Received full packet: 010a0200c2ddbc000000
TRACE +0.001 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5800f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005800f03fc0
TRACE +0.030 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a020000b0005c0000c0
TRACE +0.000 Received full packet: 010a020000b0005c0000
TRACE +0.001 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5400f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005400f03fc0
TRACE +0.030 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a0200857f00020000c0
TRACE +0.000 Received full packet: 010a0200857f00020000
TRACE +0.001 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5000f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005000f03fc0
TRACE +0.030 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a02000000b55b0000c0
TRACE +0.000 Received full packet: 010a02000000b55b0000
Chip is ESP8266EX
TRACE +0.001 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5c00f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005c00f03fc0
TRACE +0.031 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a0200c2ddbc000000c0
[... 809 lines cut ...]
    2c2339620c068601 007221245d0b7cf6 | ,#9b.....r!$].|.
    2616144b26cc6246 03000b77c2ccf882 | &..K&.bF...w....
    212477b802f68cf1 8168fa2198fa3198 | !$w......h.!..1.
    fac9784d0c526134 6261357261338261 | ..xM.Ra4ba5ra3.a
    32b26136c5840082 2132922128a22126 | 2.a6....!2.!(.!&
    0b2299e8922129e0 e210a26810722133 | ."...!)....h.r!3
    a22124522134b221 36622135f9f8e268 | .!$R!4.!6b!5...h
    14926815a0d7dbdc b0c5dbdcfd069656 | ..h............V
    0e3185faf8d82d0c 057d00f0e0f44d02 | .1....-..}....M.
    f0f0f57d0c0c7862 2135b22136462500 | ...}..xb!5.!6F%.
    0000920200a20202 eae9920201ea999a | ................
    eefafee202039a9a 9aff9a9ee202049a | ................
    ff9a9ee202059aff 9a9ee202069aff9a | ................
    9ee202079aff9aee eaff8b223a924739 | ...........":.G9
    dbdc402341b022b0 b090604602000032 | ..@#A."...`F...2
    02001b223aeeeaff 2a39bd024733ef31 | ...":...*9..G3.1
    67fa2d0e426131c0                  | g.-.Ba1.
TRACE +6.836 Read 1 bytes: c0
TRACE +0.000 Read 7 bytes: 01070200c2ddbc
TRACE +0.016 Read 1 bytes: 00
TRACE +0.000 Read 3 bytes: 0107c0
TRACE +0.000 Received full packet: 01070200c2ddbc000107

A fatal error occurred: Failed to write to target RAM (result was 0107)
```

s0mm3rb, 2021-03-29:
holding the reset button while monitoring I was able to grab this

```
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2592, room 16
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4
tail 0
chksum 0x22
csum 0x22

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : QIO
SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000
```

gibo77, 2021-04-04:
I have the same issue after I did an OTA several times that got stuck halfway like 56%.

I have read a thread before of downloading to make the chip blank slate. I will keep it posted here if I find it.

s0mm3rb, 2021-04-07:
FYI: closing this, because it is most likely a HW issue and not esptool

I tried to dump the FW with the onboard serial connection on the raspi, but it gave me the same 0107 error

I have just flashed a sonoff RF bridge with the same raspi and I guess one of the additional MCUs on the tuya board interfere with the flashing process

I found this post, where they were successfull after cutting the RX trace to the additional MCU
https://forum.iobroker.net/topic/9886/tuya-jinvoo-unterputz-wandschalter/16?_=1617829763723

I might try this next and will post the result here if any

closed 2021-04-07 by s0mm3rb