espressif/esptool #618: ESP8266 via Arduino - Timed out waiting for packet header after sync

matgiani, 2021-03-28:
----------------------------- Delete below -----------------------------

Hello,

I am attempting to flash an ESP8266EX (installed on a custom board, a RGBW light controller).
Connection to the PC is via the integrated USB-Serial interface of an Arduino Mega 2560.

The Arduino itself is bypassed (Reset tied to Ground).

Connections are otherwise as follows: (TX to TX and RX to RX due to bypassing the Arduino)
Arduino RX0 -> (via voltage divider) -> ESP RX
Arduino TX0 -> ESP TX (no voltage shift performed, but receiving works, see below)
ESP powered via external 5V supply (it has a voltage regulator onboard)
GPIO0 tied to ground momentarily upon reset to enter bootloader

**What works:**

- if I open a terminal at 74880 baud, I can see the ESP boot messages in flash boot (default) mode, running the original firmware. Device connects to network and functions normally in this case and when powered by the same power supply I am using to attempt flashing
- Similarly if I open a terminal at 74880 baud after resetting with GPIO0 to ground, I see the following `ets Jan  8 2013,rst cause:2, boot mode:(1,7)`. Boot mode 1 indicates to me that it is now in the correct mode to talk to esptool.

**The issue:**
Esptool seems to be able to sync with the device, but anything after that results in a "timed out waiting for packet header".
Sample trace below for chip_id.
Any other commands (read_mac, flash_id etc) behave the same way.

----------------------------- Delete above -----------------------------

# Operating system
Ubuntu 20.10

# Python version
Python 3.8.6

# What Chip
ESP8266

# What development board or other hardware is the chip attached to

ESP8266 on this device, see [this page](https://templates.blakadder.com/iwoole_table_lamp.html)

# Is anything else attached to the development board, except for the serial flasher connections?

GPIO4, 12, 13 and 14 are connected to the LED drivers

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
No IDE. Ubuntu command prompt.

# Full esptool.py command line that was run:

esptool.py --port /dev/serial/by-id/usb-Arduino__www.arduino.cc__0042_5593034363635170A141-if00 --before no_reset --trace chip_id

# Full output from esptool.py

`matteo@Ubuntu-skylake-itx:~$ esptool.py --port /dev/serial/by-id/usb-Arduino__www.arduino.cc__0042_5593034363635170A141-if00 --before no_reset --trace chip_id
esptool.py v3.0
Serial port /dev/serial/by-id/usb-Arduino__www.arduino.cc__0042_5593034363635170A141-if00
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 Timed out waiting for packet header
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
[... 113 lines cut ...]
TRACE +0.000 Read 47 bytes: 
    00c0c00108020007 0712200000c0c001 | .......... .....
    0802000707122000 00c0c00108020007 | ...... .........
    0712200000c0c001 08020007071220   | .. ........... 
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.004 Read 1 bytes: 00
TRACE +0.000 Read 2 bytes: 00c0
TRACE +0.000 Received full packet: 01080200070712200000

Detecting chip type...TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=00100040
TRACE +0.000 Write 14 bytes: c0000a04000000000000100040c0
TRACE +3.003 Timed out waiting for packet header

A fatal error occurred: Timed out waiting for packet header
matteo@Ubuntu-skylake-itx:~$ `

# Do you have any other information from investigating this?

See additional information in text above - serial connection seems to work fine in terminal.

# Is there any other information you can think of which will help us reproduce this problem?
Not at this stage.

dobairoland, 2021-03-28:
Hi @matgiani. Why do you need `--before no_reset`? Could you please try without it?

Could you try to increase [DEFAULT_TIMEOUT](https://github.com/espressif/esptool/blob/master/esptool.py#L76) in the code and see if it helps?

matgiani, 2021-03-28:
Hey @dobairoland and thanks for looking into this.

> Why do you need `--before no_reset`? Could you please try without it?
In my case, the only lines connected that esptool can control are the data lines, hence I wanted to avoid it trying to reset the ESP and potentially upsetting the Arduino USB to serial converter.
Essentially I need to reset the ESP myself by powering it up as esptool is trying to sync.
In any case I've removed `--before no_reset` but same result.

> Could you try to increase [DEFAULT_TIMEOUT](https://github.com/espressif/esptool/blob/master/esptool.py#L76) in the code and see if it helps?
Sure thing, I've increased it to 10 seconds but the outcome does not change - it waits for 10 seconds and then times out.

matgiani, 2021-03-28:
I noticed one more thing: it seems like I can get a response to *one* packet at a time - see for example the following.

This is a trace just after reset, in line with the behaviour above: it syncs but does not go further than that.

```
esptool.py --port /dev/serial/by-id/usb-Arduino__www.arduino.cc__0042_5593034363635170A141-if00 --trace chip_id
esptool.py v3.0
Serial port /dev/serial/by-id/usb-Arduino__www.arduino.cc__0042_5593034363635170A141-if00
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 Timed out waiting for packet header
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.008 Read 1 bytes: c0
TRACE +0.000 Read 35 bytes: 
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 0108020007071220 | ... ........... 
    0000c0                            | ...
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.004 Read 1 bytes: c0
TRACE +0.000 Read 46 bytes: 
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 0108020007071220 | ... ........... 
    0000c0c001080200 070712200000     | ........... ..
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.004 Read 1 bytes: c0
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Read 12 bytes: c001080200070712200000c0
TRACE +0.000 Received full packet: 01080200070712200000

Detecting chip type...TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=10.000 data=00100040
TRACE +0.000 Write 14 bytes: c0000a04000000000000100040c0
TRACE +10.010 Timed out waiting for packet header

A fatal error occurred: Timed out waiting for packet header
```

I then wanted to see what happened if i told esptool to skip the sync, without resetting or resyncing, and retry from where it left basically:

```
esptool.py --port /dev/serial/by-id/usb-Arduino__www.arduino.cc__0042_5593034363635170A141-if00 --before no_reset_no_sync --trace chip_id
esptool.py v3.0
Serial port /dev/serial/by-id/usb-Arduino__www.arduino.cc__0042_5593034363635170A141-if00
Connecting...
Detecting chip type...TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=10.000 data=00100040
TRACE +0.000 Write 14 bytes: c0000a04000000000000100040c0
TRACE +0.002 Read 1 bytes: c0
TRACE +0.000 Read 7 bytes: 010a020001c1f0
TRACE +0.004 Read 1 bytes: ff
TRACE +0.000 Read 3 bytes: 0000c0
TRACE +0.000 Received full packet: 010a020001c1f0ff0000
 ESP8266
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=10.000 data=5c00f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005c00f03fc0
TRACE +10.010 Timed out waiting for packet header

A fatal error occurred: Timed out waiting for packet header
```

To my surprise this time it identifies the chip type, but then times out waiting for a response to the second command.

matgiani, 2021-03-28:
Final update from me: interestingly, when I tried using esptool.py in Windows 10 (same machine and same hardware serial setup as above) it went further, i.e. correctly reading MAC, chip ID and flash info, etc. However it was still struggling with more substantive data transfers (flash_read was failing with "expected 0x1000 bytes, received 0xffd" and similar).

Hence I chalked it up to a bad serial connection and decided to flash the ESP using the UART of a Raspberry Pi instead. This worked perfectly.

closed 2021-03-28 by matgiani

dobairoland, 2021-03-28:
Great! I'm happy that you found a workaround for your issue!