espressif/esptool #619: LILYGO TTGO T-Watch-2020 read_flash hangs at 72%

theycallmevirgo, 2021-03-28:
Platform: Win10 enterprise v1909 

Python: WinPython v3.9.2 x64

Command; 
`python E:\WinPython\python-3.9.2.amd64\Lib\site-packages\esptool-3.1.dev0-py3.9.egg\esptool.py --trace -b 1000000 --port COM2 read_flash 0x00000 0x1600000 E:\watch\stock_16m.bin`

Final packet; 

`TRACE +0.001 Read 511 bytes:
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffff   | ...............
TRACE +0.004 Read 1 bytes: ff
TRACE +0.001 Read 513 bytes:
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
[... 242 lines cut ...]
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
TRACE +0.014 Write 6 bytes: c000000001c0`

NOTE: not yet tested on Linux. Available distro is Rasperry Pi OS (Debian) but I can try to virtualize using hyper-v, wsl, and/or vmware if recommended.

dobairoland, 2021-03-28:
Hi @theycallmevirgo. I think you've got wrong `0x1600000` because 22MB flash is not supported by ESP32 used in that watch.

theycallmevirgo, 2021-03-28:
@dobairoland Are you saying that isn't a decimal value? I think the size is 16mb, what is the correct value?

Thanks again

Joe

dobairoland, 2021-03-28:
@theycallmevirgo Numbers starting with `0x` are hexadecimal. `hex(16*1024*1024) =  0x1000000`.

theycallmevirgo, 2021-03-28:
@dobairoland thanks so much for this, I should know this. I feel pretty dumb.

dobairoland, 2021-04-24:
I think this can be closed for lack of activity in the last month.

closed 2021-04-24 by dobairoland