espressif/esptool #621: espsecure - encrypt_flash_data gives Unknown OpenSSL Error

ThomasCham, 2021-04-14:
* Operating system: Windows 10
* Python version: 3.7.5

# Full esptool.py command line as run:
"C:\Users\ThomasB\Documents\WorkingDirectory\esptool\espsecure.py" encrypt_flash_data --aes_xts -k "C:\Users\ThomasB\Desktop\my_flash_encryption_key.bin" -a 0x0 -o "C:\Users\ThomasB\Documents\WorkingDirectory\encrypted-image.bin" "C:\Users\ThomasB\Desktop\ESP Main Image\esp32s2_combined.bin"

# Full output from esptool.py
espsecure.py v3.1-dev
Using 256-bit key
Traceback (most recent call last):
  File "C:\Users\ThomasB\Documents\WorkingDirectory\esptool\espsecure.py", line 997, in <module>
    _main()
  File "C:\Users\ThomasB\Documents\WorkingDirectory\esptool\espsecure.py", line 990, in _main
    main()
  File "C:\Users\ThomasB\Documents\WorkingDirectory\esptool\espsecure.py", line 980, in main
    operation_func(args)
  File "C:\Users\ThomasB\Documents\WorkingDirectory\esptool\espsecure.py", line 814, in encrypt_flash_data
    return _flash_encryption_operation_aes_xts(args.output, args.plaintext_file, args.address, args.keyfile, False)
  File "C:\Users\ThomasB\Documents\WorkingDirectory\esptool\espsecure.py", line 774, in _flash_encryption_operation_aes_xts
    encryptor = cipher.decryptor() if do_decrypt else cipher.encryptor()
  File "C:\Users\ThomasB\AppData\Local\Programs\Python\Python37\lib\site-packages\cryptography\hazmat\primitives\ciphers\base.py", line 110, in encryptor
    self.algorithm, self.mode
  File "C:\Users\ThomasB\AppData\Local\Programs\Python\Python37\lib\site-packages\cryptography\hazmat\backends\openssl\backend.py", line 484, in create_symmetric_encryption_ctx
    return _CipherContext(self, cipher, mode, _CipherContext._ENCRYPT)
  File "C:\Users\ThomasB\AppData\Local\Programs\Python\Python37\lib\site-packages\cryptography\hazmat\backends\openssl\ciphers.py", line 115, in __init__
    self._backend.openssl_assert(res != 0)
  File "C:\Users\ThomasB\AppData\Local\Programs\Python\Python37\lib\site-packages\cryptography\hazmat\backends\openssl\backend.py", line 241, in openssl_assert
    return binding._openssl_assert(self._lib, ok, errors=errors)
  File "C:\Users\ThomasB\AppData\Local\Programs\Python\Python37\lib\site-packages\cryptography\hazmat\bindings\openssl\binding.py", line 88, in _openssl_assert
    errors_with_text,
cryptography.exceptions.InternalError: Unknown OpenSSL error. This error is commonly encountered when another library is not cleaning up the OpenSSL error stack. If you are using cryptography with another library that uses OpenSSL try disabling it before reporting a bug. Otherwise please file an issue at https://github.com/pyca/cryptography/issues with information on how to reproduce this. ([_OpenSSLErrorWithText(code=101511351, lib=6, func=207, reason=183, reason_text=b'error:060CF0B7:digital envelope routines:aesni_xts_init_key:xts duplicated keys')])

# What is the expected behaviour?
It should generate an encrypted image using the given aes-xts keyfile and the image file.

# Do you have any other information from investigating this?
My installed Python modules are:
Package            Version
appdirs            1.4.3
bitstring          3.1.7
cffi               1.14.0
cryptography       3.4.7
distlib            0.3.0
filelock           3.0.12
future             0.18.2
importlib-metadata 1.6.0
invoke             1.4.1
pip                20.0.2
pycparser          2.20
pyserial           3.4
reedsolo           1.5.4
setuptools         41.2.0
six                1.14.0
virtualenv         20.0.17
zipp               3.1.0

# Is there any other information you can think of which will help us reproduce this problem?
Tried on 2 PCs with similar packages to the listed ones above and both give the same Unknown OpenSSL Error. Not sure if I'm missing a package or something similar but can't find any information on prerequiste packages needed?

dobairoland, 2021-04-19:
Hi @ThomasCham. I haven't seen this issue before. Could you please try to run the related unit tests on your computer?

I can run the following on Windows and Linux as well without any issue:
```.\test\test_espsecure.py AesXtsFlashEncryptionTests```

Are you following the [flash encryption guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/security/flash-encryption.html) of ESP-IDF? It would be worth going through the steps (without actually flashing encrypted images) to see if at any stage you get the same issue. It would help us to reproduce it.

ThomasCham, 2021-04-21:
Hi @dobairoland,

Sorry for the late reply. I ran the command you gave above and it ran the 3 tests and came back OK.
I am following the flash encryption guide you have linked. I've gotten to the stage where I need to encrypt the image with the generated flash encryption key from the previous step and this is the error I get.

dobairoland, 2021-04-21:
I cannot reproduce the issue. I have tried it on Linux and Windows as well. It works for me.

I've found a very similar issue: https://github.com/pyca/cryptography/issues/4885 I'm not sure if it is relevant.

Could you check the content of your `my_flash_encryption_key.bin`? It contains 32 bytes?

Could you please try to install an older cryptography package? Could you try [this one]( https://dl.espressif.com/pypi/cryptography/cryptography-3.4.6-cp36-abi3-win_amd64.whl) (this is the one used by most of ESP-IDF users)? I'm also using binaries from [there](https://dl.espressif.com/pypi/cryptography/).

ThomasCham, 2021-04-21:
I've already tried the linked issue you mentioned when I googled the error :)

When I generate a flash encryption key, I do get a 32 byte file as you say.

I've also tried downgrading to cryptography 3.3.1 as that is the version used by the PC which does have it working.

I am using the "fairly" latest master clone of esptool so that I have esptool version 3.1-dev. Is there a stable release of some kind which may give better results? I think that version 3 doesn't support the aes-xts encryption that we require?

dobairoland, 2021-04-21:
> I've also tried downgrading to cryptography 3.3.1 as that is the version used by the PC which does have it working.

Does this mean that you get failure with 3.3.2 and success with 3.3.1 when trying repeatedly?

> I am using the "fairly" latest master clone of esptool so that I have esptool version 3.1-dev. Is there a stable release of some kind which may give better results? I think that version 3 doesn't support the aes-xts encryption that we require?

This issue seems to me entirely cryptography + openssl one. Esptool have a minimal version required (`cryptography>=2.1.4`), so the newest cryptography will be used even if you switch to older version of esptool. You can downgrade cryptography for yourself if that solves your issue.

ThomasCham, 2021-04-21:
> Does this mean that you get failure with 3.3.2 and success with 3.3.1 when trying repeatedly?
I've tried using the flash encryption methods on a total of 5 PCs. On 1 PC, I've managed to get it to work straight away. The other 4 gave this error. I did manage to fix it on 1 PC by using 3.3.1 cryptography and uninstalling over packages I think (to be honest, I tried lots of random installing/uninstalling and eventually it worked). 

I've since tried using `pip freeze` to and install the exact versions of packages from a working PC to other non-working ones but am still hit with this issue so it appears to be random?

> You can downgrade cryptography for yourself if that solves your issue.
I've tried various versions of cryptography being: latest, 3.3.1, 3.3.0 and none seem to make it work.

dobairoland, 2021-04-22:
@ThomasCham I'm really sorry for the inconvenience.

The fact that you have this issue on 4 PCs really surprises me. We are using this package a lot in [ESP-IDF](
https://github.com/espressif/esp-idf/blob/master/requirements.txt#L11) as well but I don't remember reports about this.

What other software are using on those PCs which might interfere? Are you using only ESP-IDF & Esptool, or some related tools as well, i.e. Arduino, Micropython, Platformio, ...?

dobairoland, 2021-04-22:
Can you please share your key? I assume you won't use that in any final product of yours. Any key is sufficient which you've generated with esptool and produces the above issue.

ThomasCham, 2021-04-22:
[test_flash_encryption_key.zip](https://github.com/espressif/esptool/files/6356628/test_flash_encryption_key.zip)

I've attached a key generated using espsecure on an affected PC. In terms of other tools like you mentioned, we don't use any of the ones you specified. It's difficult to say whether its a common tool that is on all 4 as they are all used for different purposes (software enginerering, manufacturing and general use). I'll try and investigate whether there is a common tool between the 4 PCs in the meantime, but it might be quite difficult.

Thanks for bearing with me on this one :)

dobairoland, 2021-04-22:
I'm sorry but I'm out of ideas. I'll ask around if someone could help with your issue.

ThomasCham, 2021-04-22:
Apologies for the trouble, but it appears it was a problem my end. The keyfile I manually hex edited to test with was the problem.

[manual_flash_encryption_key.zip](https://github.com/espressif/esptool/files/6358674/manual_flash_encryption_key.zip)

On another question then, as this is basically 16 bytes repeated to make the 32 byte key, why would this cause an error? Is there a possibility that espsecure would generate an invalid key like this?

Thanks and sorry for the confusion.

projectgus, 2021-04-23:
Hi @ThomasCham,

Glad you got to the bottom of this!

> On another question then, as this is basically 16 bytes repeated to make the 32 byte key, why would this cause an error?

Yes, openssl added this check to prevent encrypting with a known-vulnerable key:
https://github.com/openssl/openssl/blob/c39352/crypto/evp/e_aes.c#L3119

The error message seem unintuitive, though. We could add a check for this case and more user-friendly error message in esptool, but I think the most appropriate place to fix it is in the cryptography package, so have opened an issue there to ask about it.

projectgus, 2021-06-09:
As soon as their next release comes out, cryptography package will now generate a more meaningful error for this case: https://github.com/pyca/cryptography/pull/6085

Thanks for letting us know about this!

closed 2021-06-09 by projectgus