espressif/esptool #622: ESP8266 CrystalFreq Select

means0nothing, 2021-04-23:
ESP8266 DownloadTool allows you to change CrystalFreq to 24M, 26M and 40M (according to ESP8266's datasheet, all of them are supported)
Is it possible to flashing ESP8266 with 24M Crystal using esptool?

radimkarnis, 2021-04-27:
Hello @means0nothing,
esptool supports crystal frequency autodetection only for 40MHz and 26MHz. It should operate fine even with 24MHz crystals, though. There is no need to manually edit any config files. 
Are you having trouble with a particular situation or is this just generally speaking?

means0nothing, 2021-04-29:
I'm trying to download AT Firmware. Esptool calculate crystal frequency as 24MHz and warns about big error regarding 26MHz. After downloading esp8266 reacts to AT commands with baud 115200*24/26.

If I change the corresponding byte in esp_init_data from default value "1" (0: 40MHz, 1: 26MHz, 2: 24MHz) to "2", esp8266 reacts to AT commands with baud 115200 properly.

ESP8266 DownloadTool aloows to select CrystalFreq 24MHz, and download AT Firmware with esp_init_data_default, and after that everything works fine with baud 115200.

dobairoland, 2021-04-30:
Hi @means0nothing. Esptool doesn't do modification of bin files on-the-fly in order to change the crystal frequency. I suggest you to change the frequency in the build tool (or use supported crystals). That detected crystal frequency doesn't change anything for flashing process, it is just a warning for you that you use an unsupported crystal frequency.

I'm not an ESP8266 developer but I looked now at the datasheet and the TRM. Both comment the steps  distinguishing between 26M and 40M just the same way Esptool does. It states that the lower range is 24M but it is interesting to take a look at the history of the document which states that the lower range was 26M and got modified to 24M in 2016. Looking at the official ESP8266 SDK, I see only 26M and 40M (https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/components/esp8266/Kconfig#L14). I guess 24M never got properly supported.

But no matter what is the lower supported frequency limit, please don't expect esptool to modify the binaries before flashing. That is the responsibility of the build system which generates the binaries. I don't know the DownloadTool you are referencing, but if it does this kind of hacking then I don't like it.

closed 2021-05-09 by radimkarnis