espressif/esptool #624: Impossible to consult the esptool's version

kristofmulier, 2021-04-24:
I tried to consult the version of the `esptool.py` script:

```sh
$ python esptool.py -v
esptool: error: unrecognized arguments: -v
```

and:
```sh
$ python esptool.py --version
esptool: error: unrecognized arguments: -version
```

I think it's an important feature. Most software let their users consult the version nr this way.

kristofmulier, 2021-04-24:
Apparently, it works without a `'-'`, so like this:

```sh
$ python esptool.py version
```

My apologies.

radimkarnis, 2021-04-24:
Hello @kristofmulier,
indeed, the option to display the current version is implemented as a command, not an optional argument. See the esptool help screen for clarification.
There is an ongoing initiative to improve esptool docs, we will make sure to properly document this option.

closed 2021-04-26 by radimkarnis