espressif/esptool #626: Could not connect to an Espressif device on any of the available serial ports

mohamadnoor991, 2021-04-27:
**##  fatal error occurred: Could not connect to an Espressif device on any of the 1 available serial ports.**

Hello, I am trying to Erasing ESP32 Flash Memory but when I press the reset button and try to execute the command

` esptool.py --chip esp32 erase_flash` --> I get error "'_esptool.py' is not recognized as an internal or external command,
operable program or batch file."_

But when I am trying with `esptool --chip esp32 erase_flash` i get this error --> _COM17 failed to connect: could not open port 'COM17': PermissionError(13, 'Access is denied.', None, 5)

A fatal error occurred: Could not connect to an Espressif device on any of the 1 available serial ports._

[image]

In both case, I have a problem, how I can solve this problem.
thanks

radimkarnis, 2021-04-27:
Hello @mohamadnoor991, 
it seems like esptool is installed correctly and can be used just by calling `esptool erase_flash`. To run it directly from a source file, you need to use python: `python esptool.py erase_flash`.
Anyway, your connection problem is caused by an inaccessible COM port; it can only be open in one application at a time.
Try unplugging the device, then plug it in again and run `esptool erase_flash`.

mohamadnoor991, 2021-04-27:
@radimkarnis  thank you for your help that solves my problem.

note: I had opened the visual studio and I have some plugs-in installed on VS,to work with serial ports maybe that was one of the problems. for the people who will search later for such problem

closed 2021-04-27 by mohamadnoor991