espressif/esptool #630: MicroPython -> Cannot access Serial REPL Prompt after successful reflashing

toddparker3233, 2021-05-30:
# Operating system

Ubuntu 18.04

# Python version

Can run `python -V` to check this.

2.7.17

# What Chip

(For example: ESP8266, ESP32-PICO-D4, ESP32-WROOM32 module, etc)

ESP8266

# What development board or other hardware is the chip attached to

(For example: DevKitC, NodeMCU board, plain module on breadboard, etc)

If your hardware is custom or unusual, please attach a photo to the issue.

NodeMCU

# Is anything else attached to the development board, except for the serial flasher connections?

Example: GPIO 18 & 19 are connected to I2C devices.

No.

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

No. I'm running esptool.py from my terminal.

# Full esptool.py command line that was run:

1) sudo esptool.py --port /dev/ttyUSB0 erase_flash
2) sudo esptool.py --port /dev/ttyUSB0 write_flash --flash_size=detect 0 esp8266-20210418-v1.15.bin

# Full output from esptool.py

1)
esptool.py v2.0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.0s
Hard resetting...

2)
esptool.py v2.0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0040
Compressed 632632 bytes to 415633...
Wrote 632632 bytes (415633 compressed) at 0x00000000 in 36.8 seconds (effective 137.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

# Do you have any other information from investigating this?

I try to minicom into the device. I press enter multiple times. No MicroPython REPL prompt pops up. I've done this on both 115200 and 9600 baud.

# Is there any other information you can think of which will help us reproduce this problem?

Nope.

dobairoland, 2021-05-31:
Hi @toddparker3233. Maybe it is an access issue. I see that you are running `esptool` with `sudo` so probably minicom doesn't have access to the port neither.

It is not recommended to use `sudo` for these tools. You could add access to the serial port for your OS user. This is the related information from the README:

> In Linux, the current user may not have access to serial ports and a "Permission Denied" error will appear. On most Linux distributions, the solution is to add the user to the dialout group with a command like sudo usermod -a -G dialout <USERNAME>. Check your Linux distribution's documentation for more information.
 
You can check if the `dialout` group is the right one for you by running: 
```sh
ls -al /dev/ttyUSB*
```

radimkarnis, 2021-06-09:
Hello @toddparker3233, any update? have you been able to resolve your issue?
Radim

radimkarnis, 2021-06-18:
Closing this issue for inactivity. Feel free to reopen if further discussion is needed.

closed 2021-06-18 by radimkarnis

Erkman84, 2022-10-02:
Hi,
I have the (almost) the same issue: after flashing with micropython (Raspberry OS) i do not get any repl promt, and what is strange: i cannot upload any script (main.py, boot.py), In this case I used a ESP C3 module. But unfortunatelly with some esp8266 boards the same thing i cannot understand.
Repeating myself: I successfully flashed every board (!) (mentioned like on the man pages) BUT for some models after that i cannot upload any code, picocom repl answert, but there is input seems to be freezd. I thought it is conneted with CH340 driver but the system is up-to-date, so no kernel problems should be tried on antoher ubuntu 18 up-to-date system but it seems not to be an issue with my system.

Other Example:
Wemos D1 mini eps8266 V1 --> flash: ok, write script: ok. That's okay.
Wemos D1 mini esp8266 V2 --> other chip (?): flash: ok, write script fails, no repl prompt in picocom.

Tried windows 7 (with updated driver) with putty to connect, also no repl promt for the selected devices.

Anyone an idea? Micropython 1.19 bin --> this should be also not the problem, with the non-problematic devices it's working fine.