espressif/esptool #631: Need to change this from ERROR to INFO or something else "less alarming"

UnexpectedMaker, 2021-06-03:
```Leaving...
Hard resetting via RTS pin...
ERROR: ESP32-S2 chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this error, set --after option to 'no_reset'.
```

I get 3-4 community contacts a week from people telling me they are unable to flash one of my boards via the Arduino IDE because they see this final error and think the flashing didn't work - instead of reading further up to see that it did and that it's just that esptool can't reset the board for them.

Calling it an ERROR is not only totally confusing for people, but it's really not an error at all.

radimkarnis, 2021-06-09:
Hello @UnexpectedMaker,

Thanks for your awesome contributions to the community and for your valuable input. 

I agree this is unnecessarily confusing and might cause some misunderstanding. 
We will fix that!

Radim

closed 2021-07-07 by espressif-bot

spikey1973, 2022-12-21:
hello I just got the same "error" msg after flashing micropython on my ESP32-S2FNR2 (esp32 s2 mini). 
Unfortunately, just like esptool, i can't seem to get out of this download mode.
even after manually pressing reset button, it still keeps connecting into the download mode. how can i force this board to exit?

radimkarnis, 2022-12-22:
Hi @spikey1973,
If your board keeps entering the download mode after resetting or even after disconnecting/reconnecting power, the GPIO0 is probably being pulled to GND. Try disconnecting everything and check for bridges on the pin.

juggledad, 2023-06-23:
I've been having the same problem trying to flash MicroPython on the ESP32-S2 mini using a Mac with Ventura. I finally figured out what I needed to do
1) Plug a usb-c cable into the board and then the other end of the cable into the Mac.
2) Press and hold the 0 button then the RST. Hold them for 6 seconds and release the RST then 0 button. At this point if I run **ls /dev/cu*** it shows
**/dev/cu.usbmodem01** 
3) Erase the firmware using:
**esptool.py --chip esp32s2 --port /dev/cu.usbmodem01 erase_flash**
4) Write new Firmware: (make sure to change teh path to where you have the firmware bin file)
**esptool.py --chip esp32s2 --port /dev/cu.usbmodem01 write_flash -z 0x1000 /Volumes/Gremil-two/Paul_Stuff/Downloads/firmware-LOLIN_S2_MINI-v1.19.1-1061-g20b3470e2.bin**
5) Now, unplug the USB cable and plug it back in and run **ls /dev/cu*** and it nows shows
**/dev/cu.usbmodem1234561**
and at this point, I can use Thonny to program the device.

I hope this helps someone else.

lp85d, 2023-10-14:
Arduino IDE
Version: 2.2.1
Date: 2023-08-31T14:35:44.802Z (1 мес. назад)
CLI Version: 0.34.0
Copyright © 2023 Arduino SA

eaving... WARNING: ESP32-S2FNR2 (revision v0.0) chip was placed into download mode using GPIO0. esptool.py can not exit the download mode over USB. To run the app, reset the chip manually. To suppress this note, set --after option to 'no_reset'. Failed uploading: uploading error: exit status 1

wayzha, 2024-03-28:
still the same error! i guess that’s some how related to a specific hw, but better to change from error to warning only.

Docteh, 2024-04-10:
@wayzha check the version of your copy of esptool? I ran into this message last night and decided to look it up.

have to do `esptool -h` to get a version, my particularly old copy from 2020 says

> esptool.py v3.0-dev - ESP8266 ROM Bootloader Utility

Looking at the commit that closed this issue, this change to a warning should be in any version of esptool after v3.2

wayzha, 2024-04-11:
@Docteh Thx for reply. below is the version:
esptool.py v4.7.0 - Espressif chips ROM Bootloader Utility.

I have no idea how to change the behavior from Arduino IDE, so, it works only if i use command lines with --after no_reset