espressif/esptool #633: error: argument <address> <filename>: Address "\--flash_freq" must be a number

hypo-thesis, 2021-06-19:
* Operating system:
* Python version: (`python -V` to check this)
Python 3.9.5
# Full esptool.py command line as run:

```
python esptool.py --port COM10 --baud 460800 write_flash \
  --flash_freq 80m --flash_mode qio --flash_size 4MB-c1 \
  0x0000 "boot_v1.6.bin" 0x1000 espruino_esp8266_user1.bin \
  0x3FC000  esp_init_data_default.bin 0x3FE000 blank.bin

```

# Full output from esptool.py

```
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION]
                           [--no-progress] [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address> <filename>
                           [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Address "\--flash_freq" must be a number
```

# What is the expected behaviour?

Flash the new bin

Example: Print the version number

# Do you have any other information from investigating this?

I am on windows 10 and my chip is Nodemcu ch340g.

hypo-thesis, 2021-06-19:
I ended up using` python esptool.py --port COM10 --baud 460800 write_flash --flash_size=detect -fm dio 0 espruino_2v09_esp8266_4mb_combined_4096.bin` to flash it but still pretty weird the error I got.

radimkarnis, 2021-06-21:
Hello @hypo-thesis,

The issue seems to be the way the command is input into the terminal. Clearly, the `\` symbol gets merged with the `--flash_freq` argument. The line-continuation fails if you have whitespace (spaces or tab characters) after the backslash and before the newline.

Did you run the command manually (copying it from somewhere) or was it invoked by another script/program?

radimkarnis, 2021-07-01:
Closing this issue, feel free to reopen if you believe this needs further discussion.

closed 2021-07-01 by radimkarnis