espressif/esptool #636: Problem with compiling with Arduino and Big Sur.

roontoon, 2021-06-27:
# Operating system
Mac OS 11.4 Big Sur

# Python version
Python 3.9.5

# What Chip
ESP32-WROOM-32

# What development board or other hardware is the chip attached to
ESP32-DEVKITC-32UE

# Is anything else attached to the development board, except for the serial flasher connections?
No

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
Arduino 1.8.15

# Full esptool.py command line that was run:
N/A via IDE

# Full output from esptool.py

Arduino: 1.8.15 (Mac OS X), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

Traceback (most recent call last):
  File "esptool.py", line 57, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports.py", line 29, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports_posix.py", line 31, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports_osx.py", line 32, in <module>
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
Failed to execute script esptool
Multiple libraries were found for "WiFi.h"
 Used: /Users/roontoon/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 255
/Applications/Arduino.app/Contents/Java/arduino-builder returned 255
Error compiling for board ESP32 Dev Module.

# Do you have any other information from investigating this?
I have fully deleted the install of Arduino 1.8.15 and all associated files and reinstalled, updated Python to the most current version.

# Is there any other information you can think of which will help us reproduce this problem?

dobairoland, 2021-06-27:
Hi @roontoon. This issue has already been fixed for esptool by a workaround (if you specify a port then the issue is prevented to happen), by a pyserial update and a fix on the MacOS side. Please see https://github.com/espressif/esptool/issues/540 for further details.

Please look for a solution in https://github.com/espressif/arduino-esp32 and report the issue there if you cannot find one.

closed 2021-06-27 by dobairoland

dobairoland, 2021-06-27:
Here is the related Arduino issue: https://github.com/espressif/arduino-esp32/issues/4408