espressif/esptool #640: esp32 does not boot after read protecting block2

romatetemadze, 2021-07-06:
Hello guys, I have a following issue and I am not sure how to proceed. I have ab ESP32 Wrover 16mb SoC, with software that I am loading as follows:

1. Bootloader, Partition Table, Initial OTA, App, Ota0, Ota1 are encrypted
2. I am using secure boot v2 (chip revision is obviously 3)
3. I was trying to secure as much fuses as I could, but I fall into an issue the moment I read protected the BLOCK2

Right after read protecting the chip, it won't boot anymore (before that it would boot normally into the app that was active). Am I not supposed to read protect block2 at all? 

I ran this command to secure block 2:
`python espefuse.py --port COM3 read_protect_efuse BLOCK2`

Here's output of fuse summary:
`C:\Users\dev\app1>python espefuse.py summary --chip esp32 --port=COM3
Connecting......
espefuse.py v3.0
EFUSE_NAME (Block)                       Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
BLK3_PART_RESERVE (BLOCK0):              BLOCK3 partially served for ADC calibration data   = False -/- (0b0)
ADC_VREF (BLOCK0):                       Voltage reference calibration                      = 1121 R/W (0b00011)

Config fuses:
XPD_SDIO_FORCE (BLOCK0):                 Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = True R/- (0b1)
XPD_SDIO_REG (BLOCK0):                   If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = True R/- (0b1)
XPD_SDIO_TIEH (BLOCK0):                  If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 3.3V R/- (0b1)
CLK8M_FREQ (BLOCK0):                     8MHz clock freq override                           = 54 R/W (0x36)
SPI_PAD_CONFIG_CLK (BLOCK0):             Override SD_CLK pad (GPIO6/SPICLK)                 = 0 R/W (0b00000)
SPI_PAD_CONFIG_Q (BLOCK0):               Override SD_DATA_0 pad (GPIO7/SPIQ)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_D (BLOCK0):               Override SD_DATA_1 pad (GPIO8/SPID)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_HD (BLOCK0):              Override SD_DATA_2 pad (GPIO9/SPIHD)               = 0 R/W (0b00000)
SPI_PAD_CONFIG_CS0 (BLOCK0):             Override SD_CMD pad (GPIO11/SPICS0)                = 0 R/W (0b00000)
DISABLE_SDIO_HOST (BLOCK0):              Disable SDIO host                                  = False R/W (0b0)

Efuse fuses:
WR_DIS (BLOCK0):                         Efuse write disable mask                           = 62880 R/W (0xf5a0)
RD_DIS (BLOCK0):                         Efuse read disable mask                            = 11 R/W (0xb)
CODING_SCHEME (BLOCK0):                  Efuse variable block length scheme
   = NONE (BLK1-3 len=256 bits) -/- (0b00)
KEY_STATUS (BLOCK0):                     Usage of efuse block 3 (reserved)                  = False -/- (0b0)

Identity fuses:
MAC (BLOCK0):                            Factory MAC Address
   = c4:5b:be:34:4b:88 (CRC 0x94 OK) R/W
MAC_CRC (BLOCK0):                        CRC8 for factory MAC address                       = 148 R/W (0x94)
CHIP_VER_REV1 (BLOCK0):                  Silicon Revision 1                                 = True R/W (0b1)
CHIP_VER_REV2 (BLOCK0):                  Silicon Revision 2                                 = True R/W (0b1)
CHIP_VERSION (BLOCK0):                   Reserved for future chip versions                  = 2 R/W (0b10)
CHIP_PACKAGE (BLOCK0):                   Chip package identifier                            = 1 R/W (0b001)
MAC_VERSION (BLOCK3):                    Version of the MAC field                           = 0 R/W (0x00)

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                Flash encryption mode counter                      = 1 R/W (0b0000001)
UART_DOWNLOAD_DIS (BLOCK0):              Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):             Flash encryption config (key tweak bits)           = ? -/- (0x0)
CONSOLE_DEBUG_DISABLE (BLOCK0):          Disable ROM BASIC interpreter fallback             = True R/- (0b1)
ABS_DONE_0 (BLOCK0):                     Secure boot V1 is enabled for bootloader image     = False R/- (0b0)
ABS_DONE_1 (BLOCK0):                     Secure boot V2 is enabled for bootloader image     = True R/- (0b1)
JTAG_DISABLE (BLOCK0):                   Disable JTAG                                       = True R/- (0b1)
DISABLE_DL_ENCRYPT (BLOCK0):             Disable flash encryption in UART bootloader        = True R/- (0b1)
DISABLE_DL_DECRYPT (BLOCK0):             Disable flash decryption in UART bootloader        = True R/- (0b1)
DISABLE_DL_CACHE (BLOCK0):               Disable flash cache in UART bootloader             = True R/- (0b1)
BLOCK1 (BLOCK1):                         Flash encryption key
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
BLOCK2 (BLOCK2):                         Secure boot key
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
BLOCK3 (BLOCK3):                         Variable Block 3
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Flash voltage (VDD_SDIO) set to 3.3V by efuse.
`

# Operating system: FreeRTOS
# Python version 3.7.3
# What Chip ESP32-WROVER module, etc)

Many thanks in advance.

KonstantinKondrashov, 2021-07-12:
Hi @romatetemadze
Secure boot V2 is supposed to have a valid hash of the public key in the eFuse block, it should be readable and write-protected! (if you set read protection bit for this eFuse block it means all zeros to be read, ).
Do not do this `espefuse.py read_protect_efuse BLOCK2`, 
Use it `espefuse.py burn_key_digest`. you will get readable and read-protected eFuse BLOCK.

romatetemadze, 2021-07-14:
> Hi @romatetemadze
> Secure boot V2 is supposed to have a valid hash of the public key in the eFuse block, it should be readable and write-protected! (if you set read protection bit for this eFuse block it means all zeros to be read, ).
> Do not do this `espefuse.py read_protect_efuse BLOCK2`,
> Use it `espefuse.py burn_key_digest`. you will get readable and write-protected eFuse BLOCK.

Thanks for the answer. I do understand the logic, but my expectation was that it would still be available for the core system to read it. Like for example BLOCK1 - it's read protected by default, yet the system uses the key to decrypt flash content. Also I can read protect FLASH_CRYPT_CONFIG without any negative impact on the encrypted flash operations.

I believe you should reflect this in the documentation, or even in the espfuse script, so it won't let you read protect BLOCK2 when ABS_DONE_1 is set to 1.

closed 2021-08-04 by espressif-bot