espressif/esptool #642: Timed out waiting for packet content: different behaviour between v2.3.2 and v3.0

brazoayeye, 2021-07-14:
All works well if I use v2.3.2, but I have a problem using v3.0. It seems that the received packed is missing last part, even if it's present

Here the --trace with v2.3.2 (only sync + reply)
```
esptool.exe --trace --chip esp32 --no-stub --after soft_reset --port COM8 --baud 38400 write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x19000 C:\test.bin  --no-compress

esptool.py v2.3.2-dev
Connecting...
TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=b'\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
TRACE +0.005 Write 46 bytes: b'\xc0\x00\x08$\x00\x00\x00\x00\x00\x07\x07\x12 UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\xc0'
TRACE +0.029 Read 1 bytes: b'\xc0'
TRACE +0.001 Read 1 bytes: b'\x01'
TRACE +0.000 Read 2 bytes: b'\x08\x04'
TRACE +0.001 Read 1 bytes: b'\x00'
TRACE +0.000 Read 2 bytes: b'\x00\x00'
TRACE +0.001 Read 1 bytes: b'\x00'
TRACE +0.004 Read 6 bytes: b'\x00\x00\x00\x00\x00\xc0'
TRACE +0.001 Full packet: b'\x01\x08\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00'
```

And here what I have with v3.0
```
esptool.py --trace --chip esp32 --no-stub --after soft_reset --port COM8 --baud 38400 write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x19000 C:\test.bin  --no-compress

esptool.py v3.0
Serial port COM8
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.006 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.036 Read 1 bytes: c0
TRACE +0.001 Read 1 bytes: 01
TRACE +0.005 Read 6 bytes: 080400000000
TRACE +0.126 Timed out waiting for packet content
```

With the previous version I receive 1+1+2+1+2+1+6=14 bytes With the previous v3 version I receive 1+1+6=8 bytes

Sniffing in the serial, using Realterm, I see
```
C00008240000000000070712205555555555555555555555555555555555555555555555555555555555555555C0
C0010804000000000000000000C0
```

Thanks

# Operating system
Winzozz 10 Pro Build 19043.1083

# Python version
Python 3.8.2

# What Chip
ESP32-WROOM 4MB

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
No IDE, Windows command prompt, same problem running from pyCharm inside other code

# Full esptool.py command line that was run:
esptool.py --trace --chip esp32 --no-stub --after soft_reset --port COM8 --baud 38400 write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x19000 C:\test.bin  --no-compress

# Full output from esptool.py
esptool.py v3.0
Serial port COM8
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.006 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.036 Read 1 bytes: c0
TRACE +0.001 Read 1 bytes: 01
TRACE +0.005 Read 6 bytes: 080400000000
TRACE +0.126 Timed out waiting for packet content
.TRACE +0.074 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=

# Is there any other information you can think of which will help us reproduce this problem?
Previous version, v2.3.2-dev, works without problems

brazoayeye, 2021-07-14:
I have the same problem downloading the 3.1exe and the 2.3.1 py, probably starting 2.3.2 exe i have has been customized to fix that problem.... 

My serial settings are 38400 N 8 1 , are they right - how can be set in the py?

brazoayeye, 2021-07-14:
I failed, the problem was the serial adapter that works only with dtr = True for an electrical reason. 

btw I needed to change also a part of a function where command was called with op=None. 
```
    def sync(self):
        self.command(self.ESP_SYNC, b'\x07\x07\x12\x20' + 32 * b'\x55',
                     timeout=SYNC_TIMEOUT)
        #for i in range(7):
        #    self.command()
```

Why should we wait 7 replies more if only one command is sent?

Thanks

radimkarnis, 2021-07-15:
Hello @brazoayeye,
I am glad you could solve your problem!

Bootloader responds to the SYNC request with eight identical SYNC responses. The stub reacts the same way so SYNC is possible with the flasher stub as well. This helps in cases when the chip cannot be reset and the flasher stub keeps running.

Running command() with op=None seven more times is just a measure needed in order to catch all of these 8 SYNC responses.

radimkarnis, 2021-07-16:
Feel free to reopen this issue if you think further discussion is needed!

closed 2021-07-16 by radimkarnis