espressif/esptool #643: Cannot read/write/erase FLASH after encryption has been enabled

amiotk, 2021-07-15:
# Problem Description

I am unable to perform any of the FLASH operations. Although it may be a coincidence but the last successful programming was when enabling FLASH encryption. Below are the exact commands:
```
espefuse.py --port COM9 burn_efuse FLASH_CRYPT_CNT 0x7F
espefuse.py --port COM9 burn_key flash_encryption key.bin --no-protect-key
espefuse.py --port COM9 burn_efuse FLASH_CRYPT_CONFIG 0xF
espefuse.py --port COM9 burn_efuse DISABLE_DL_ENCRYPT 1
espefuse.py --port COM9 burn_efuse DISABLE_DL_DECRYPT 1
espefuse.py --port COM9 burn_efuse DISABLE_DL_CACHE 1
espefuse.py --port COM9 burn_efuse JTAG_DISABLE 1
espefuse.py --port COM9 read_protect_efuse BLOCK1
espefuse.py --port COM9 write_protect_efuse BLOCK1
esptool.py --port COM9 -b 921600 write_flash 0 flash_all_encrypted.bin
```

I don't want to use self-encryption supported by the ESP32 - just want to keep the module updatable via serial interface thus leaving `UART_DOWNLOAD_DIS` eFuse not set.

Next time I tried to reprogram I found out that any read/write/erase or chip_id command performed by esptool results in an error message `A fatal error occurred: Timed out waiting for packet header`.

Please see setup details and eFuses summary below and advise if it's an issue with the eFuses settings (and the device is a brick) or could be anything else.

# Operating system
Windows 10

# Python version
3.9

# What Chip
ESP32-D0WD-V3

# What development board or other hardware is the chip attached to
DevKitC

# Is anything else attached to the development board, except for the serial flasher connections?
No

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
No IDE, Windows command prompt

# Full esptool.py command line that was run:

```
esptool.py -p COM9 -b 19200 --no-stub --trace read_flash 0x1000 16 dump.bin
```

# Full output from esptool.py
```
esptool.py v3.1
Serial port COM9
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.105 Timed out waiting for packet header
.TRACE +0.065 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.084 Read 1 bytes: c0
TRACE +0.001 Read 111 bytes:
    0108040020555555 00000000c0c00108 | .... UUU........
[... 135 lines cut ...]
CHIP_VER_REV2 (BLOCK0):                  Silicon Revision 2                                 = True R/W (0b1)
CHIP_VERSION (BLOCK0):                   Reserved for future chip versions                  = 2 R/W (0b10)
CHIP_PACKAGE (BLOCK0):                   Chip package identifier                            = 1 R/W (0b001)
MAC_VERSION (BLOCK3):                    Version of the MAC field                           = 0 R/W (0x00)

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                Flash encryption mode counter                      = 127 R/W (0b1111111)
UART_DOWNLOAD_DIS (BLOCK0):              Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):             Flash encryption config (key tweak bits)           = 15 R/W (0xf)
CONSOLE_DEBUG_DISABLE (BLOCK0):          Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                     Secure boot V1 is enabled for bootloader image     = False R/W (0b0)
ABS_DONE_1 (BLOCK0):                     Secure boot V2 is enabled for bootloader image     = False R/W (0b0)
JTAG_DISABLE (BLOCK0):                   Disable JTAG                                       = True R/W (0b1)
DISABLE_DL_ENCRYPT (BLOCK0):             Disable flash encryption in UART bootloader        = True R/W (0b1)
DISABLE_DL_DECRYPT (BLOCK0):             Disable flash decryption in UART bootloader        = True R/W (0b1)
DISABLE_DL_CACHE (BLOCK0):               Disable flash cache in UART bootloader             = True R/W (0b1)
BLOCK1 (BLOCK1):                         Flash encryption key
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
BLOCK2 (BLOCK2):                         Secure boot key
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK3 (BLOCK3):                         Variable Block 3
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).
```

amiotk, 2021-07-16:
Problem solved!
I turned out to be an issue with JTAG adapter helding IO12 high during power-up.

closed 2021-07-16 by amiotk

radimkarnis, 2021-07-16:
Hi @amiotk,
I am glad you managed to solve your issue and thanks for letting us know!