espressif/esptool #645: Traceback error while trying to read flash

TinManAkshay, 2021-07-22:
Hi all,

I've a custom board which has SAMD51 MCU as well as Ublox NINA W152 WiFi module. This Wifi chip is based on ESP32 core.

I've been wanting to flash ESP32 AT firmware into the wifi chip using Espressif esptool so that I will make an .ino in Arduino IDE.

So far I've been trying to run these commands on a Linux system as follow: 

1. $ esptool.py flash_id

Outcome - 

esptool.py v3.2-dev
Found 1 serial ports
Serial port /dev/ttyACM0
Connecting...
Traceback (most recent call last):
  File "/home/daniel/.local/bin/esptool.py", line 7, in <module>
    exec(compile(f.read(), __file__, 'exec'))
  File "/esptool/esptool.py", line 4830, in <module>
    _main()
  File "/esptool/esptool.py", line 4823, in _main
    main()
  File "/esptool/esptool.py", line 4283, in main
    before=args.before)
  File "/esptool/esptool.py", line 118, in get_default_connected_device
    connect_attempts)
  File "/esptool/esptool.py", line 365, in detect_chip
    detect_port.connect(connect_mode, connect_attempts, detecting=True)
  File "/esptool/esptool.py", line 644, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False, usb_jtag_serial=usb_jtag_serial)
  File "/esptool/esptool.py", line 611, in _connect_attempt
    self.sync()
  File "/esptool/esptool.py", line 499, in sync
    timeout=SYNC_TIMEOUT)
  File "/esptool/esptool.py", line 437, in command
    self.write(pkt)
  File "/esptool/esptool.py", line 400, in write
    self._port.write(buf)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 552, in write
    raise writeTimeoutError
serial.serialutil.SerialTimeoutException: Write timeout

2. $ esptool.py --trace flash_id  (or)  esptool.py --trace --no-stub flash_id 

Outcome - 

esptool.py v3.2-dev
Found 1 serial ports
Serial port /dev/ttyACM0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.

Traceback (most recent call last):
  File "/home/daniel/.local/bin/esptool.py", line 7, in <module>
    exec(compile(f.read(), __file__, 'exec'))
  File "/esptool/esptool.py", line 4830, in <module>
    _main()
  File "/esptool/esptool.py", line 4823, in _main
    main()
  File "/esptool/esptool.py", line 4283, in main
    before=args.before)
  File "/esptool/esptool.py", line 118, in get_default_connected_device
    connect_attempts)
  File "/esptool/esptool.py", line 365, in detect_chip
    detect_port.connect(connect_mode, connect_attempts, detecting=True)
  File "/esptool/esptool.py", line 644, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False, usb_jtag_serial=usb_jtag_serial)
  File "/esptool/esptool.py", line 611, in _connect_attempt
    self.sync()
  File "/esptool/esptool.py", line 499, in sync
    timeout=SYNC_TIMEOUT)
  File "/esptool/esptool.py", line 437, in command
    self.write(pkt)
  File "/esptool/esptool.py", line 400, in write
    self._port.write(buf)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 552, in write
    raise writeTimeoutError
serial.serialutil.SerialTimeoutException: Write timeout

3. esptool.py -p /dev/ttyACM0 -b 115200 read_flash 0 0x400000 flash_dump.bin

Outcome - 

esptool.py v3.2-dev
Serial port /dev/ttyACM0
Connecting...
Traceback (most recent call last):
  File "/home/daniel/.local/bin/esptool.py", line 7, in <module>
    exec(compile(f.read(), __file__, 'exec'))
  File "/esptool/esptool.py", line 4830, in <module>
    _main()
  File "/esptool/esptool.py", line 4823, in _main
    main()
  File "/esptool/esptool.py", line 4283, in main
    before=args.before)
  File "/esptool/esptool.py", line 118, in get_default_connected_device
    connect_attempts)
  File "/esptool/esptool.py", line 365, in detect_chip
    detect_port.connect(connect_mode, connect_attempts, detecting=True)
  File "/esptool/esptool.py", line 644, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False, usb_jtag_serial=usb_jtag_serial)
  File "/esptool/esptool.py", line 611, in _connect_attempt
    self.sync()
  File "/esptool/esptool.py", line 499, in sync
    timeout=SYNC_TIMEOUT)
  File "/esptool/esptool.py", line 437, in command
    self.write(pkt)
  File "/esptool/esptool.py", line 400, in write
    self._port.write(buf)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 552, in write
    raise writeTimeoutError
serial.serialutil.SerialTimeoutException: Write timeout

So you have observed whenever I try to read the flash id or flash contents, I will get the same traceback error. I cannot figure out why is it happening like that. I'd appreciate any kind of help or feedback here. 

Thanks,
Akshay

dobairoland, 2021-07-23:
I've just noticed that you use `/dev/ttyACM0`. That indicates to me that a native USB is used and you don't connect directly to ESP32 (which has only UART so the port would be `/dev/ttyUSBx` with an USB-UART bridge).

If the RTS, DTR signals are not properly transferred through that native USB bridge then esptool won't be able to communicate properly. Manually changing the EN and GPIO0 pins might help as Radim suggests.

Other solution could be to make a direct connection with an USB dongle to EN, GPIO0, RxD, TxD pins.

TinManAkshay, 2021-07-23:
Hi - 

Please look at this sketch how NINA is connected ->

[image]

[image]

Your thoughts on how the pins are connected??

TinManAkshay, 2021-07-23:
Hi @dobairoland ,

I used installed esptool via command prompt on Windows as well. When I run this command "esptool.py --trace -p COM5 flash_id", the same output shows as posted yesterday right off the bat. 

You can look at the schematic above how RTS and DTR are connected. Hope connections are not a problem here. 

Can you please tell what do you mean by EN and GPIO0 pins as I am afraid I cannot hard-wire the wifi signals as these are traced inside the PCB. In other words, we don't have external connections to the wifi module pinout.

dobairoland, 2021-07-26:
Please take a look at this page for a detailed description: https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection

ESP32 has to been put into bootloader mode for esptool. If you cannot put ESP32 into bootloader mode then esptool cannot work with ESP32.

If the serial interface is connected to EN & GPIO0 of ESP32 through the serial DTR & RTS signals then esptool can put the ESP32 into bootloader mode automatically. Otherwise, the only option to put the ESP32 into bootloader mode manually by toggling EN & GPIO0. Of course, for this you need direct access to the ports. Probably, this could be achieved with the BOOT and RESET button as Radim suggests a couple of posts earlier. 

Maybe your issue should be discussed with NINA engineers? Please note that we don't know any inside information about that WIFI chip. Esptool supports ESP8266- and ESP32-series chips but not third-party chips based on them.

closed 2021-07-26 by dobairoland