espressif/esptool #646: esptool.py does not connect to MCU for flashing.

KingZuluKing, 2021-07-23:
# Operating system  = Windows 7

# Python version 3.8

# What Chip ESP32 WRoverB

# What development board or other hardware is the chip attached to
If your hardware is custom or unusual, please attach a photo to the issue.

# Is anything else attached to the development board, except for the serial flasher connections?
Nothing is attached to the board, only the UART Comm. channel GPIO with Silicon Labs USB-UART Bridge

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
Arduino with ESP tool

# Full esptool.py command line that was run:

esptool.py -p COM999 write_flash 0x0 bootloader.bin

# Full output from esptool.py

D:\Users\Arhitect\AppData\Local\Temp\Rar$EXa2968.16225\arduino-nightly\arduino-builder -dump-prefs -logger=machine -hardware D:\Users\Arhitect\AppData\Local\Temp\Rar$EXa2968.16225\arduino-nightly\hardware -hardware D:\Users\Arhitect\AppData\Local\Arduino15\packages -tools D:\Users\Arhitect\AppData\Local\Temp\Rar$EXa2968.16225\arduino-nightly\tools-builder -tools D:\Users\Arhitect\AppData\Local\Temp\Rar$EXa2968.16225\arduino-nightly\hardware\tools\avr -tools D:\Users\Arhitect\AppData\Local\Arduino15\packages -built-in-libraries D:\Users\Arhitect\AppData\Local\Temp\Rar$EXa2968.16225\arduino-nightly\libraries -libraries D:\Users\Arhitect\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32wrover:PartitionScheme=default,FlashMode=qio,FlashFreq=80,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10815 -build-path D:\Users\Arhitect\AppData\Local\Temp\arduino_build_511987 -warnings=none -build-cache D:\Users\Arhitect\AppData\Local\Temp\arduino_cache_308774 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=D:\Users\Arhitect\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=D:\Users\Arhitect\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.mkspiffs.path=D:\Users\Arhitect\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\Users\Arhitect\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\Users\Arhitect\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=D:\Users\Arhitect\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -verbose D:\Users\Arhitect\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\ESP32\examples\ChipID\GetChipID\GetChipID.ino
D:\Users\Arhitect\AppData\Local\Temp\Rar$EXa2968.16225\arduino-nightly\arduino-builder -compile -logger=machine -hardware D:\Users\Arhitect\AppData\Local\Temp\Rar$EXa2968.16225\arduino-nightly\hardware -hardware D:\Users\Arhitect\AppData\Local\Arduino15\packages -tools D:\Users\Arhitect\AppData\Local\Temp\Rar$EXa2968.16225\arduino-nightly\tools-builder -tools D:\Users\Arhitect\AppData\Local\Temp\Rar$EXa2968.16225\arduino-nightly\hardware\tools\avr -tools D:\Users\Arhitect\AppData\Local\Arduino15\packages -built-in-libraries D:\Users\Arhitect\AppData\Local\Temp\Rar$EXa2968.16225\arduino-nightly\libraries -libraries D:\Users\Arhitect\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32wrover:PartitionScheme=default,FlashMode=qio,FlashFreq=80,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10815 -build-path D:\Users\Arhitect\AppData\Local\Temp\arduino_build_511987 -warnings=none -build-cache D:\Users\Arhitect\AppData\Local\Temp\arduino_cache_308774 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=D:\Users\Arhitect\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=D:\Users\Arhitect\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.mkspiffs.path=D:\Users\Arhitect\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\Users\Arhitect\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\Users\Arhitect\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=D:\Users\Arhitect\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -verbose D:\Users\Arhitect\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\ESP32\examples\ChipID\GetChipID\GetChipID.ino
Using board 'esp32wrover' from platform in folder: D:\Users\Arhitect\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6
Using core 'esp32' from platform in folder: D:\Users\Arhitect\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6
cmd /c if exist "D:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\libraries\\ESP32\\examples\\ChipID\\GetChipID\\partitions.csv" copy /y "D:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\libraries\\ESP32\\examples\\ChipID\\GetChipID\\partitions.csv" "D:\\Users\\Arhitect\\AppData\\Local\\Temp\\arduino_build_511987\\partitions.csv"
cmd /c if not exist "D:\\Users\\Arhitect\\AppData\\Local\\Temp\\arduino_build_511987\\partitions.csv" copy "D:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\partitions\\default.csv" "D:\\Users\\Arhitect\\AppData\\Local\\Temp\\arduino_build_511987\\partitions.csv"
Detecting libraries used...
"D:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/config" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_trace" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_update" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/asio" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bootloader_support" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bt" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/coap" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/console" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/driver" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/efuse" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp-tls" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp32" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_adc_cal" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_event" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_http_client" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_http_server" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_https_ota" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_https_server" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_ringbuf" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_websocket_client" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/espcoredump" "-ID:\\Users\\Arhitect\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/ethernet" "-ID [... cut ...] .bin 
esptool.py v3.0-dev
Serial port COM37
Connecting........_____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch
_

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

# Do you have any other information from investigating this?

Since Powering up, it was never possible to program it over this tool....

# Is there any other information you can think of which will help us reproduce this problem?
[image]

igrr, 2021-07-23:
@KingZuluKing Regarding the SW2 and SW3 buttons in the schematic, are you doing anything with them when or prior to uploading the sketch?

KingZuluKing, 2021-07-23:
no, they are built-in only due to the datasheet of ESP32, to program and reboot the MCU.

radimkarnis, 2021-07-23:
> to program and reboot the MCU

Exactly; can you try holding the `Boot_Option` button and pressing the `Reset` button at the same time prior to uploading the sketch? There is no auto-reset circuitry implemented in your schematic, therefore the only option to enter the download mode is with the buttons.

KingZuluKing, 2021-07-23:
I did, same issue occurs. The ESP WRoverB ist soldered directly from the producer package, that is, I dunno, does it come with bootloader pre-burnt? normally for the AVR MCUs you have to burn bootloader before you do something else.

dobairoland, 2021-07-26:
Please try to connect any terminal program (eg. putty) to the same port (baud 115200, 8 bits, 1 stop bit, no parity, no flow control). After that please use your switch button to put the chip into bootloader mode. Do you capture something similar in the terminal program?

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x6 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_FEO_V2))
waiting for download
```

More on this topic: https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection

KingZuluKing, 2021-07-31:
Hello guys, I've tried what you have proposed, connected to the same port via terminal, and then hit the boot option, there is nothing showing up inside the window.
It seems that no communication is happening.

dobairoland, 2021-08-01:
> and then hit the boot option,

Have you used reset as well?

> Exactly; can you try holding the Boot_Option button and pressing the Reset button at the same time prior to uploading the sketch?

KingZuluKing, 2021-08-01:
update on issue: I have connected the Rx-Tx pins on CP2102N-A02-GQFN28 usb-uart bridge, together to see if there is echo happening on the bus. When I connect to port and open terminal, everything I type, I am getting back, that means the usb-uart side is working properly. 

On ESP32 WRover: When I hit the button "RESET_ESP_32" aka "EN", I can see on Scope trace, that it goes LOW. But what I also see, on the seconds channel on scope, which is connected to GPIO-0, is that the same LOW pulse. What I understand is why there is also LOW pulse on Boot aka GPIO-0 when I press the "RESET_ESP_32" aka "EN"? are they somehow connected ?
In scope trace, the upper is button "RESET_ESP_32" aka "EN", and the lower curve is the Boot aka GPIO-0. 
As you can see, pressing  button "RESET_ESP_32" appears on GPIO-0 as well.

[image]

KingZuluKing, 2021-08-12:
okay guys, I found the issue. It was on this Silicon Labs CP xxxx USB-UART Bridge. The evil QFN package was faulty or similar.
Thanks a lot everyone for their input. 

greetings from Physalis Labs Berlin ^_^

closed 2021-08-13 by KingZuluKing

dobairoland, 2021-08-13:
Thank you @KingZuluKing for the follow-up information. I'm happy that you were able to resolve the issue.