espressif/esptool #647: Protocol error when flashing ESP32S2

xtrinch, 2021-07-30:
Hi, I am having trouble flashing my board (with platformio and manually). I always have to put it into bootloader mode, and then I can flash it. I'm using https://github.com/chegewara/EspTinyUSB as my usb stack and I can see the output of my board via the serial monitor, I just can't flash it.

Operating system: Linux
Python version: 3.8
What Chip/Board:
ESP32-S2-DevKit-Lipo-USB from Olimuex
https://www.olimex.com/Products/IoT/ESP32-S2/ESP32-S2-DevKit-Lipo-USB/open-source-hardware

The board has two SPI devices, a display and a sensor board attached.

Full esptool.py command line that was run:
```
xtrinch@elizabeta:~/repos/esp32s2-bme280-webusb$ /home/xtrinch/.platformio/penv/bin/esptool.py --chip esp32s2  --port /dev/ttyACM0 write_flash 0x00000 .pio/build/esp32s2doit-devkit-v1/firmware.bin
esptool.py v3.1
Serial port /dev/ttyACM0
Connecting...
Traceback (most recent call last):
  File "/home/xtrinch/.platformio/penv/bin/esptool.py", line 4582, in <module>
    _main()
  File "/home/xtrinch/.platformio/penv/bin/esptool.py", line 4575, in _main
    main()
  File "/home/xtrinch/.platformio/penv/bin/esptool.py", line 4074, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
  File "/home/xtrinch/.platformio/penv/bin/esptool.py", line 121, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "/home/xtrinch/.platformio/penv/bin/esptool.py", line 629, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False, usb_jtag_serial=usb_jtag_serial)
  File "/home/xtrinch/.platformio/penv/bin/esptool.py", line 590, in _connect_attempt
    self.bootloader_reset(esp32r0_delay, usb_jtag_serial)
  File "/home/xtrinch/.platformio/penv/bin/esptool.py", line 553, in bootloader_reset
    self._setDTR(False)  # IO0=HIGH
  File "/home/xtrinch/.platformio/penv/bin/esptool.py", line 499, in _setDTR
    self._port.setDTR(state)
  File "/home/xtrinch/.platformio/penv/lib/python3.8/site-packages/serial/serialutil.py", line 603, in setDTR
    self.dtr = value
  File "/home/xtrinch/.platformio/penv/lib/python3.8/site-packages/serial/serialutil.py", line 473, in dtr
    self._update_dtr_state()
  File "/home/xtrinch/.platformio/penv/lib/python3.8/site-packages/serial/serialposix.py", line 715, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIC, TIOCM_DTR_str)
OSError: [Errno 71] Protocol error
```

```
xtrinch@elizabeta:~/repos/esp32s2-bme280-webusb$ /home/xtrinch/.platformio/penv/bin/esptool.py -h
esptool.py v3.1 - ESP8266 ROM Bootloader Utility
```

radimkarnis, 2021-07-30:
Hello @xtrinch!

Thank you for the detailed issue report. 
This is not an esptool issue and this should be fixed in the used USB stack (and [recently was!](https://github.com/chegewara/EspTinyUSB/commit/ca4e924ad0341bde28af0f1e7fad2772bfd60e08)). @chegewara explains his insights in this thread https://github.com/chegewara/EspTinyUSB/issues/43, I would recommend you to finish solving your issue there.

I will close this one, feel free to reopen if you think we should address this here.

Radim

closed 2021-07-30 by radimkarnis

xtrinch, 2021-07-31:
Thanks, I was actualy just hoping to gain some insight into what this error is from the esptool side.

As far as I can see it fails at the fnctl.ioctl system call when trying to set the DTR, so it hasn't actually had the chance to read anything yet. What are the possible reasons for that?

Since my usb device has 2 in interfaces (one CDC class and one vendor class), how does esptool know which one to pick?