espressif/esptool #648: Unable to flash ESP32-WROVER-E

persan666, 2021-08-03:
# Operating system 

Windows10

# Python version 

Python 3.8.7

# What Chip

ESP32-WROVER-E

# What development board or other hardware is the chip attached to

Custom hardware

# Is anything else attached to the development board, except for the serial flasher connections?

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

ESP-IDF Command prompt

# Full esptool.py command line that was run:

idf.py --port COM3 -b 921600 build app-flash monitor

# Full output from esptool.py

C:\source\dev\esp\INWEN>idf.py --port COM3 -b 921600 build app-flash monitor
Executing action: all (aliases: build)
Running ninja in directory c:\source\dev\esp\inwen\build
Executing "ninja all"...
[1/3] Performing build step for 'bootloader'
ninja: no work to do.
Executing action: app-flash
Running ninja in directory c:\source\dev\esp\inwen\build
Executing "ninja app-flash"...
[0/1] cmd.exe /C "cd /D C:\source\dev\esp\esp-idf\components\esptool_py && C:\Users\persan\.espressif\tools\c...ORY="C:/source/dev/esp/INWEN/build" -P C:/source/dev/esp/esp-idf/components/esptool_py/run_serial_tool.cmake"
esptool.py esp32 -p COM3 -b 921600 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 16MB 0x10000 INWEN.bin
esptool.py v3.1-dev
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
CMake Error at run_serial_tool.cmake:50 (message):
  C:/Users/persan/.espressif/python_env/idf4.3_py3.8_env/Scripts/python.exe
  C:/source/dev/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip
  esp32 failed

FAILED: esp-idf/esptool_py/CMakeFiles/app-flash
cmd.exe /C "cd /D C:\source\dev\esp\esp-idf\components\esptool_py && C:\Users\persan\.espressif\tools\cmake\3.16.4\bin\cmake.exe -D IDF_PATH="C:/source/dev/esp/esp-idf" -D SERIAL_TOOL="C:/Users/persan/.espressif/python_env/idf4.3_py3.8_env/Scripts/python.exe C:/source/dev/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @app-flash_args" -D WORKING_DIRECTORY="C:/source/dev/esp/INWEN/build" -P C:/source/dev/esp/esp-idf/components/esptool_py/run_serial_tool.cmake"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

# Do you have any other information from investigating this?
I am using custom hardware and have copied the USB-to-Uart circuit with the transistors for EN and IO0 from the dev-kit.
I have successfully used this tool for several years with both WROOM chips and earlier versions of WROVER.
Now when using the new ESP32-WROVER-E it wont flash. I have to force the IO0 pin to low to enter bootloader.

When analyzing the signals with a oscilloscope, one can see that the signals EN and IO0 will not be low at the same time and that EN do not go high during IO0 low. From the schematic of the Espressif Dev-kit V4 the hardware is the same as earlier versions. As I understand it, it prevents EN and IO0 to be low at the same time. Isn't that what we want? And when both signals are low we then raise EN to enter bootloader? Something is different in the boot ROM code in ESP32-WROVER-E from earlier versions.
I wonder how it could have worked before now that I know how the signals behaves.
[image]

Sarah-C, 2021-08-08:
Same here!

I'm using the M5Stack Core2 board... the Core Grey I have works fine.

dobairoland, 2021-08-09:
Hi @persan666. Do you have a capacitor between EN and GND? That is also necessary for the reset circuitry to work.

persan666, 2021-08-12:
> 
> 
> Hi @persan666. Do you have a capacitor between EN and GND? That is also necessary for the reset circuitry to work.

@dobairoland I noticed that just now when looking at the devkit schematic and that was exactly how we solved it:)
Thanks for pointing that out!

closed 2021-08-12 by persan666