espressif/esptool #651: Wiki

chegewara, 2021-08-13:
Any chance to update this wiki page for new "family members":
https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection

dobairoland, 2021-08-16:
Hi @chegewara. Thank you, you are right. We are planning to improve the documentation and removing those old Wiki pages. This suggestion will be implemented right after that.

chegewara, 2021-11-24:
Why?
I dont want to be mean, but my request was to add info about other chips into wiki, not to delete wiki.
Now we dont have info about boot mode and bootstrap pins even for old esp32, unless i am missing something.

radimkarnis, 2021-11-24:
Hi @chegewara,
as @dobairoland mentioned:

> We are planning to improve the documentation and remove those old Wiki pages. This suggestion will be implemented right after that.

We have launched new esptool docs and moved all the information there. This way it is not scattered across multiple different places like the readme, wiki, and PyPI description.

See https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html.

I am currently updating the new docs with requests from users, including this one.

chegewara, 2021-11-24:
Thanks, i saw new docs, but i could not find that page.

closed 2021-11-30 by espressif-bot

chegewara, 2021-11-30:
The same info we can get from datasheet, which actually is part of this doc for other chips.
> As well as the above mentioned pins, other ones influence the serial bootloader, please consult the ESP32-S2 Datasheet, section “Strapping Pins”.

What we are missing is this section:
https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#boot-mode-message

so we could know exactly which pins are in the wrong state during boot.

Thanks

radimkarnis, 2021-12-01:
I would advise you to open a new feature request and describe the desired changes there, it would make tracking this easier. Thank you!