espressif/esptool #652: Where can I downdown version 2.6 executable file

ianho7, 2021-08-17:
Hi guys, I met some problems while using 3.1 release executable file, after testing, version 2.6 works well, but I have only the  executable file of windows platform, and I also need it for Mac OS.However, the release page only provides version 3.1 executable files, so could you provide version 2.6 executable file or download link for me?🙏🙏🙏

# Operating system
Mac OS 11.4

# Python version
2.7.16

# What Chip
ESP32

radimkarnis, 2021-08-17:
Hello @ianho7,

could you please describe your problems when switching from v2.6 to v3.1? It would help us to track any possible regressions.

We’ve started releasing executables in the esptool project just recently (introduced in v3.1, to be exact). These are built automatically with `PyInstaller` by GitHub Actions. We do not plan to do this with older releases.

It is possible to checkout any older version and build the executable for your system manually. To do so, just install `PyInstaller` and run `pyinstaller --distpath . -F esptool.py` in the esptool folder.

Let me know if you find this solution helpful or not.

Radim

radimkarnis, 2021-08-30:
Closing due to inactivity. Feel free to reopen.

closed 2021-08-30 by radimkarnis