espressif/esptool #655: "Packet content transfer stopped" flash failed with USB-CDC serial port under windows

wuxx, 2021-08-29:
* Operating system: Windows 10
* Python version: (`python -V` to check this)
Python 3.8.0
# Full esptool.py command line as run:
esptool.py -t --chip esp32c3 -p COM319 -b 115200 --before=default_reset --after=hard_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 F:\oss\nanoESP32-C3\demo\flash_image_blink_1000.bin 
esptool.py version
esptool.py v3.2-dev
3.2-dev

# Full output from esptool.py
F:\20210827\esptool>esptool.py -t --chip esp32c3 -p COM319 -b 115200 --before=default_reset --after=hard_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 F:\oss\nanoESP32-C3\demo\flash_image_blink_1000.bin 
esptool.py v3.2-dev
Serial port COM319
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.103 No serial data received.
.TRACE +0.067 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.005 Read 1 bytes: c0
TRACE +0.000 Read 6 bytes: 010804000707
TRACE +0.001 Read 1 bytes: 12
TRACE +0.000 Read 10 bytes: 2000000000c0c0010804
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.011 Read 28 bytes:
    0007071220000000 00c0c00100c0c001 | .... ...........
    0800c0c001080400 07071220         | ...........
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.011 Received full packet: 0100
TRACE +0.000 Received full packet: 010800
TRACE +3.011 Packet content transfer stopped (received 8 bytes)
.TRACE +0.065 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.005 Read 1 bytes: c0
TRACE +0.000 Read 7 bytes: 01080400070712
TRACE +0.001 Read 1 bytes: 20
TRACE +0.000 Read 11 bytes: 00000000c0c00108040007
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.011 Read 28 bytes:
    07122000000000c0 c0010804c0c00108 | .. .............
    04c0010804000707 12200000         | ......... ..
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.011 Received full packet: 010804
TRACE +0.000 Received full packet: 010804
TRACE +0.000 Read invalid data:
    07122000000000c0 c0010804c0c00108 | .. .............
    04c0010804000707 12200000         | ......... ..
TRACE +0.000 Remaining data in serial buffer:
.TRACE +0.062 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
[... 166 lines cut ...]
    val, _ = self.command()
  File "F:\20210827\esptool\esptool.py", line 452, in command
    p = self.read()
  File "F:\20210827\esptool\esptool.py", line 395, in read
    return next(self._slip_reader)
  File "F:\20210827\esptool\esptool.py", line 3287, in slip_reader
    read_bytes = port.read(1 if waiting == 0 else waiting)
  File "C:\Users\Lenovo\AppData\Local\Programs\Python\Python38\lib\site-packages\serial\serialwin32.py", line 286, in read
    result_ok = win32.GetOverlappedResult(
KeyboardInterrupt
^C
F:\20210827\esptool>

# What is the expected behaviour?

succ expected 

# Do you have any other information from investigating this?
already check uart with logic analyzer, the ESP_SYNC send succ, and the response of chip is also ok, but the response in esptool.py recv is incomplete, so ESP_SYNC response always fail.

test with the nanoESP32-C3 (with a USB-CDC serial port on-board)
https://github.com/wuxx/nanoESP32-C3

it work well on linux, but always failed on windows, and after exploration and test, found that if add a  sleep after ESP_SYNC frame, the windows will also success
https://github.com/wuxx/esptool/commit/723810dac013107cd95f70686e394b2f6a5fc247

radimkarnis, 2021-08-31:
Hello @wuxx,
you are mentioning a `USB-CDC serial port onboard`, then `ESPLink with USB-to-Serial capability`. This is not a very common programmer and is probably causing the timing problems.

Would you mind giving more information about how this works (google reveals multiple unrelated `ESPLink` projects) and share the logic analyzer traces with us?

wuxx, 2021-08-31:
hello @radimkarnis , the ESPLink is based on open source project ARMMbed DAPLink, the source code here https://github.com/ARMmbed/DAPLink, it support USB-HID JTAG and USB-CDC Serial, and I only modify the drag-n-drop related code, the USB-CDC related code was not modified, and here is the uart traces, need opened with pulseview (https://sigrok.org/)
[usb_cdc_fail.zip](https://github.com/espressif/esptool/files/7082374/usb_cdc_fail.zip)

radimkarnis, 2021-09-06:
Hello @wuxx,
thank you for the clarification and provided traces. It seems like the ESPLink is not fast enough to pass through data during the sync process. Just to check - does anything change if you increase the `SYNC_TIMEOUT` value?

wuxx, 2021-09-07:
hello, @radimkarnis 
In fact, the first thing I tried was to modify SYNC_TIMEOUT, try modify it to 1 and other values (origin is 0.1), nothing changes.

radimkarnis, 2021-09-10:
One last thing, have you tried lowering the baudrate? (`-b 115200`)

To be honest, it is not a viable solution to increase the time between individual sync commands for all esptool users. If an unusual solution is used and proper timing during communication can't be guaranteed, a creative patch has to be issued - see [this example](https://github.com/espressif/esptool/blob/master/esp_rfc2217_server.py#L88).

wuxx, 2021-09-12:
understand it, low baudrate already tried, still nothing changed... I think this relate to how the windows process with USB-CDC serial data. a low-level problem

radimkarnis, 2021-09-15:
Hi @wuxx,
sorry this is taking so long. I am trying to come with an acceptable fix for everyone. 

Please try applying the patch I suggested here: https://github.com/wuxx/esptool/commit/723810dac013107cd95f70686e394b2f6a5fc247#commitcomment-56495707. Does waiting for the output buffer to empty before continuing help?

wuxx, 2021-09-19:
hello, @radimkarnis, I tried this patch, still fail on SYNC request
F:\20210827\esptool-master>esptool.py -p COM319 flash_id
esptool.py v3.2-dev
Serial port COM319
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: Invalid head of packet (0x00)

F:\20210827\esptool-master>

I will try to capture and analyze the USB event when have some free time.

radimkarnis, 2021-09-20:
@wuxx thank you, that would be great. Other than that, I am currently out of ideas

radimkarnis, 2021-09-20:
@wuxx, could you please provide your PySerial version so we can rule out as many factors as possible?

wuxx, 2021-09-21:
hello, @radimkarnis , the pyserial version is 3.5. check with following command
>>> import serial
>>> help(serial)
Help on package serial:

NAME
    serial

DESCRIPTION
    # This is a wrapper module for different platform implementations
    #
    # This file is part of pySerial. https://github.com/pyserial/pyserial
    # (C) 2001-2020 Chris Liechti <cliechti@gmx.net>
    #
    # SPDX-License-Identifier:    BSD-3-Clause

PACKAGE CONTENTS
    __main__
    rfc2217
    rs485
    serialcli
    serialjava
    serialposix
    serialutil
    serialwin32
    threaded (package)
    tools (package)
    urlhandler (package)
    win32

FUNCTIONS
    serial_for_url(url, *args, **kwargs)
        Get an instance of the Serial class, depending on port/url. The port is not
        opened when the keyword parameter 'do_not_open' is true, by default it
        is. All other parameters are directly passed to the __init__ method when
        the port is instantiated.

        The list of package names that is searched for protocol handlers is kept in
        ``protocol_handler_packages``.

        e.g. we want to support a URL ``foobar://``. A module
        ``my_handlers.protocol_foobar`` is provided by the user. Then
        ``protocol_handler_packages.append("my_handlers")`` would extend the search
        path so that ``serial_for_url("foobar://"))`` would work.

DATA
    CR = b'\r'
    EIGHTBITS = 8
    FIVEBITS = 5
    LF = b'\n'
    PARITY_EVEN = 'E'
    PARITY_MARK = 'M'
    PARITY_NAMES = {'E': 'Even', 'M': 'Mark', 'N': 'None', 'O': 'Odd', 'S'...
    PARITY_NONE = 'N'
    PARITY_ODD = 'O'
    PARITY_SPACE = 'S'
    SEVENBITS = 7
    SIXBITS = 6
    STOPBITS_ONE = 1
    STOPBITS_ONE_POINT_FIVE = 1.5
    STOPBITS_TWO = 2
    VERSION = '3.5'
    XOFF = b'\x13'
    XON = b'\x11'
    absolute_import = _Feature((2, 5, 0, 'alpha', 1), (3, 0, 0, 'alpha', 0...
    basestring = (<class 'str'>,)
    protocol_handler_packages = ['serial.urlhandler']

VERSION
    3.5

FILE
    c:\users\lenovo\appdata\local\packages\pythonsoftwarefoundation.python.3.9_qbz5n2kfra8p0\localcache\local-packages\python39\site-packages\serial\__init__.py

>>>

wuxx, 2021-10-05:
[Uploading decoder-fail.txt…](
[decoder-succ.txt](https://github.com/espressif/esptool/files/7282612/decoder-succ.txt)
)

wuxx, 2021-10-05:
[20211003.zip](https://github.com/espressif/esptool/files/7282614/20211003.zip)
please check  the decoder-fail.txt and decoder-succ.txt, just search "C0" to locate the esp frame.
code difference: https://github.com/wuxx/esptool/commit/723810dac013107cd95f70686e394b2f6a5fc247
notice that the esptool.py (or pyserial) send some strange USB SETUP packet after the esp SYNC frame sent in decode-fail.txt.

wuxx, 2021-10-05:
the strange packets are here, I check the DAPLink code, it's host request  to get the cdc port settings.
relate to function usbd_cdc_acm.c:USBD_CDC_ACM_GetLineCoding()
https://github.com/ARMmbed/DAPLink/blob/ce5e0d33be585e3bdf14fb908f83a6d309066c15/source/usb/cdc/usbd_cdc_acm.c#L247

286818,5920320320.00000000000000000000,SETUP ADDR 32 EP 0
286819,5920323580.00000000000000000000,DATA0 [ 21 20 00 00 01 00 07 00 ]
286820,5920332060.00000000000000000000,ACK
286821,5920351060.00000000000000000000,SOF 1137
286822,5920354320.00000000000000000000,IN ADDR 32 EP 1
286823,5920357460.00000000000000000000,NAK
286824,5920359400.00000000000000000000,IN ADDR 32 EP 4
286825,5920362560.00000000000000000000,NAK
286826,5920368380.00000000000000000000,OUT ADDR 32 EP 0
286827,5920371660.00000000000000000000,DATA1 [ 00 C2 01 00 00 00 08 ]

wuxx, 2021-10-05:
ok... it may be relate to some junk print log in my ESPLink firmware code, after delete these logs, the SYNC frame can response succ, though still after about few second try SYNC, I will try again and close this issue, thanks

wuxx, 2021-10-06:
After repeated testing, it's better than before, but still easy to fail...
I think it's related to low-level of the pyserial cdc process under windows, in linux everything is ok.

radimkarnis, 2021-10-06:
Thank you @wuxx for the thorough investigation, nice job locating the issue! This is certainly an interesting one.

I believe it is now safe to say this is not an esptool issue and the problem should be fixed elsewhere. If you agree and don't have any more ideas, we can close this ticket.

closed 2021-10-10 by wuxx

alexeyrusnak, 2023-05-12:
> ok... now found a workaround to match the strange timing of esptool.py with USB-CDC. close the issue thanks radimkarnis ~!

Hi
Please say what workaround you found? I have the same problem

wuxx, 2023-05-13:
I add some delay before send the data, this can temporarily solve the problem, however it will affect performance

alexeyrusnak, 2023-05-14:
> I add some delay before send the data, this can temporarily solve the problem, however it will affect performance

In esptool.py? Can you show code please?

wuxx, 2023-05-15:
you can modify the esptool.py like this
https://github.com/wuxx/esptool/commit/723810dac013107cd95f70686e394b2f6a5fc247
or modify the UART-CDC code
[image]