espressif/esptool #656: Automatic reset not working. Only able to connect with --before no_reset

MrJake222, 2021-09-06:
Hi,

I'm on linux and experiencing weird behavior related to auto-reset function. The esptool times out and doesn't receive the data the module is clearly sending.
> [image]
[image]

Carrying out the same procedure with `--before no_reset` and reseting the module manually works (logic analyzer graph stays the same). But it connects after several packets received, and starts communicating after the longer one.
> [image]

MrJake222, 2021-09-06:
I've fixed the issue by editing the number of tries in `_connect_attempt` ([line 609](https://github.com/espressif/esptool/blob/master/esptool.py#L609)) from `5` to for ex. `8`. It might be worth a config option.

Compizfox, 2021-09-06:
Same issue here: worked fine in the past, but suddenly couldn't flash my NodeMCU boards anymore, whatever I tried. I've been struggling with this for over a week now. Thankfully, I found this issue now!

I can confirm that `--before no_reset` and resetting the board manually is the only thing that works.

FWIW, I'm also on Arch Linux.

MrJake222, 2021-09-06:
I suspect some kernel/driver update since it's a rolling-release distro.

dobairoland, 2021-09-07:
@MrJake222 I see in your PR that you are on Archlinux and using NodeMCU. You could be affected by https://github.com/espressif/esptool/issues/653. Archlinux at this moment still runs on an affected kernel version. The fixed version is in "testing".

radimkarnis, 2021-09-07:
Hello @Compizfox and @MrJake222,
could you please help and confirm the source of the problem? See this issue explaining a bug in the kernel: https://github.com/espressif/esptool/issues/653

MrJake222, 2021-09-07:
I think I'm affected. The thing happened after two months (update probably happened) and I'm at linux `5.10.60-1-lts`.

dd871, 2021-09-07:
Same issue here with Arch on 5.13.10. Was starting to pull my hairs out until I found this. No matter what I do, always getting a timeout but working perfectly fine on Windows (albeit it seems to indeed work with the --before no_reset parameter as workaround).

MrJake222, 2021-09-07:
I've resoldered my board yesterday because It's the cheapest one. But to no avail. People mentioned downgrading kernel as a solution.

0x0aa, 2021-09-10:
Linux 5.10.62 has the [fix](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/usb/serial?h=v5.10.62&id=8437e07c370fc4dfcaf639b4372e16fc5f32acf5) for #653 and now there's linux-lts 5.10.63 available in the Arch Linux core repository.

radimkarnis, 2021-09-13:
I am closing this issue and the linked PR as there is a confirmed fix in https://github.com/espressif/esptool/issues/653. Feel free to reopen if you think further discussion is needed.

closed 2021-09-13 by radimkarnis

adriafp, 2021-09-30:
Up to the version 5.11.0-36 was working fine for me, but upgraded to 5.11.0-37 and stop working.

ferdinand0815, 2021-10-03:
> Up to the version 5.11.0-36 was working fine for me, but upgraded to 5.11.0-37 and stop working.

Same here, stopped working with update from 5.11.0-36 to 5.11.0-37. Also 5.4.0-84 works but not 5.4.0-88.

MrJake222, 2021-10-16:
> Linux 5.10.62 has the [fix](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/usb/serial?h=v5.10.62&id=8437e07c370fc4dfcaf639b4372e16fc5f32acf5) for #653
Or update.

Doubletop12, 2021-11-03:
> 
> 
> ```
> upload_speed = 115200
> upload_resetmethod = soft_reset
> ```
> 
> Seemed to work for me once, but only once.
> 
> The problem has been identified as being in kernel 5.11.0-37. Reverting to -36 supposedly works.
> 
> What worked for me was using the "Advanced boot options" screen during boot. If there is an older kernel listed, select it. No fuss, no mess. I plan to make the older kernel the default with Grub Customizer.

It worked for me, but also only once. The second run fails with "A fatal error occurred: Invalid head of packet (0x06)" I guessed that was because the UART was outputting data and missed the soft reset command. So I commented out

`;upload_resetmethod = soft_reset` and ran it again that then fails to connect (but has done a reset)

Then remove the comment back to

`upload_resetmethod = soft_reset` and it works again.

A real cludge but saves having to go to the hardware and  press the reset button. Especially if you are working remotely.

Pete