espressif/esptool #658: read_flash hang with Lolin S2 Mini ESP32-S2FNR2

durapensa, 2021-09-07:
# Operating system
Ubuntu Impish Indri (development branch) Linux 5.14.0-051400-generic

# Python version
Python 3.9.7

# What Chip
ESP32-S2FNR2

# What development board or other hardware is the chip attached to
Lolin S2 Mini https://www.wemos.cc/en/latest/s2/s2_mini.html

# Is anything else attached to the development board, except for the serial flasher connections?
NO

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
NO

# Full esptool.py command line that was run:
`esptool.py --chip esp32s2 --port /dev/ttyACM0 --baud 115200 read_flash 0 0x400000 lolin_s2_mini_stock_firmware_4M.bin`

# Full output from esptool.py
```
esptool.py v3.2-dev
Serial port /dev/ttyACM1
Connecting......
Chip is ESP32-S2FNR2
Features: WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V1
Crystal is 40MHz
MAC: 7c:df:a1:96:85:f2
Uploading stub...
Running stub...
Stub running...
20480 (0 %)
```
hang; no further output

# Do you have any other information from investigating this?

Board with factory-flashed firmware (MicroPython). When flashed with TinyUF2 & CircuitPython, progress reaches 458752 (10 %) and hangs.

Same failure with multiple Lolin S2 mini boards, USB C cables, USB ports, Baud (down to 9600)

Same failure with esptool 3.1 from https://pypi.org/project/esptool/

# Is there any other information you can think of which will help us reproduce this problem?
TBD

radimkarnis, 2021-09-08:
Hello @durapensa,
could you please run the same command again with the `--trace` flag and post the resulting log?

Does the same thing happen if you try to reflash the board with stock firmware according to [this tutorial](https://www.wemos.cc/en/latest/tutorials/s2/get_started_with_micropython_s2.html)?

Radim

durapensa, 2021-09-08:
Reflashing the board works fine (flashed MicroPython 1.17 using the tutorial, and TinyUF2 0.5.2 / CircuitPython 7.0.0-rc1 using Adafruit's WebSerial ESPTool).

Output of `esptool.py --chip esp32s2 --port /dev/ttyACM0 --baud 115200 --trace read_flash 0 0x400000 lolin_s2_mini_circuitpython_firmware_4M.bin` attached.
[esptool.out.txt](https://github.com/espressif/esptool/files/7129134/esptool.out.txt)

radimkarnis, 2021-09-13:
This output log shows that most of the firmware image has been successfully read from the flash. Does the command always hang at the same spot (percentage)? Does it sometimes succeed if you try multiple times?

Your command reads the entire contents of the 4MB flash. You can also try to read just a single partition (like the factory image), but you'll need to determine the partition offset and size and pass these to read_flash instead. Usually, this information is printed to the serial console by the bootloader.

durapensa, 2021-09-13:
The command always hangs at the same spot and never succeeds, even after multiple tries.

I'll look for the partition information on the serial console and try to `read_flash` just the factory image (Wemos' custom build of MicroPython in this case) using the partition offset and size. Thanks for your help!

durapensa, 2021-09-13:
This board does not break out TX/RX pins from the SoC but instead uses USB CDC for the serial console (and flashing), so early messages from the bootloader are not displayed, i.e. the first and only thing displayed on a hard reset, while USB-connected & viewing the output from /dev/ttyACM0 using tio, is MicroPython's REPL prompt `>>> ` (with "Disconnected" and "Connected" messages from tio after the RESET button is pressed and released). 

I went looking and found this method to read the partition table directly from flash:

`esptool.py --chip esp32s2 read_flash 0x8000 0xc00 ptable.img`
`gen_esp32part.py ptable.img`

```
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1536K,
vfs,data,fat,0x200000,2M,
```

So tried `esptool.py --chip esp32s2 --port /dev/ttyACM0 --baud 115200 read_flash 0x10000 0x180000 lolin_s2_mini_micropython_firmware.bin`, and it hangs every time at 282624 (17 %).

radimkarnis, 2021-09-14:
Nice detective work, thanks for trying! 

I have tried to replicate your problem - flashed an ESP32-S2 based devboard via USB-CDC with micropython and then read the whole flash. This worked all just fine with no errors.

Two more ideas I have: 
- try with the `--no-stub` argument
- try reading the flash "per partes", trying to download smaller arbitrary chunks

durapensa, 2021-09-14:
`--no-stub` argument worked, thank you! (silly slow, but it worked). I'm happy to try getting further debugging info if this would be helpful for improving esptool.py.

radimkarnis, 2021-09-21:
The only thing I could ask you to try is to switch the OS. If you have access to a Windows or Mac machine, I would be grateful for a confirmation that this is/isn't related to OS/drivers. Other than that, I can't reproduce the issue and would need to order the board for further investigation.

radimkarnis, 2021-09-24:
Closing this, the issue can't be reproduced and a working solution has been provided.

closed 2021-09-24 by radimkarnis