espressif/esptool #659: what does this mean "[directory] which is not on PATH" any why in CAPS?

mlohdael, 2021-09-11:
* Operating system: Debian
* Python version: (`python -V` to check this)
2.7
# Full esptool.py command line as run:
N/A

# Full output from esptool.py

this is the terminal response from "pip install esptool"
[pipinstallesptool.txt](https://github.com/espressif/esptool/files/7148650/pipinstallesptool.txt)

# What is the expected behaviour?
install script with no warnings

# Do you have any other information from investigating this?
as you see everything seems to have installed fine, but i dont know what 

# Is there any other information you can think of which will help us reproduce this problem?

i am aware what a "file path" is but i dont know why it's capitalized abut it also uses the term "directory".
just want to make sure this wont end up doing bricking my OS in some way. i am bad at that.

radimkarnis, 2021-09-13:
Hello @mlohdael,

The PATH variable is an environment variable that contains an ordered list of paths that Linux will search for executables (python modules and packages) when running a command. 

In your case, the `/home/pi/.local/bin` folders contains the `pyserial-miniterm` and `pyserial-ports` python modules, but isn't on PATH. There is nothing to worry about! If esptool works, you don't have to change anything. 

Let me know if your installation works or not.

radimkarnis, 2021-09-23:
I am closing this, it seems like it is not relevant anymore.

closed 2021-09-23 by radimkarnis