espressif/esptool #660: esptool.py not detecting hardware via USB

andogq, 2021-09-14:
# Operating system

`Linux 5.10.61-1-MANJARO #1 SMP PREEMPT Thu Aug 26 20:36:54 UTC 2021`

# Python version

`Python 3.9.6`

# What Chip

ESP8266

# What development board or other hardware is the chip attached to

WeMos D1 R2

# Is anything else attached to the development board, except for the serial flasher connections?

No connections to development board aside from USB

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

No IDE, running from command prompt

# Full esptool.py command line that was run:

`esptool.py --chip esp8266 chip_id`

(no commands work, this is the one I've been used to test though)

# Full output from esptool.py

```
esptool.py v3.1
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/ttyUSB0 failed to connect: Failed to connect to ESP8266: Timed out waiting for packet header

A fatal error occurred: Could not connect to an Espressif device on any of the 1 available serial ports.
```

# Do you have any other information from investigating this?

After searching through the Wiki I found the `--before` [reset option](https://github.com/espressif/esptool/wiki/Advanced-Options#reset-before-operation), and running the command (and any other command) with `--before usb_reset` works as expected.

The Wiki states that Esptool should be able to detect the USB connection (I presume this applies to this board, as it uses a CH340 serial converter) and it appears this check happens on [line 659 of `esptool.py`](https://github.com/espressif/esptool/blob/master/esptool.py#L659) by checking the PID of the device to the `USB_JTAG_SERIAL_PID` constant which is set as `0x1001` on [line 305](https://github.com/espressif/esptool/blob/master/esptool.py#L305). Changing this value to `7523` (the PID of the CH340 serial converter) and running the command without the `--before` flag works as expected.

I'm unsure if this is the actual problem (due to my unfamiliarity with these tools and hardware) or if it's just a coincidence and something else is wrong elsewhere. This 'fix' works with multiple boards (same model, I own a few) and USB cables ect.

Any clarity on what's happening and where/what the original PID of `0x1001` refers to would be greatly appreciated.

radimkarnis, 2021-09-15:
Hello @andogq,
thank you for the report and your efforts to debug this.
At the moment, there is a known bug in Linux kernel affecting C340 USB-serial bridges. See https://github.com/espressif/esptool/issues/653 for more information.

To clarify: some of Espressif's new chips (ESP32-C3 and ESP32-S3) integrate a USB Serial/JTAG controller. This peripheral allows CDC-ACM virtual serial port and JTAG adapter functionality using only two pins. Esptool can use this to communicate and flash the chip. The only caveat is that it requires a slightly different reset sequence. To decide which reset sequence needs to be triggered, esptool detects the PID of the USB Serial/JTAG controller (which is `0x1001`!).

Your solution is a happy coincidence of using a slightly different reset sequence to cover a completely unrelated bug in Linux kernel 😄 .

Please, let me know if your problem gets resolved following the linked GH issue.

Radim

andogq, 2021-09-15:
Thanks for your quick reply!

I can confirm that upgrading my kernel to 5.14 completely fixed this issue. I had a feeling I was going to be a bit off haha.

Thank you again for your help!

closed 2021-09-15 by andogq