espressif/esptool #661: Issue with misaligned padding from elf2image

FrozenDroid, 2021-09-16:
* Operating system: macOS 
* Python version: 3.8.2

# Full esptool.py command line as run:
`esptool.py --chip esp32c3 elf2image target/riscv32imc-esp-espidf/debug/rust-esp32-std-hello`

# Full output from esptool.py
```
 ▎ € esptool.py --chip esp32c3 image_info target/riscv32imc-esp-espidf/debug/rust-esp32-std-hello.bin
esptool.py v3.1-dev
Image version: 1
Entry point: 403802e6
8 segments

Segment 1: len 0x118c8 load 0x3c030020 file_offs 0x00000018 [DROM]
Segment 2: len 0x01bf8 load 0x3fc8ae00 file_offs 0x000118e8 [DRAM,BYTE_ACCESSIBLE]
Segment 3: len 0x0acbc load 0x40380000 file_offs 0x000134e8 [IRAM]
Segment 4: len 0x00010 load 0x50000000 file_offs 0x0001e1ac [RTC_IRAM,RTC_DRAM]
Segment 5: len 0x01e4c load 0x00000000 file_offs 0x0001e1c4 [PADDING]
Segment 6: len 0x2a4fc load 0x42000020 file_offs 0x00020018 [IROM]
Segment 7: len 0x0ffee load 0x00000000 file_offs 0x0004a51c [PADDING]
Segment 8: len 0x00980 load 0x4202a51a file_offs 0x0005a512 [IROM]
Checksum: 5e (valid)
Validation Hash: b4d49c7d256de8de958ada81d50fc9c69f6f37c7c1f146a76e6067fc0a6f85ae (valid)
```

# What is the expected behaviour?
Segment 7 should be 4-byte aligned, but it is not. This causes an [error](https://github.com/espressif/esp-idf/blob/master/components/bootloader_support/src/esp_image_format.c#L646) on the chip: 
```
esp_image: invalid segment length 0xffee
```

# Do you have any other information from investigating this?

https://github.com/ivmarkov/rust-esp32-std-hello/issues/14

It seems that perhaps, in [this line](https://github.com/espressif/esptool/blob/master/esptool.py#L2538), padding should also be aligned?

# Is there any other information you can think of which will help us reproduce this problem?
From the repo in the linked issue above, commenting out all the code in the main and putting a simple "Hello world" print compiles a binary, which, if used with `elf2image`, causes this erroneous image.

FrozenDroid, 2021-09-16:
Additional info, I have tried this with the release 3.1 version and the latest dev version, but this happens with both. According to @ivmarkov this issue does not happen with ESP32, but this might be due to the fact it doesn't even generate any padding there.

FrozenDroid, 2021-09-22:
@igrr @radimkarnis 
[rust-esp32-std-hello.zip](https://github.com/espressif/esptool/files/7210892/rust-esp32-std-hello.zip)
Attached you'll find a zip with the requested ELF file zipped.

radimkarnis, 2021-09-22:
Thank you @FrozenDroid for the ELF file, can you please confirm it is the right one?

I can't reproduce your output, `image_info` reports no padding segments (similar to https://github.com/ivmarkov/rust-esp32-std-hello/issues/14#issuecomment-921018208):
```
➜ esptool.py --chip esp32c3 elf2image rust-esp32-std-hello
esptool.py v3.2-dev
Merged 1 ELF section

➜ esptool.py --chip esp32c3 image_info rust-esp32-std-hello.bin
esptool.py v3.2-dev
Image version: 1
Entry point: 403802e6
6 segments

Segment 1: len 0x29200 load 0x3c0c0020 file_offs 0x00000018 [DROM]
Segment 2: len 0x03500 load 0x3fc8e200 file_offs 0x00029220 [DRAM,BYTE_ACCESSIBLE]
Segment 3: len 0x038e8 load 0x40380000 file_offs 0x0002c728 [IRAM]
Segment 4: len 0xb15b8 load 0x42000020 file_offs 0x00030018 [IROM]
Segment 5: len 0x0a7a0 load 0x403838e8 file_offs 0x000e15d8 [IRAM]
Segment 6: len 0x00010 load 0x50000000 file_offs 0x000ebd80 [RTC_IRAM,RTC_DRAM]
Checksum: 05 (valid)
Validation Hash: a5b363e36e395cc3a65f24e5a103631a9d44c3f1d6a22b3acc59bc8a77be3bb4 (valid)
```

FrozenDroid, 2021-09-22:
@radimkarnis 
Ah, my apologies, I must have sent an ELF of the full example. Try the attached ELF file which I just compiled anew, which I can confirm still outputs the erroneous padding section. 
[rust-esp32-std-hello.zip](https://github.com/espressif/esptool/files/7211487/rust-esp32-std-hello.zip)

radimkarnis, 2021-09-27:
Thank you @FrozenDroid, I can confirm I am now getting the same erroneous output as you.
At first glance, I can't seem to locate a bug in esptool. I will look into this more when we get a confirmation the issue is not with the generated elf/linker script (as mentioned here: https://github.com/ivmarkov/rust-esp32-std-hello/issues/14#issuecomment-925926890)

fkohlgrueber, 2021-11-16:
Is there any update on this? I'm running into the problem quite frequently. Is there a workaround other than inserting more code until it works? Let me know if I can help. Thanks in advance!

radimkarnis, 2021-11-18:
I am sorry @fkohlgrueber, I did no investigation on the esptool side. I am still not sure if this is build script / build related or not.

dobairoland, 2022-01-11:
Segment 8 is `.rwtext` which was created by mistake and is not generated usually. The alignment of this segment is not controlled by ESP-IDF.

The fix was submitted [here](https://github.com/esp-rs/esp-idf-hal/commit/ed6d8f783435bb033ea578a0058f1d875542db3a). See https://github.com/ivmarkov/rust-esp32-std-demo/issues/14 for more information.

closed 2022-01-11 by dobairoland