espressif/esptool #662: A fatal error occurred: MD5 of file does not match data in flash!

lennardsisik, 2021-09-21:
Hello im currently student and new in esp32
The binary is ESP32 MicroPython esp32-20210902-v1.17.bin
when I install this I get this error:

"C:\Program Files (x86)\Thonny\python.exe" -u -m esptool --port COM3 erase_flash
esptool.py v3.1
Serial port COM3
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 7c:9e:bd:45:90:c0
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.0s
Hard resetting via RTS pin...
Erasing done
------------------------------------

"C:\Program Files (x86)\Thonny\python.exe" -u -m esptool --port COM3 write_flash --flash_mode keep --flash_size detect 0x1000 C:/Users/CLIENT/Desktop/esp32-20210623-v1.16.bin
esptool.py v3.1
Serial port COM3
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 7c:9e:bd:45:90:c0
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00171fff...
Compressed 1510544 bytes to 976957...
Writing at 0x00001000... (1 %)
Writing at 0x000111e1... (3 %)
Writing at 0x00019c9f... (5 %)
Writing at 0x00022690... (6 %)
Writing at 0x00029089... (8 %)
Writing at 0x00031b4a... (10 %)
Writing at 0x0003d0df... (11 %)
Writing at 0x00047987... (13 %)
Writing at 0x00053b47... (15 %)
Writing at 0x000599d5... (16 %)
Writing at 0x0005f3db... (18 %)
Writing at 0x00064979... (20 %)
Writing at 0x000698ff... (21 %)
Writing at 0x0006e9e0... (23 %)
Writing at 0x00073e23... (25 %)
Writing at 0x00079127... (26 %)
Writing at 0x0007e140... (28 %)
Writing at 0x00082d19... (30 %)
Writing at 0x000886ca... (31 %)
Writing at 0x0008e80b... (33 %)
Writing at 0x00093c3c... (35 %)
Writing at 0x00098e11... (36 %)
Writing at 0x0009ef14... (38 %)
Writing at 0x000a44a4... (40 %)
Writing at 0x000a9a58... (41 %)
Writing at 0x000af8b4... (43 %)
Writing at 0x000b5add... (45 %)
Writing at 0x000bb950... (46 %)
Writing at 0x000c0995... (48 %)
Writing at 0x000c670c... (50 %)
Writing at 0x000cc267... (51 %)
Writing at 0x000d14bc... (53 %)
Writing at 0x000d6384... (55 %)
Writing at 0x000db342... (56 %)
Writing at 0x000e0572... (58 %)
Writing at 0x000e56f9... (60 %)
Writing at 0x000ead9b... (61 %)
Writing at 0x000f0916... (63 %)
Writing at 0x000f6209... (65 %)
Writing at 0x000faeee... (66 %)
Writing at 0x0010094c... (68 %)
Writing at 0x00105da0... (70 %)
Writing at 0x0010b1bd... (71 %)
Writing at 0x001108cb... (73 %)
Writing at 0x00115a6a... (75 %)
Writing at 0x0011af8d... (76 %)
Writing at 0x00120281... (78 %)
Writing at 0x00125957... (80 %)
Writing at 0x0012adc0... (81 %)
Writing at 0x001304ef... (83 %)
Writing at 0x00135d36... (85 %)
Writing at 0x0013b6ed... (86 %)
Writing at 0x00141383... (88 %)
Writing at 0x0014632c... (90 %)
Writing at 0x0014bec4... (91 %)
Writing at 0x00152a2c... (93 %)
Writing at 0x0015ca92... (95 %)
Writing at 0x00162ada... (96 %)
Writing at 0x001680de... (98 %)
Writing at 0x0016d422... (100 %)
Wrote 1510544 bytes (976957 compressed) at 0x00001000 in 86.9 seconds (effective 139.0 kbit/s)...
File  md5: c0ddf3b5f2eb47a495c1270f8e33ef5b
Flash md5: 98f7ea6cf80b36d9fa1b08b5ae85d9c5
MD5 of 0xFF is de783540df443b3deb667ac4e7ca8593

A fatal error occurred: MD5 of file does not match data in flash!

Write command returned with error code 2

how to get rid this?

radimkarnis, 2021-09-21:
Hi @lennardsisik,
After esptool wrote the new binary to the flash it read back the contents and it didn't match. So something was corrupted during the data transfer. 

Some of the causes of an unsuccessful flash are:

- A bad quality or too long USB cable, try a different one.
- Weak power supply or when there are more things sharing it (causing the chip to brown out). Provide sufficient power.
- Peripherals connected to GIPO pins, try disconnecting everything.
- Fast baud rate, you can try with `--baud 115200` instead `921600`.

Please let me know if any of these or our [troubleshooting guide](https://github.com/espressif/esptool#troubleshooting) helped!

Radim

lennardsisik, 2021-09-22:
i try different different usb cable but it still the same error and also no peripherals connected to GIPO pins and I try to change the baud rate to --baud 115200 and it still the same. it is possible I brick my esp32? but this is new

radimkarnis, 2021-09-22:
Don't worry, your board is not bricked, let's investigate more:

If you try more than once, does this `Flash md5: 98f7ea6cf80b36d9fa1b08b5ae85d9c5` change? This will help us to determine if the corruption happens randomly or not.

Does the command work with `--no-stub` option?

lennardsisik, 2021-09-22:
it didn't change. whant syntax i'll be use in --no-stub option?

lennardsisik, 2021-09-22:
[image]
i tried the --no-stub but the same issue

radimkarnis, 2021-09-23:
The md5 hash is the same, that means we can rule out serial noise / random data corruption. 
Can you please try flashing with no compression? (delete the `-z` option and use `-u`, try both with stub and no stub)

If this doesn't help, some of your flash might be write protected or defective.

lennardsisik, 2021-09-23:
it still the same error @radimkarnis

radimkarnis, 2021-09-23:
Ok, let's try to debug hardware flash chip-related problems:

- Please run these two commands:
`esptool.py read_flash_status`
`esptool.py flash_id`
(Please copy or screenshot the output and post it here.)

- If that all seems to go well, you can try:
`esptool.py write_flash_status --non-volatile 0`

Run these commands and check if flashing succeeds.
Thanks for your patience!

lennardsisik, 2021-09-23:
[image]
[image]
@radimkarnis

radimkarnis, 2021-09-24:
Can you please confirm if flashing works after running the last command (`esptool.py write_flash_status --non-volatile 0`)?

If not, would you mind sharing the binary (`esp32-20210623-v1.16.bin`) and a dump of the flash (acquired with `esptool.py -b 460800 read_flash 0 0x400000 flash_contents.bin`)?

lennardsisik, 2021-09-24:
it not work (esptool.py write_flash_status --non-volatile 0)
how to dump the flash? sorry im student and this is my first time to use esp32

radimkarnis, 2021-09-24:
run the command `esptool.py -b 460800 read_flash 0 0x400000 flash_contents.bin`, it will create a file `flash_contents.bin`

lennardsisik, 2021-09-24:
@radimkarnis  
[image]
what is next?

radimkarnis, 2021-09-26:
the flash contents were dumped into a file called `flash_contents.bin`, you can find it in `C:\Users\CLIENT\ve397-esp32\Scripts`. Please upload this file and the binary you are trying to flash (`esp32-20210623-v1.16.bin`) here.

lennardsisik, 2021-09-26:
https://drive.google.com/file/d/1T3MxJQk_eBh73ydwBTeMDLkPUnGeEf-a/view?usp=sharing 
@radimkarnis  this is the file

radimkarnis, 2021-09-27:
Thank you @lennardsisik, 
It seems like absolutely no data is being stored in the flash. This and the fact that `Chip erase completed successfully in 0.0s` (full erase usually takes around 15s) lead me to the explanation that the hardware is somehow faulty or damaged. I'm afraid moving to a new development board is probably the best option at this stage.

Will close this issue, feel free to reopen if you think there is more we can do to resolve the problem.

closed 2021-09-27 by radimkarnis

simtcrom, 2024-06-30:
@radimkarnis 
Sorry to bump on a closed issue.
But I have exact problem,

```
PS C:\Users\xxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\tools\esptool> python ./esptool.py read_flash_status
esptool.py v2.8
Found 2 serial ports
Serial port COM8
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ec:64:c9:d3:b1:f1
Uploading stub...
Running stub...
Stub running...
Status value: 0x1b64
Hard resetting via RTS pin...
```
```
PS C:\Users\xxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\tools\esptool> python ./esptool.py flash_id
esptool.py v2.8
Found 2 serial ports
Serial port COM8
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ec:64:c9:d3:b1:f1
Uploading stub...
Running stub...
Stub running...
Manufacturer: 85
Device: 2016
Detected flash size: 4MB
Hard resetting via RTS pin...
```
```
PS C:\Users\xxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\tools\esptool> python ./esptool.py write_flash_status --non-volatile 0
esptool.py v2.8
Found 2 serial ports
Serial port COM8
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ec:64:c9:d3:b1:f1
Uploading stub...
Running stub...
Stub running...
Initial flash status: 0x1b64
Setting flash status: 0x0000
After flash status:   0x1b64
Hard resetting via RTS pin...
```

uploading flash_contents.bin (renamed to .txt)
[flash_contents.bin.txt](https://github.com/user-attachments/files/16043327/flash_contents.bin.txt)

If you could please take a look before I ask for a replacement from seller.

radimkarnis, 2024-07-09:
@simtcrom I have tried to reproduce your issue, but flashing works normally on my side. Sorry, I don't see any obvious solutions to your problem. Please try switching your devkit.