espressif/esptool #663: Documentation on DISABLE_SDIO_HOST efuse?

henrygab, 2021-09-23:
* Operating system:
* Python version: (`python -V` to check this)

# Full esptool.py command line as run:

Example: python ./espefuse.py --port COM22 summary

# Full output from esptool.py

Note: MAC address replaced with `*` characters; Nothing in the output is surprising.

<details><summary>full output (click to expand)</summary></P>

```
C:\T\ESPixelStick>espefuse.py --port COM11 summary
espefuse.py v2.8
Connecting.....
EFUSE_NAME             Description = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Config fuses:
XPD_SDIO_FORCE         Ignore MTDI pin (GPIO12) for VDD_SDIO on reset    = 1 R/W (0x1)
XPD_SDIO_REG           If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset   = 1 R/W (0x1)
XPD_SDIO_TIEH          If XPD_SDIO_FORCE & XPD_SDIO_REG, 1=3.3V 0=1.8V   = 1 R/W (0x1)
CLK8M_FREQ             8MHz clock freq override                          = 53 R/W (0x35)
SPI_PAD_CONFIG_CLK     Override SD_CLK pad (GPIO6/SPICLK)                = 0 R/W (0x0)
SPI_PAD_CONFIG_Q       Override SD_DATA_0 pad (GPIO7/SPIQ)               = 0 R/W (0x0)
SPI_PAD_CONFIG_D       Override SD_DATA_1 pad (GPIO8/SPID)               = 0 R/W (0x0)
SPI_PAD_CONFIG_HD      Override SD_DATA_2 pad (GPIO9/SPIHD)              = 0 R/W (0x0)
SPI_PAD_CONFIG_CS0     Override SD_CMD pad (GPIO11/SPICS0)               = 0 R/W (0x0)
DISABLE_SDIO_HOST      Disable SDIO host                                 = 0 R/W (0x0)

Efuse fuses:
WR_DIS                 Efuse write disable mask                          = 0 R/W (0x0)
RD_DIS                 Efuse read disablemask                            = 0 R/W (0x0)
CODING_SCHEME          Efuse variable block length scheme                = 0 R/W (0x0)
KEY_STATUS             Usage of efuse block 3 (reserved)                 = 0 R/W (0x0)

Identity fuses:
MAC                    Factory MAC Address
  = **:**:**:**:**:** (CRC 0xfe OK) R/W
CHIP_VER_REV1          Silicon Revision 1                                = 1 R/W (0x1)
CHIP_VER_REV2          Silicon Revision 2                                = 0 R/W (0x0)
CHIP_VERSION           Reserved for future chip versions                 = 2 R/W (0x2)
CHIP_PACKAGE           Chip package identifier                           = 1 R/W (0x1)

Calibration fuses:
BLK3_PART_RESERVE      BLOCK3 partially served for ADC calibration data  = 0 R/W (0x0)
ADC_VREF               Voltage reference calibration                     = 1128 R/W (0x4)

Security fuses:
FLASH_CRYPT_CNT        Flash encryption mode counter                     = 0 R/W (0x0)
FLASH_CRYPT_CONFIG     Flash encryption config (key tweak bits)          = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE  Disable ROM BASIC interpreter fallback            = 1 R/W (0x1)
ABS_DONE_0             secure boot enabled for bootloader                = 0 R/W (0x0)
ABS_DONE_1             secure boot abstract 1 locked                     = 0 R/W (0x0)
JTAG_DISABLE           Disable JTAG                                      = 0 R/W (0x0)
DISABLE_DL_ENCRYPT     Disable flash encryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_DECRYPT     Disable flash decryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_CACHE       Disable flash cache in UART bootloader            = 0 R/W (0x0)
BLK1                   Flash encryption key
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLK2                   Secure boot key
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLK3                   Variable Block 3
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Flash voltage (VDD_SDIO) set to 3.3V by efuse.
```
</details><hr/>

# What is the expected behaviour?

The tool is displaying information for the `DISABLE_SDIO_HOST` efuse.  That's great.

But, where is the documentation for this fuse?  What changes, if this EFUSE is burnt / set to one?  I'm thrilled to read technical documentation.  IO12's value at boot can no longer cause boot problems on this board.

# Do you have any other information from investigating this?

I had hoped that this was a fuse to disable the SDIO Slave boot mode (see [ESP_Sprite's post](https://esp32.com/viewtopic.php?t=297#p1323)).  However, that is seeming less likely as I dig deeper.  Reaching out, hoping there's updated documentation on what the fuse bits do (esp. as v3 / WROOM-32E are now more common).

# Is there any other information you can think of which will help us reproduce this problem?

No.

radimkarnis, 2021-12-01:
Hello @henrygab,

I am sorry for not responding for so long. If you are still interested, I will try to provide some more information.

henrygab, 2021-12-01:
Hello @radimkarnis,

Yes, I would still appreciate help finding the documentation on the purpose of each fuse bit. 
 In particular, the `DISABLE_SDIO_HOST` bit sounded like it'd disable the probing for SDIO input during boot?

Thanks!

henrygab, 2021-12-09:
@radimkarnis ... out of curiosity, is there no public documentation yet, or is it just hard to find?

radimkarnis, 2021-12-09:
Hello @henrygab, 
thanks for your patience and sorry for keeping you waiting again.

I actually also do not know the exact function of this efuse and didn't find any external/internal documentation on it.
It exists (see  [here](https://github.com/espressif/esp-idf/blob/master/components/soc/esp32/include/soc/efuse_reg.h#L293-L298)), but lacks any description.

I have asked around but don't have the answer for you yet. We will try to either provide documentation or remove any irrelevant efuses.

henrygab, 2022-03-15:
@radimkarnis ... Just periodically checking if this has an answer.  Is monthly a good cadence to check for updates?

radimkarnis, 2022-03-25:
@henrygab sorry to keep you waiting, this got lost deep in the backlog.

I will try to provide an answer next week so you don't have to awkwardly wait for another month.

radimkarnis, 2022-04-06:
Hello @henrygab, 
SDIO host is described in the [ESP32 TRM](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf), chapter `9.4.1 SD/MMC Host Controller Architecture`.

Some more information can also be found in the [ESP-IDF documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sdio_slave.html). It states that 

> This means SDIO slave driver can only runs on the second set of pins while SDIO host is not using it.

The second set of pins also includes GPIO 12. Because of this, there is an option to burn the DISABLE_SDIO_HOST efuse, which essentially frees the gpio pins used by the host for other operations.

Can't really get more info on this. The SDIO host has been removed on newer chips and it seems the efuse itself wasn't really useful.

I am afraid I won't be able to help you any more with this.

henrygab, 2022-04-06:
Ah... Thanks for finding what you can.  I appreciate your help.

closed 2022-04-06 by henrygab

togsterx, 2023-04-28:
Disabling this made my esp32 wroom 32d 16MB useless. FLASHING a new bin file to the esp32 is not possible anymore.
espfuse.py still works.

DISABLE_SDIO_HOST      Disable SDIO host                                 = 1 R/W (0x1)

Be also aware that this cannot be reversed.

I did this in the hope my ESP32 will boot accordingly also when GPIO7 and GPIO8 are is use. I used those in the pcb design.
..