espressif/esptool #664: ESP32 can't flash - Write timeout

oldo439, 2021-09-27:
Hi.
I have an interesting problem with the TTGO T-Beam T22 v1.1 board with the ESP32 processor. I use Windows 10.
I bought 2 boards. The first board I flash with the firmware devel20210315-B10-full.bin from http://rdzsonde.mooo.com/download.html
I connected the display with the ILI9225 chip and used the following pins:
GPIO - Display
4 - SDA
13 - CLK
14 - RST
2 - RS
0 - CS
3.3 - VCC board power
GND - GND
The board worked well. The display showed the data.
After a few months I wanted to update the firmware, but I had to make a full flash (file devel20210919-B16-full.bin) because the firmware changed more (B10 to B16). According to the advice on the RDZSonde website.
But I forgot to use BOOT MODE (GPIO0 to GND). The board was working, I connected it to USB and used the flash_download_tool_v3.8.5 program. I set the parameters according to the instructions and pressed START. Then the display remained white and in the download tool it threw out an error and HALT. I didn't pay attention and tried the flash again. But again ERROR. Then I use BOOTMODE, so I connected GPIO0 to GND. But there was still ERROR, so I unplugged the whole display, but it didn't help. Still ERROR.
So I tried to use esptool.py directly. I used the command:
`esptool.py --chip ESP32 -p COM8 --before default_reset --after hard_reset write_flash 0x1000 devel20210919-B16-full.bin`

Listing:
esptool.py v3.1
Serial port COM8
Connecting....
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:61:05:0d:58:80
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x003fffff...
Compressed 4190208 bytes to 698639...
Writing at 0x00001000... (2 %)Traceback (most recent call last):
  File "C:\Users\Andrej\AppData\Local\Programs\Python\Python39\Scripts\esptool.py-script.py", line 33, in <module>
    sys.exit(load_entry_point('esptool==3.1', 'console_scripts', 'esptool.py')())
  File "C:\Users\Andrej\AppData\Local\Programs\Python\Python39\lib\site-packages\esptool.py", line 4575, in _main
    main()
  File "C:\Users\Andrej\AppData\Local\Programs\Python\Python39\lib\site-packages\esptool.py", line 4122, in main
    operation_func(esp, args)
  File "C:\Users\Andrej\AppData\Local\Programs\Python\Python39\lib\site-packages\esptool.py", line 3424, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
  File "C:\Users\Andrej\AppData\Local\Programs\Python\Python39\lib\site-packages\esptool.py", line 149, in inner
    return func(*args, **kwargs)
  File "C:\Users\Andrej\AppData\Local\Programs\Python\Python39\lib\site-packages\esptool.py", line 874, in flash_defl_block
    self.check_command("write compressed data to flash after seq %d" % seq,
  File "C:\Users\Andrej\AppData\Local\Programs\Python\Python39\lib\site-packages\esptool.py", line 462, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:\Users\Andrej\AppData\Local\Programs\Python\Python39\lib\site-packages\esptool.py", line 425, in command
    self.write(pkt)
  File "C:\Users\Andrej\AppData\Local\Programs\Python\Python39\lib\site-packages\esptool.py", line 388, in write
    self._port.write(buf)
  File "C:\Users\Andrej\AppData\Local\Programs\Python\Python39\lib\site-packages\serial\serialwin32.py", line 325, in write
    raise SerialTimeoutException('Write timeout')
serial.serialutil.SerialTimeoutException: Write timeout

I tried to use a second backup board, I used the same command via esptool.py as for the first board. And the flash was successful, the board worked. So the problem will not be in the computer or the cable.
I also tried the command"
`test_esptool.py COM8 esp32 TestFlashing`

This is part of the listing:
Running esptool.py tests...
E
Stdout:
**************************************************
Running C:\Users\Andrej\AppData\Local\Programs\Python\Python39\python.exe C:\Users\Andrej\.platformio\packages\tool-esptoolpy\test\..\esptool.py --chip esp32 --port COM8 --baud 115200 write_flash 0x0 images/sector.bin 0x1000 images/fifty_kb.bin...
b'esptool.py v3.1\r\nSerial port COM8\r\nConnecting......\r\nChip is ESP32-D0WDQ6-V3 (revision 3)\r\nFeatures: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None\r\nCrystal is 40MHz\r\nMAC: 3c:61:05:0d:58:80\r\nUploading stub...\r\nRunning stub...\r\nStub running...\r\nConfiguring flash size...\r\nFlash will be erased from 0x00000000 to 0x00000fff...\r\nFlash will be erased from 0x00001000 to 0x0000dfff...\r\nCompressed 4096 bytes to 4107...\r\nWriting at 0x00000000... (100 %)\r\nWrote 4096 bytes (4107 compressed) at 0x00000000 in 3.0 seconds (effective 10.9 kbit/s)...\r\nHash of data verified.\r\nCompressed 51200 bytes to 51226...\r\nWriting at 0x00001000... (25 %)\r\nTraceback (most recent call last):\r\n  File "C:\\Users\\Andrej\\.platformio\\packages\\tool-esptoolpy\\esptool.py", line 4582, in <module>\r\n    _main()\r\n  File "C:\\Users\\Andrej\\.platformio\\packages\\tool-esptoolpy\\esptool.py", line 4575, in _main\r\n    main()\r\n  File "C:\\Users\\Andrej\\.platformio\\packages\\tool-esptoolpy\\esptool.py", line 4122, in main\r\n    operation_func(esp, args)\r\n  File "C:\\Users\\Andrej\\.platformio\\packages\\tool-esptoolpy\\esptool.py", line 3424, in write_flash\r\n    esp.flash_defl_block(block, seq, timeout=timeout)\r\n  File "C:\\Users\\Andrej\\.platformio\\packages\\tool-esptoolpy\\esptool.py", line 149, in inner\r\n    return func(*args, **kwargs)\r\n  File "C:\\Users\\Andrej\\.platformio\\packages\\tool-esptoolpy\\esptool.py", line 874, in flash_defl_block\r\n    self.check_command("write compressed data to flash after seq %d" % seq,\r\n  File "C:\\Users\\Andrej\\.platformio\\packages\\tool-esptoolpy\\esptool.py", line 462, in check_command\r\n    val, data = self.command(op, data, chk, timeout=timeout)\r\n  File "C:\\Users\\Andrej\\.platformio\\packages\\tool-esptoolpy\\esptool.py", line 425, in command\r\n    self.write(pkt)\r\n  File "C:\\Users\\Andrej\\.platformio\\packages\\tool-esptoolpy\\esptool.py", line 388, in write\r\n    self._port.write(buf)\r\n  File "C:\\Users\\Andrej\\AppData\\Local\\Programs\\Python\\Python39\\lib\\site-packages\\serial\\serialwin32.py", line 325, in write\r\n    raise SerialTimeoutException(\'Write timeout\')\r\nserial.serialutil.SerialTimeoutException: Write timeout\r\n'
E
Stdout:
**************************************************
Running C:\Users\Andrej\AppData\Local\Programs\Python\Python39\python.exe C:\Users\Andrej\.platformio\packages\tool-esptoolpy\test\..\esptool.py --chip esp32 --port COM8 --baud 115200 write_flash 0x0 images/sector.bin...
b'esptool.py v3.1\r\nSerial port COM8\r\nConnecting....\r\nChip is ESP32-D0WDQ6-V3 (revision 3)\r\nFeatures: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None\r\nCrystal is 40MHz\r\nMAC: 3c:61:05:0d:58:80\r\nUploading stub...\r\nRunning stub...\r\nStub running...\r\nConfiguring flash size...\r\nFlash will be erased from 0x00000000 to 0x00000fff...\r\nCompressed 4096 bytes to 4107...\r\nWriting at 0x00000000... (100 %)\r\nWrote 4096 bytes (4107 compressed) at 0x00000000 in 3.2 seconds (effective 10.1 kbit/s)...\r\nHash of data verified.\r\n\r\nLeaving...\r\nHard resetting via RTS pin...\r\n'
Running C:\Users\Andrej\AppData\Local\Programs\Python\Python39\python.exe C:\Users\Andrej\.platformio\packages\tool-esptoolpy\test\..\esptool.py --chip esp32 --port COM8 --baud 115200 --before default_reset read_flash 0 4096 C:\Users\Andrej\AppData\Local\Temp\tmpqoti3sih...
b'esptool.py v3.1\r\nSerial port COM8\r\nConnecting....\r\nChip is ESP32-D0WDQ6-V3 (revision 3)\r\nFeatures: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None\r\nCrystal is 40MHz\r\nMAC: 3c:61:05:0d:58:80\r\nUploading stub...\r\nRunning stub...\r\nStub running...\r\n4096 (100 %)\r\n4096 (100 %)\r\nRead 4096 bytes at 0x0 in 0.4 seconds (81.9 kbit/s)...\r\nHard resetting via RTS pin...\r\n'
Readback 4096 bytes
Running C:\Users\Andrej\AppData\Local\Programs\Python\Python39\python.exe C:\Users\Andrej\.platformio\packages\tool-esptoolpy\test\..\esptool.py --chip esp32 --port COM8 --baud 115200 write_flash 0x1000 images/fifty_kb.bin...
b'esptool.py v3.1\r\nSerial port COM8\r\nConnecting....\r\nChip is ESP32-D0WDQ6-V3 (revision 3)\r\nFeatures: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None\r\nCrystal is 40MHz\r\nMAC: 3c:61:05:0d:58:80\r\nUploadi [... cut ...]
I tried to write sector.bin to offset 0x10000 and it burned fine. I also tried offset 0x100000, also ok. Wherever I wrote sector.bin, it was successful. But if the BIN file was bigger than 16kB, there was a write timeout.
Where can there be a mistake?
Could I damage something when I flash with the display connected?
RTS is connected to RST, so a reset was performed before burning. And also DTR is connected to GPIO0, so esptool switches the board to Download Mode. I verified that.
Thank you.
Sorry for my english, i use translator :)

radimkarnis, 2021-10-10:
Thanks for investigating further, according to your logs erasing the flash works as intended.

> A fatal error occurred: Failed to enter Flash download mode (result was 01060000)

Using the `--no-stub` option revealed a `106 = Operation or feature not supported` error being returned by the flash chip. This means that the flash chip for some reason refuses/doesn't register the write command.

This is strange behaviour, can you please confirm that the two boards are identical and this only happens only on one of them? If so, there is a possibility of a damaged or write protected flash chip.

oldo439, 2021-10-10:
I can confirm that both boards are the same, bought together. I tried to do a no-stub on the other board as well, it returned the same result 0106. It follows that esptool.py does not work very well with the ESP32 ROM loader. But the software loader (stub) works well.
Why does the flash chip refuses / doesn't register the write command? But only if I write more than 16kB. If I write, for example, 1kB, then the writing is successful. And it doesn't matter where I write the data (offset). This means that it is not write protected. And also that it probably won't be damaged.
I need to rule out all possibilities. What else could I test?
Could something go wrong when I had a display connected while writing the firmware? I used pins:
GPIO - Display
4 - SDA
13 - CLK
14 - RST
2 - RS
0 - CS
3.3 - VCC board power
GND - GND

How about pin GPIO02 - does it have any function? Could it affect the flash firmware in any way?
And what is the difference when writing data less than 16kB and more than 16kB? What changes when I write more data than 16kB? I want to look for a mistake there. This is probably the only difference between the boards. Otherwise, they behave the same way.
I read somewhere that during a flash, a block of data is first stored in RAM, in the meantime a part of the flash memory is erased and then the block of data is written to the flash. And then another block is loaded into RAM and the process is repeated.
Can there be a mistake somewhere?
On both boards I compared flash_status, chip_id, efuses, voltage on board. Everything is the same. I don't understand what's wrong.

radimkarnis, 2021-10-12:
> But the software loader (stub) works well.

Sorry, but now I am getting confused. Are you saying that flashing with the stub works fine?

To be honest, I am out of ideas. Esptool is well-tested for ESP32 using both the ROM and the stub. I would probably start thinking about some kind of HW issue. Can you try running the `flash_id` command and tell me the reported flash chip manufacturer and device ID?

The rdz_ttgo_sonde mainpage mentions the esptool command used to flash the binary:
```esptool --chip esp32 --port /dev/cu.SLAB_USBtoUART --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 <filename.bin>```

The `-z` option for compressed upload is used, does this make any difference?

Other than that, I really don't see anything wrong with your approach. I was not able to reproduce the issue on any of my boards.

oldo439, 2021-10-13:
So I still tried to burn the firmware without compression. Unbelievable, no mistake.
The board works.
But it certainly didn't solve the problem, I just found a way to upload the firmware there to make it work.
But look at how long it took. Almost 1 hour.

```
esptool.py --no-stub --chip esp32 --port COM8 --baud 115200 --before default_reset --after hard_reset write_flash -u --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 devel20210930-B16-full.bin
esptool.py v3.1a
Serial port COM8
Connecting....
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:61:05:0d:58:80
Enabling default SPI flash mode...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x003fffff...
Erasing flash...
Took 7.78s to erase flash block
Wrote 4190208 bytes at 0x00001000 in 3132.8 seconds (10.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

And now the question is what follows. What does this mean where there may be a mistake?
Stub doesn't work, compression makes mistakes.
I used a ROM loader, no compression and baud 115200.

oldo439, 2021-10-16:
Today I tried to write 3 times with the same command to check MD5.
The result was always the same. So corruption does not happens randomly.

And I also compared the current consumption on both boards. There is little difference. I made the comparison with EN turned on, so the firmware was not running. To eliminate the change in consumption when the other peripherals on the board are running. The good board had a consumption of 75 mA and the bad ones a little more, 85 mA. But I don't know why there is a difference. I did the measurement on a USB power supply. The battery in the board has not been inserted.

I have not tested the capacitors yet. There was no time yet.

radimkarnis, 2021-11-08:
Will close this issue now. Please reopen if you decide to continue with the investigation or if you feel like further discussion is needed.

closed 2021-11-08 by radimkarnis

josuah, 2023-02-11:
Someone I discussed with had this same problem, and could work around it.

Tested on two of these boards both presenting the same behavior:
https://bpi-steam.com/Leaf_S3_doc/en/

The flash chip could have its status read without trouble and erased as well.

```
python -m esptool --port COM11 write_flash_status --non-volatile 0
esptool.py v4.4
Serial port COM11
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:f2:a3:bc
Uploading stub...
Running stub...
Stub running...
Initial flash status: 0x0200
Setting flash status: 0x0000
After flash status:   0x0200
Hard resetting via RTS pin...
```

But programming did fail, at a percentage roughly between 1% and 6%:

```
esptool.py v4.4
Serial port COM10
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:c1:98:58
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00157fff...
Compressed 1406576 bytes to 928185...
Writing at 0x00000000... (1 %)Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
```

It was tried on a computer running Windows 10.

Trying the same thing on an Ubuntu laptop did work with the `esptool.py` freshly installed from pip, likely esptool v4.4 as well.

I suspect some antivirus wanting to prevent software to reprogram chips inside the computer, to fight against virus who would install malicious firmware, but could not confirm this hypothesis.

mrweaver, 2023-02-19:
I appear to be having a similar issue. I realised that previous code that I was running on ESP32-S3 had compiled and uploaded through the USB fine. I've been working on custom boards and running into this issue continuously over the last couple of weeks - the percentage of writing at failure in my case seems to arbitrarily change ( sometimes with early failure, sometimes with success). Made me think that it wasn't a hardware issue after all.

During writing of this reply, I seem to have potentially found a workaround- but haven't investigated thoroughly yet obvious.
I'm not sure about the versions of `esptool`

I could successfully upload using the following platform:
```platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.4.1/platform-espressif32-2.0.4.1.zip```

but not simply
```platform = espressif32```

That was the first I found that worked, I don't have time to look into it further at the moment, but it might get you started...? Oh maybe I've lead you astray with a rushed response - good luck!

oldo439, 2023-02-19:
Hi. I gave up a long time ago. I came to the conclusion that the board must be damaged. I bought two identical boards, one was a backup. The first one got damaged, so I started using the second, backup board. And it still works today. So it couldn't be blocked by antivirus or virus. Not even a bad version of esptool.
The only way I managed to upload the firmware to that old board was to reduce the baud rate to 9600. Then the write was successful.
I used the board for a while after that, but it was doing weird things. Sometimes the software did not start, etc. The damage was apparently quite severe, so even though I was able to load the FW at a slow speed, the board still struggled to use. So I eliminated her.
The second backup board is still working.

jcyinnovations, 2024-02-24:
I had this problem this week with a Lilygo T-Display S3 AMOLED board. Finally fixed it this afternoon by using their recommended reset: press and hold the boot button, then press the reset button, release the reset button then release the boot button. Board works normally after that.

I specified my board because I don't know if this solution will work universally across all ESP32 boards.