espressif/esptool #665: How can I upload to ESP32 board via OTG-USB from Android?

jinwook4217, 2021-09-28:
There was a problem in using esptool directly on Android because it was written in Python code.

So, what we are curious about is how to use esptool to learn more about how to upload to ESP32/ESP8266 board via USB from Android device via OTG.

Is it possible to run esptool code as cli command on Android?
Do I have to port all esptool code to java/kotlin?
Is there any open source to upload from android device using esptool ?

Sorry for the off-topic issue.

dobairoland, 2021-10-04:
I'm sorry, this is not supported. I can suggest to search for a solution of running a Python script from Android.

Beside that here is a list of other implementations which could help:
- [esptool-js](https://github.com/espressif/esptool-js) can run from a browser.
- There is also a C-based implementation: https://github.com/espressif/esp-serial-flasher
- Rust implementation is under development: https://github.com/esp-rs/esp-flash-loader
- Windows gui application: https://www.espressif.com/en/support/download/other-tools

However, none of the mentioned implementations offers the full set of features of Esptool.

closed 2021-10-04 by dobairoland

bdureau, 2022-02-02:
It would be really good to have it on android. I am building android applications that are uploading the latest firmware to many microcontrolers to make sure that my Android front end is compatible with the firmware . I have looked at many ESPTools applications in javascript or C# and I have tried to use the Physicaloid library to do an Android port and it is hardwork. I am trying to do the sync and I am not getting any response from the ESP32 .... Could it be possible to explain how the Python program works. Why do you need to encapsulate frames and replace some bytes such as 0xC0 by 0xDB and 0xDC?

radimkarnis, 2022-02-02:
Hi @bdureau,
the serial protocol used by esptool is explained [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/serial-protocol.html). This page also explains SLIP packet framing, which is the reason why some bytes are being replaced.

bdureau, 2022-02-03:
Thanks I am taking a look at it

bdureau, 2022-02-03:
If anybody has started a port on Android or java please contact me

bdureau, 2022-02-07:
Ok here is my issue. I just cannot get it to sync. I have tried with 2 different USB libraries (libUSB and Physicaloid) and it does not respond at all. I have set the baudrate to 115200
I am doing the reset and boot mode manually using the button. I can print a message which says "waiting for download"
Here is my code. I would appreciate if you could help
int sync() {
        int x;
        int response = 0;
        byte cmddata [] = new byte[36];

        cmddata[0] =  (byte)(0x07);
        cmddata[1] =  (byte)(0x07);
        cmddata[2] =  (byte)(0x12);
        cmddata[3] =  (byte)(0x20);
        for (x = 4; x < 36; x++)
        {
            cmddata[x] =  (byte)(0x55);
        }

        for (x = 0; x < 7; x++)
        {

            byte cmd =(byte)0x08;
            sendCommand(cmd, cmddata, 0);
        }
        return response;
    }
    void sendCommand(byte opcode, byte buffer[], int chk) {
        int CorrectReply = 0;

        int i = 0;
        byte data[]= new byte[8+buffer.length];
        data[0] =  (byte)0x00;
        data[1] =  opcode;
        data[2] = (byte) ((buffer.length )&0xFF);
        data[3] = (byte)((buffer.length >> 8)&0xFF );
        data[4] = (byte)((chk&0xFF) );
        data[5] = (byte)((chk >> 8)&0xFF) ;
        data[6] = (byte)((chk >> 16)&0xFF) ;
        data[7] = (byte)((chk >> 24) &0xFF);
        for (i = 0; i < buffer.length; i++) {
            data[8 + i] = buffer[i];
        }
        usbService.write(slipEncode(data));
    }
    byte[] slipEncode(byte buffer[]) {
        int ESP_FLASH_BLOCK = 0x400;
        int i = 0;
        byte encoded[] = new byte[ESP_FLASH_BLOCK * 2];
        encoded[i] = (byte) 0xC0;

        for (int x = 0; x < buffer.length; x++)
        {
            if (buffer[x] == (byte) (0xC0)) { encoded[i++] = (byte)(0xDB); encoded[i++] =(byte) (0xDC); }
            else if (buffer[x] == (byte) (0xDB)) { encoded[i++] = (byte)(0xDB); encoded[i++] = (byte)(0xDD); }
            else { encoded[i++] = buffer[x]; }
        }
        encoded[i++] = (byte)(0xC0);
        return buffer;
    }

bdureau, 2022-10-03:
Hello all
I have done a quick  Java port of the Esptool. Currently only working on ESP32
https://github.com/bdureau/ESPLoader

I am planning on porting it to Android so that it can flash my firmware from my phone

bdureau, 2022-10-08:
I have also done a port to Android using the physicaloid lib 
https://github.com/bdureau/BearConsole2/tree/master/app/src/main/java/com/altimeter/bdureau/bearconsole/Flash
However it needs improvement, I have added some timers so that it worked .... but is is slow!!!!

TheButlah, 2023-05-03:
> I'm sorry, this is not supported. I can suggest to search for a solution of running a Python script from Android.
>
> Beside that here is a list of other implementations which could help:
>
>     * [...]

should be mentioned that a fairly featureful tool/library exists in rust: https://github.com/esp-rs/espflash

bdureau, 2023-05-03:
Thanks for your answer
I eventually wrote my own pure java as well as Android implementation
 
DeÂ : Ryan Butler ***@***.*** 
EnvoyÃ©Â : 03 May 2023 11:26
Ã€Â : espressif/esptool
CcÂ : Bear altimeters; Mention
ObjetÂ : Re: [espressif/esptool] How can I upload to ESP32 board via OTG-USB from Android? (ESPTOOL-314) (#665)
 
I'm sorry, this is not supported. I can suggest to search for a solution of running a Python script from Android.
Beside that here is a list of other implementations which could help:
* [esptool-js]( <https://github.com/espressif/esptool-js> https://github.com/espressif/esptool-js) can run from a browser.
 
* There is also a C-based implementation:  <https://github.com/espressif/esp-serial-flasher> https://github.com/espressif/esp-serial-flasher
 
* Rust implementation is under development:  <https://github.com/esp-rs/esp-flash-loader> https://github.com/esp-rs/esp-flash-loader
 
* Windows gui application:  <https://www.espressif.com/en/support/download/other-tools> https://www.espressif.com/en/support/download/other-tools
However, none of the mentioned implementations offers the full set of features of Esptool.
should be mentioned that a fairly featureful tool/library exists in rust:  <https://github.com/esp-rs/espflash> https://github.com/esp-rs/espflash

xCarlost, 2024-11-08:
Hi @bdureau,
thank you for https://github.com/bdureau/BearConsole2! I tried it and it worked with my esp32dev. Unfortunately, it is not compatible with my esp32-S3. So I have implemented android support in pyserial (the dependency that esptool.py is using to establish serial connection). 
Here is my demo app: https://github.com/xCarlost/FirmwareFlasher
And the pr: https://github.com/pyserial/pyserial/pull/780

bdureau, 2024-12-07:
hello, I have now fixed it for the ESP32S3 and ESP32C3

bdureau, 2025-05-13:
> Any update ?

yes if you look at my github I have done all the ESP32 that can work on the Arduino environement