espressif/esptool #667: can't flash esp32 use 460800bps baud rate in wsl

guation, 2021-09-30:
# Operating system
Windows 10 Pro 21H1 + Ubuntu 20.04.3 LTS (WSL1)
# Python version
Python 3.8.10

# What Chip
ESP32-WROOM32

# What development board or other hardware is the chip attached to
NodeMCU DEVKITV1

# Is anything else attached to the development board, except for the serial flasher connections?
no connected

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
No IDE, Linux command prompt.

# Full esptool.py command line that was run:
`
esptool.py -p /dev/ttyS7 -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode qio --flash_size detect --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/Guation.bin
`

# Full output from esptool.py
```
$ esptool.py -p /dev/ttyS7 -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode qio --flash_size detect --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/Guation.bin
esptool.py v3.1-dev
Serial port /dev/ttyS7
Connecting........__
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: a8:03:2a:46:8b:a8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Traceback (most recent call last):
  File "/home/guation/esp-idf-v4.3.1/components/esptool_py/esptool/esptool.py", line 4535, in <module>
    _main()
  File "/home/guation/esp-idf-v4.3.1/components/esptool_py/esptool/esptool.py", line 4528, in _main
    main()
  File "/home/guation/esp-idf-v4.3.1/components/esptool_py/esptool/esptool.py", line 4053, in main
    esp.change_baud(args.baud)
  File "/home/guation/esp-idf-v4.3.1/components/esptool_py/esptool/esptool.py", line 149, in inner
    return func(*args, **kwargs)
  File "/home/guation/esp-idf-v4.3.1/components/esptool_py/esptool/esptool.py", line 864, in change_baud
    self._set_port_baudrate(baud)
  File "/home/guation/esp-idf-v4.3.1/components/esptool_py/esptool/esptool.py", line 334, in _set_port_baudrate
    self._port.baudrate = baud
  File "/home/guation/.espressif/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/serial/serialutil.py", line 299, in baudrate
    self._reconfigure_port()
  File "/home/guation/.espressif/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/serial/serialposix.py", line 517, in _reconfigure_port
    termios.tcsetattr(
termios.error: (5, 'Input/output error')
```

# Do you have any other information from investigating this?
Use flash_download_tool.exe can flash esp32 in 460800bps baud rate.
```
test offset :  4096 0x1000
case ok
test offset :  32768 0x8000
case ok
test offset :  65536 0x10000
case ok
======
CONNECT BAUD: 115200
============
..Uploading stub...
[... 127 lines cut ...]
MAC: a8:03:2a:46:8b:a8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x0003afff...
Flash params set to 0x0020
Compressed 26848 bytes to 16524...
Wrote 26848 bytes (16524 compressed) at 0x00001000 in 0.8 seconds (effective 270.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 384.0 kbit/s)...
Hash of data verified.
Compressed 174288 bytes to 94299...
Wrote 174288 bytes (94299 compressed) at 0x00010000 in 2.7 seconds (effective 512.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

dobairoland, 2021-10-04:
We are sorry for this but this seems to be a long standing issue: https://github.com/espressif/esp-idf/issues/4008

guation, 2021-10-04:
> We are sorry for this but this seems to be a long standing issue: [espressif/esp-idf#4008](https://github.com/espressif/esp-idf/issues/4008)

Can I set the 921600 bps baud rate to default?

dobairoland, 2021-10-06:
You can set the `ESPTOOL_BAUD` baud environment variable and all esptool runs without a `-b` argument will use that value.

dobairoland, 2021-10-11:
In ESP-IDF you can fix the baud rate by skconfig: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-esptoolpy-baud. You can set it in `idf.py menuconfig` and then copy the relevant config(s) from your `sdkconfig` into `sdkconfig.defaults` which you can use for all of your project. Further information can be found in the guide.

radimkarnis, 2021-10-13:
Closing. Feel free to reopen for any further discussions!