espressif/esptool #668: BIN esptool.py doesn't work?

exrector, 2021-10-01:
I'm not sure what's going on. I have NodeMCU PyFlasher5.  I use bin files for firmware. It stopped working a few days ago. I am trying to flash ESP8266 in the terminal 
esptool.py -p /dev/cu.STLAB_USB to UART write_flash --flash_mode dia --flash_size 4MB 0x0 homekidd_blinds.bin
or 
esptool.py --chip esp8266 write_flash --flash_mode dio --flash_size detect 0x0 homekidd_blinds.bin

in three cases, the firmware ends well, but port monitor shows and **The program DOESN'T work on the device**
17:54:53.668 -> sd d⸮⸮| ⸮$⸮| ⸮ l⸮ c|⸮⸮ ⸮2⸮⸮| c⸮ B⸮⸮oo⸮$nn⸮⸮⸮ b p⸮⸮drlrl ⸮r⸮⸮22NNbp ⸮. 
The funny thing is if you take a standard arduino sketch blink or WIFI Scan, then everything is flashed and works great. The program **works** on the device. The problem is only in the firmware of BIN files

in three cases, the same one is used esptool.py v3.1-dev (also was a study on 3.0). So why did flashing the bin files stop???
**_HELP_**
[image]
[image]

dobairoland, 2021-10-04:
> The funny thing is if you take a standard arduino sketch blink or WIFI Scan, then everything is flashed and works great. The program works on the device. The problem is only in the firmware of BIN files

Could you please specify what is the difference? I think Arduino uses BIN files as well. What is a "firmware of BIN files"?

exrector, 2021-10-04:
@dobairoland 
Sorry about my English. I probably can't express a normal thought in this language. I have a bin file. I can't flash it.(I can flash but the devices don't work) Nodemcu, command line and arduino ide. All three options use pylool. But it is possible to flash the device only in the Arduino ide. What is the error on my mac?

exrector, 2021-10-06:
@radimkarnis @dobairoland 
Yes, exactly. The flash happens perfectly. I'll repeat myself. I changed the mac. There are too many patches and workarounds on the old system. ESP is ESP32/8266. File mine.bin is absolutely working. One of them is now installed in esp8266 and the firmware works perfectly. A week ago, with a successful flash, the esp32/8266 boards stopped working. The firmware flashes but does not work. The monitor shows garbage or nothing.  Accordingly, wifi and the device does not appear on the wifi network. I think the latest macos monterey update is breaking something. I don't know how else to explain. Well install monterey

exrector, 2021-10-12:
**SUMMARIZING**
I took a sketch of the arduino ide "blink". Everything is extremely simple! And flash a raw sketch on the board. Everything works! The LED blinks. Life is wonderful.
The arduino ide has a "binary file export". I used it. I made a bin file from a 100% working sketch!
Guess what?
I flash this bin via NodeMCU PyFlasher and the result is 0. The device does not work, does not blink, nothing.
From what I conclude (and I have done a lot of work) that NodeMCU PyFlasher has some critical errors

exrector, 2021-10-14:
@radimkarnis @dobairoland 
This problem does not concern NodeMCU PyFlasher. NodeMCU PyFlasher is just a tool (shell). If the problem was in the application, then I would open the issue in the NodeMCU PyFlasher section. I was talking about critical errors that stretch from esptool

If the command esptool.py --port /dev/cu.usbserial-0001 write_flash 0x1000 Blink.info.node mcu.bin is correct, then why doesn't the device start?

`MacExrector:Downloads exrector$ cd /Users/exrector/Documents/Arduino/Blink
MacExrector:Blink exrector$ esptool.py --port /dev/cu.usbserial-0001 write_flash 0x1000 Blink.ino.nodemcu.bin
esptool.py v3.1
Serial port /dev/cu.usbserial-0001
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: e8:db:84:e2:81:fb
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00041fff...
Compressed 265040 bytes to 195064...
Wrote 265040 bytes (195064 compressed) at 0x00001000 in 17.3 seconds (effective 122.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
MacExrector:Blink exrector$ 
`
I used these two files.Previously, this worked well. These files are not working right now. Compiled in arduino BIN does not work either.
Try:
https://github.com/edx-Achim/ESP8266-HomeKit-Security-System/releases/tag/0.0.1
https://github.com/HomeKidd/ESP8266-Homekit-Smart-Blinds/releases/tag/1.0.1

radimkarnis, 2021-10-15:
> If the command esptool.py --port /dev/cu.usbserial-0001 write_flash 0x1000 Blink.info.node mcu.bin is correct, then why doesn't the device start?

A successful flash write by esptool doesn't automatically mean a working app. The `Hash of data verified` message is a guarantee that the bin file has been flashed perfectly (every single bit), but not a guarantee that it is vaild. You can use esptool to flash even a bin file full of gibberish. This means that your code or other settings are probably wrong.

I am sorry but I can't help you with debugging your application. I will close this issue now as it's clearly out of esptool scope.

exrector, 2021-10-21:
I was still able to flash one of the two BIN files with a specific command:
esptool.py -p /dev/cu.usbserial-0001 --baud 115200 write_flash -fs 32m -fm dio -ff 40m 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 main.bin