espressif/esptool #669: ESP8266 devkit flash_id fails (trace included)

digdugg, 2021-10-03:
# Operating system
Windows 10

# Python version
Python 3.8 x64

# What Chip
ESP8266 (hiletgo)
(For example: ESP8266, ESP32-PICO-D4, ESP32-WROOM32 module, etc)

# What development board or other hardware is the chip attached to

DevKitC

# Is anything else attached to the development board, except for the serial flasher connections?
USB

# Are you running esptool.py from an IDE such as Arduino or Eclipse?
Windows command prompt

# Full esptool.py command line that was run:

python esptool.py --trace flash_id

# Full output from esptool.py

esptool.py v3.2-dev
Found 1 serial ports
Serial port COM3
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.001 Read 93 bytes:
    040c8c1f006cdb24 241800ec1b92248c | .....l.$$.....$.
    90701272036ee492 6e7e12028c842424 | .p.r.n..n~....$$
    e0806204ec8c2424 70238302237b8384 | ..b...$$p#..#{..
    e46c606f248e7084 ec230c32620f6c6c | .l`o$.p..#.2b.ll
    db9b6f020cdf6f8c dbe26f6cdf7c1262 | ..o...o...ol.|.b
    1b0c00026c0c0404 6c6003726c       | ....l...l`.rl
TRACE +0.002 Read invalid data:
    040c8c1f006cdb24 241800ec1b92248c | .....l.$$.....$.
    90701272036ee492 6e7e12028c842424 | .p.r.n..n~....$$
    e0806204ec8c2424 70238302237b8384 | ..b...$$p#..#{..
    e46c606f248e7084 ec230c32620f6c6c | .l`o$.p..#.2b.ll
    db9b6f020cdf6f8c dbe26f6cdf7c1262 | ..o...o...ol.|.b
    1b0c00026c0c0404 6c6003726c       | ....l...l`.rl
TRACE +0.001 Remaining data in serial buffer:
.TRACE +0.118 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.002 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.113 No serial data received.
.TRACE +0.120 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.116 No serial data received.
[... 324 lines cut ...]
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.113 No serial data received.
.TRACE +0.120 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.002 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.112 No serial data received.
.
COM3 failed to connect: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting

A fatal error occurred: Could not connect to an Espressif device on any of the 1 available serial ports.

# Do you have any other information from investigating this?

Command has not succeeded over many tries.

# Is there any other information you can think of which will help us reproduce this problem?

Had another dev board in the pack (1 of 3) that flashed on the first time out.  Possibly corrupt SDK on the dev kit/board?

digdugg, 2021-10-04:
Connecting to the serial device over the COM port with baud 76800 bps yields the following output:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2408, room 16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0
tail 8
chksum 0xd8
csum 0xd8

2nd boot version : 1.6
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000

rf cal sector: 1017
rf[112] : 00
rf[113] : 00
rf[114] : 01

SDK ver: 2.0.0(5a875ba) compiled @ Aug  9 2016 15:12:27
phy ver: 1055, pp ver: 10.2

▒ԪTWT

```

radimkarnis, 2021-10-13:
Closing this issue due to inactivity. Feel free to reopen if the proposed solution doesn't work.

AHMERMANZOOR, 2025-02-06:
Hi I have been using micropython on esp32 wroom 32d but now on connecting to the computer it says no serial data received every time I try to erase the flash and I have tried to use manual boot setter method but that also does not work. I want to confirm that whether the chip is causing the problem or the cp2102 module.