espressif/esptool #670: will not write to  esp on  Kubuntu 20.04

krywenko, 2021-10-04:
* Operating system: kubuntu 20.04
* Python version: Python 3.8.10 and Python 2.7.18

tried all versions of  esptools 
error range from 

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

to 

raise FatalError(‘Failed to connect to %s: %s’ % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

it seams to only be related to the OS as i have an external harddrive with the previous version on  kubuntu installed  and when I boot to that OS every thing works fine as soon as I boot to the newer OS  all I get is errors trying  writing to esp chips - I can write to UNO board with out issue .. I was able to write to  one generic  esp chipset but I had to use   a different baud rate  then the default  most of my esp are wemos
I even drag copy of  Arduino IDE from the  the older  OS to the newer OS  same different  works fine on the   older OS but not on the newer OS 
 I can  reboot from new OS ( where it does not write to esp )  leave all hardware attached  boot into the older OS and everything works fine so  it has nothing to do with cables or esp devices  or computer hardware  --

dobairoland, 2021-10-04:
Hi @krywenko. There was a bug in the kernel. I think you are affected as well. This is a releated issue: https://github.com/espressif/esptool/issues/653. Some discussion about which versions are affected can be found here as well: https://github.com/espressif/esptool/issues/656

krywenko, 2021-10-04:
hi @dobairoland   - thank you for the reply --  I looked through the threads mention - - they said it was affecting  kernel down to 5.4.0-88  I tried to the lowest kernel I have  Linux 5.4.0-64-generic still the same issue  _  i find it weird that it can get  and request info from the ESP just can not write too it  so it knows how to communicate with the esp and it funny that it works fine with arduinos and other devices  to be strictly a kernel issue

closed 2021-10-13 by radimkarnis

amf99, 2021-10-16:
> For newbies like me, a very simple solution is to use "Advanced boot
> options" (or similar) at startup and select an older kernel. No fuss, no
> mess.

I tried the "Advanced boot options", when selecting an older kernel, I had
issues with my monitors, only one would work. Great to hear it worked for
you.